realsenseai/librealsense #8205: Depth data severely out of sync with color data

mmatl, 2021-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |  Ubuntu 16.04.7  | 
| Kernel Version (Linux Only)    |  4.15.0-132                                         | 
| Platform                                 | PC |
| SDK Version                            |  2                        | 
| Language                            | python                         | 
| Segment			|  Robot                       | 

### Issue Description

When using our D415, we sometimes get major issues with our depth stream. For example, here are the color and depth components of an image:

[image]

As you can see, the top two-thirds of the depth image is severely (several minutes) out of date with respect to the rest of the depth image and color image.

What could cause this issue?

MartyG-RealSense, 2021-01-21:
Hi @mmatl  I have seen this delay effect a few times in the past, either with the upper or lower section of the image.  Those cases can be found in the discussion in the link below, and sub-links at the bottom of that discussion.  In those cases, the problem also tended to occur sometimes and not all the time.

https://github.com/IntelRealSense/librealsense/issues/5986

In the case in the above link, updating the librealsense SDK and reducing stream resolution helped.  

Are you streaming live from the camera or playing back a recorded bag file, please?

mmatl, 2021-01-21:
Hi @MartyG-RealSense! Thanks for getting back so quickly. I'm streaming live from the camera at full resolution and 15 FPS. I'll try reducing the resolution and seeing if that helps things in any way. The SDK is current.

mmatl, 2021-01-21:
Will do. We're starting by reducing the framerate rather than the resolution. Want to see if that might help.

[image]

We're seeing it happen more regularly, across a number of different cameras. The firmware and SDK are current.

MartyG-RealSense, 2021-01-21:
I would recommend keeping 15 FPS as the minimum during testing instead of reducing below that FPS value, as there may be an increased chance of errors resulting from timeouts when using a very low speed such as 6 FPS.

I would recommend as a next testing step to see whether enforcing a constant FPS makes a positive difference,  You can do so by having Auto-Exposure **enabled** and setting an option called Auto-Exposure Priority to **disabled**.

I tracked down a Python script where a RealSense users sets the status of Auto-Exposure Priority.

https://support.intelrealsense.com/hc/en-us/community/posts/360037286634-Strange-FPS-when-enable-both-color-and-depth-streams

[image]

Another way to enforce a constant FPS is to have Auto-Exposure **disabled** and set a manual exposure value within a certain range.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

You can also test constant FPS in the RealSense Viewer program by setting Auto-Exposure to enabled in the **Stereo Module** options of the Viewer's side-panel and setting Auto-Exposure Priority to diabled in the **RGB** options.

MartyG-RealSense, 2021-02-05:
Case closed due to no further comments received.

closed 2021-02-05 by MartyG-RealSense

mmatl, 2021-03-01:
@MartyG-RealSense We're still experiencing this issue with an FPS of 15 and a resolution of 848x480. It's quite troublesome, as it's difficult to detect when this is happening to the cameras.

mmatl, 2021-03-01:
We implement all of the proposed changes and we hardware-reset the camera every so often.

MartyG-RealSense, 2021-03-01:
Could you test please whether setting **global time** to False (0) makes a difference please?  The link below has a script that demonstrates how to access the enable / disable control for global time in Python using **rs.option.global_time_enabled**

https://github.com/IntelRealSense/librealsense/issues/8112

mmatl, 2021-03-01:
@MartyG-RealSense Thanks for the rapid reply, appreciate it! I'll give this a shot and report back.