realsenseai/librealsense #8208: RealSense D415 depth map issues in Python

krkkvpl, 2021-01-21:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D415|
|**Serial Number**|047422061764|
|**Firmware Version**|05.12.09.00|
|**Advanced Mode**|NO|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|048123024133|
|**Firmware Update Id**|048123024133|

Hi,

I'm testing my RealSense camera on my laptop in a Python environment. I'm using these tutorials:
https://dev.intelrealsense.com/docs/python2

In examples such as 'opencv_viewer_example' and 'align-depth2color', I'm succesfully getting a stream for both RGB and depth, but the depth map is much different and a bit weird when compared to the RealSense Viewer. Here's a picture of the image from the Python code:
[image]

And here's the same scene from the Viewer:
[image]

Does anyone know what's happening here? As additional information, the depth filter in 'align-depth2color' tutorial seems to be working somewhat correctly.

Thank already, I'm looking forward to any help!

Best regards,
Joel Lindfors

MartyG-RealSense, 2021-01-21:
Hi @krkkvpl   The RealSense Viewer applies a range of post-processing filters and depth colorization configurations by default.  Scripts do not have these applied by default though and they have to be deliberately programmed into the script.   The output from programs can therefore look different from the image generated by the default settings of the Viewer.

[image]

MartyG-RealSense, 2021-01-28:
Hi @krkkvpl   Do you require further assistance with this case, please?

closed 2021-02-05 by MartyG-RealSense