realsenseai/librealsense #8213: What causes "incomplete video frame" if it works on other platforms?

Martzi, 2021-01-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | no GUI | 
| Operating System & Version |  Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    | 4.15.0-134-generic                                        | 
| Platform                                 | Intel NUC mini PC |
| SDK Version                            | latest                           | 
| Language                            |  python                          | 
| Segment			|  Robot                           | 

### Issue Description
I am trying to send point cloud data over tcp with python, but I constantly get an error message at the begining: "Frame didn't arrive in x seconds". I read previous issues and checked for debug messages with `rs.log_to_console(rs.log_severity.debug)`
When I start my python script, randomly gives "incomplete video frame detected" info and then stops because of the timeout.
The device is not overheating, I checked from other issues.
If I invert my setup so that a laptop (ubuntu 18, kernel: 5.4.0-62-generic) with the same camera is publishing point cloud through tcp and the nuc is the subscriber, it works fine. When I want to publish from the nuc to the laptop, I get these errors.
I paste the log result below, from start to crash:

_21/01 15:35:02,563 DEBUG [139661070350144] (ds5-factory.cpp:1113) try_fetch_usb_device(...) failed.
 21/01 15:35:02,563 INFO [139661070350144] (context.cpp:365) Found 1 RealSense devices (mask 254)
 21/01 15:35:02,566 DEBUG [139661070350144] (ds5-factory.cpp:1113) try_fetch_usb_device(...) failed.
 21/01 15:35:02,566 INFO [139661070350144] (context.cpp:365) Found 1 RealSense devices (mask 254)
 21/01 15:35:02,675 DEBUG [139661070350144] (ds5-private.h:403) Loaded Valid Table: version [mjr.mnr]: 0x0202, type 32, size 240, CRC: 19da20d0
 21/01 15:35:02,675 DEBUG [139661070350144] (backend-v4l2.cpp:1273) Recognized pixel-format 16-bit Depth
 21/01 15:35:02,677 DEBUG [139661070350144] (backend-v4l2.cpp:1273) Recognized pixel-format UYVY 4:2:2
 21/01 15:35:02,677 DEBUG [139661070350144] (backend-v4l2.cpp:1273) Recognized pixel-format 8-bit Greyscale
 21/01 15:35:02,678 DEBUG [139661070350144] (backend-v4l2.cpp:1273) Recognized pixel-format YUYV 4:2:2
 21/01 15:35:02,686 INFO [139661070350144] (sensor.cpp:1342) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 21/01 15:35:02,686 DEBUG [139661070350144] (backend-v4l2.cpp:1273) Recognized pixel-format 16-bit Depth
 21/01 15:35:02,688 INFO [139661070350144] (backend-v4l2.cpp:1479) Video node was successfully configured to Z16  format, fd 16
 21/01 15:35:02,688 INFO [139661070350144] (backend-v4l2.cpp:1481) Trying to configure fourcc Z16 
 21/01 15:35:02,689 DEBUG [139661070350144] (global_timestamp_reader.cpp:176) time_diff_keeper::start: _users_count = 1
 21/01 15:35:02,741 DEBUG [139661070350144] (hw-monitor.cpp:163) hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
 21/01 15:35:02,741 DEBUG [139661070350144] (types.cpp:78) recoverable_exception: hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
 21/01 15:35:02,743 INFO [139661070350144] (sensor.cpp:1342) Request: BGR8 Color, 
Resolved to: YUYV Color, 
 21/01 15:35:02,744 INFO [139661070350144] (backend-v4l2.cpp:1479) Video node was successfully configured to YUYV format, fd 22
 21/01 15:35:02,744 INFO [139661070350144] (backend-v4l2.cpp:1481) Trying to configure fourcc YUYV
 21/01 15:35:02,746 DEBUG [139661070350144] (global_timestamp_reader.cpp:176) time_diff_keeper::start: _users_count = 2
 21/01 15:35:02,946 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65259 out of 614655 bytes (10%)
 21/01 15:35:02,947 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:02,999 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,101 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,204 INFO [139660458194688] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,413 INFO [139660458194688] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,413 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,466 INFO [139660458194688] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,466 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,625 INFO [139660458194688] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,625 INFO [139660541437696] (types.h:1100) Incomplete video frame detected!
Size 65536 out of 614655 bytes (10%)
 21/01 15:35:03,661 WARNING [139660541437696] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 21/01 15:35:03,662 DEBUG [139660541437696] (sensor.cpp:348) FrameAccepted,Depth,Counter,1,Index,0,BackEndTS,1611239703632.486084,SystemTime,1611239703660.382812 ,diff_ts[Sys-BE],27.896729,TS,1611239703661.910645,TS_Domain,System Time,last_frame_number,0,last_timestamp,0.000000
 21/01 15:35:03,662 WARNING [139660458194688] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 21/01 15:35:03,662 DEBUG [139660458194688] (sensor.cpp:348) FrameAccepted,Color,Counter,1,Index,0,BackEndTS,1611239703630.285889,SystemTime,1611239703661.047119 ,diff_ts[Sys-BE],30.761230,TS,1611239703662.602539,TS_Domain,System Time,last_frame_number,0,last_timestamp,0.000000
 21/01 15:35:03,662 DEBUG [139660541437696] (archive.cpp:293) CallbackStarted,Depth,1,DispatchedAt,1611239703662.922852
 21/01 15:35:03,663 DEBUG [139660541437696] (archive.cpp:293) CallbackStarted,Depth,1,DispatchedAt,1611239703663.016602
 21/01 15:35:03,663 DEBUG [139660541437696] (sync.cpp:147) DISPATCH (TS: )--> Depth 0 1 1611239703661.910645 

[... 311 lines cut ...]
 21/01 15:35:09,579 DEBUG [139661070350144] (global_timestamp_reader.cpp:187) time_diff_keeper::stop: _users_count = 0
 21/01 15:35:09,579 DEBUG [139661070350144] (global_timestamp_reader.cpp:190) time_diff_keeper::stop: stop object.
 21/01 15:35:09,579 DEBUG [139660678563584] (global_timestamp_reader.cpp:254) Notification: time_diff_keeper polling loop is being shut-down
 21/01 15:35:09,579 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x23ad110 stopped
 21/01 15:35:09,579 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x23ad110 stopped
 21/01 15:35:09,579 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x23ad110 stopped
 21/01 15:35:09,579 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x23ad110 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x242c210 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x244ae50 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x242c210 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x244ae50 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x242c210 stopped
 21/01 15:35:09,580 INFO [139661070350144] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x244ae50 stopped
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:148) CallbackFinished,Depth,0,DispatchedAt,1611239709580.675537
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:155) Frame Callback [Depth#0] overdue. (Duration: 5060.996338ms, FPS: 30, Max Duration: 32ms)
[... cut ...]
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:225) All frames from stream 0x0x244ae50 are now released by the user
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:148) CallbackFinished,Color,3,DispatchedAt,1611239709580.787109
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:155) Frame Callback [Color#3] overdue. (Duration: 5084.119141ms, FPS: 30, Max Duration: 32ms)
 21/01 15:35:09,580 DEBUG [139661070350144] (frame-archive.h:225) All frames from stream 0x0x23ad110 are now released by the user_

MartyG-RealSense, 2021-01-21:
Hi @Martzi   Have you seen the recent case https://github.com/IntelRealSense/librealsense/issues/8202 that discusses possible causes of the **Incomplete frame detected!** warning, please?

Also, would it be possible to provide details about how you are making use of **TCP** in your project?  Is it similar to the implementation of TCP in Intel's open-source ethernet networking white-paper?

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

Martzi, 2021-01-22:
Hi, thanks for the fast reply,
I saw these cases, and I do no use vm. I connect the camera directly through USB 3.0. Luckily the camera does not overheat, stands between 20-30°C
For tcp transport I use pyzmq. It works if I publish data from the same camera, plugged into a laptop to the Intel NUC. If I want to revert this setup, it fails.

MartyG-RealSense, 2021-01-22:
Thanks very much @Martzi for the information.  Can you confirm please if the laptop that the camera is attached to is the **network** computer and the NUC is the **client**, similar to the image below from the open-source ethernet paper?

[image]

If this is true (the laptop is the network device and the NUC is the client device), does the TCP transport fail when trying to send data from the NUC client to the laptop server?

If the NUC does not have a camera attached, what data are you trying to send from the NUC to the laptop?

Thanks very much for your patience as I learn the details of your setup!

Martzi, 2021-01-23:
The camera is connected to the NUC via usb 3.0. This NUC forwards point cloud as a numpy array through tcp with zmq. The data transport is wireless, my laptop is connected to the NUC. In this case I got the error log.

For test purposes if I change the roles so that the camera is connected to my laptop, and the NUC is the client, receiving the point cloud, it works just fine.

I assume something is wrong with the NUC but I do not have more ideas, what I should check or try.

MartyG-RealSense, 2021-01-23:
Do you have the same networking software setup on the NUC as the laptop has, so that either of them can act as the server or the client depending upon which of them the camera is currently connected to?

Martzi, 2021-01-25:
Yes, exactly

MartyG-RealSense, 2021-01-25:
TCP ought to be able to work on NUC, as **Asyncore** has been successfully tested on NUC to establish a TCP connection.

https://dev.intelrealsense.com/docs/python2#section-stream-over-ethernet-python-example

What is the camera's performance like on NUC in non TCP applications such as the RealSense Viewer?  Do you experience any incomplete frame errors in the Viewer?

Martzi, 2021-01-28:
I tried the mentioned example, but I encountered a following error:
The code from the client (my laptop):
`$ python3 EtherSenseClient.py`
`Number of arguments: 1 arguments.`
`Argument List: ['EtherSenseClient.py']`
`sending "EtherSensePing"('224.0.0.1', 1024)`

and from the server (nuc):

`$ python3 EtherSenseServer.py`
`Number of arguments: 1 arguments.`
`Argument List: ['EtherSenseServer.py']`
`Recived Multicast message b'EtherSensePing' bytes from ('10.10.1.227', 60951)`
`Launching Realsense Camera Server`
`sending acknowledgement to ('10.10.1.227', 60951)`
`<_io.TextIOWrapper name='<stderr>' mode='w' encoding='UTF-8'> b'EtherSensePing'`
`error: uncaptured python exception, closing channel <__main__.EtherSenseServer 10.10.1.227:1024 at 0x7f3604ffdfd0> (<class 'TimeoutError'>:[Errno 110] Connection timed out [/usr/lib/python3.6/asyncore.py|read|83]` `[/usr/lib/python3.6/asyncore.py|handle_read_event|420] [/usr/lib/python3.6/asyncore.py|handle_connect_event|428])`

Martzi, 2021-01-28:
Actually, my problem is solved. There was a waterproof housing around the NUC. Now I looked into to see the usb connections and for unknown reasons, the usb 3.1 cable is converted to a micro usb, then back to "normal" usb type-a with 2.0 speed...
I connected the camera directly with the original cable, now works fine.
Thanks for your help and sorry for not checking this first.

MartyG-RealSense, 2021-01-28:
No problem at all.   I'm pleased to hear that you found the solution.  :)

closed 2021-02-05 by MartyG-RealSense