realsenseai/librealsense #8216: depth values unstable(inaccurate) and drop of frames  with D410 c#

Dalinaffeti, 2021-01-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D410 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
i'm writing a wound mesaurment c# application wihch need the measure the depth of a wound and the surface of it. in my case accuracy plays a big role that's why i'm opening this Issue.

###My solution approach:
providing the doctor an infrared streaming and after clicking "Capture" he will have a depth frame which the measurment points will be selected. Than i run my calculation Algorithm after deprojecting the Points.

###Process:
-Infrared frame streaming ber´cause i don't have an RGB Camera and the project must be with Panasonic FZ-M1 which have the D410.
-aligning (Stream.Depth)
-depth frame capture
-deproject of the Points
-calculating

###Issue:
-bad DepthFrame 
-inaccurate Values
-Drop of frames(it's get's bigger with post-processing)
-and i noticed a huge light influence on my depth frame.
i will be really helpful for your advices . i really searched a lot before opening this issue but stills no solution.

MartyG-RealSense, 2021-01-21:
Hi @Dalinaffeti   There was a C# wound measurement project making use of deprojection in November 2018 that you may have already seen, and may be a useful reference for improving the accuracy of the measured depth values if you have not seen it.

https://github.com/IntelRealSense/librealsense/issues/2657

Another RealSense user in the discussion, **HippoEug**, also developed their own wound measurement application in the C++ language instead of C#.

https://github.com/HippoEug/WetlingRealsenseES

Dalinaffeti, 2021-01-22:
indeed i saw it, actually my solution is based on his solution and cs-tutorial-3-processing. Unfortunately i can't refer to  HippoEug because it's something else with c++. I just want to fix the depth frame and the drop of frames in the camera to get the highest accuracy.
[image]
[image]
and this is my software results the first with colorized depth the second with the depthframe
[image]
[image]

MartyG-RealSense, 2021-01-22:
Are you using the recommended firmware version **5.12.7.100** with SDK **2.38.1** please?  If you have a newer firmware version then I would recommend updating your RealSense SDK version, as the firmwares that come after 5.12.7.100 require at least SDK **2.39.0** due to changes in those recent firmwares.

[image]

May I also ask these diagnostic questions:

1.  Do you still get the Viewer errors if you disable a feature in the Viewer settings called **GLSL** using the instructions in the link below?

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

2.  If disabling GLSL does not make a difference, does using **Medium Density** as the preset instead of **High Accuracy** help?  Medium Density provides a balance between accuracy and having a greater amount of detail ('fill rate') rendered on the image.

3.  If 1 and 2 do not work, does going to the **More** option near the top of the Viewer side-panel and selecting **Hardware Reset** from its menu to reset the camera stop the errors from occuring?

[image]

Dalinaffeti, 2021-01-25:
yes i'm using the recommended firmware version 5.12.7.100 with SDK 2.38.1
1. yes i do
2. it's worse with Medium Density
3. also tried it, the same

MartyG-RealSense, 2021-01-25:
I have done some testing with close-range hand scanning recently, so I attempted to apply some of the principles to your situation to find some settings that may be helpful.

The most problematic aspect in this particular application may be the D410 module.  Its minimum depth sensing distance means that it is more difficult to get closer to surfaces than the D420 and D430 modules before the image starts breaking up after the distance from the camera falls below the minimum.

There are ways to reduce the minimum distance and enable the camera to get closer to the surface.  The first is to use a lower resolution.  As higher resolution = greater accuracy, this may not be ideal for a measuring application.

The other method is to set an option called **Disparity Shift**.  Increasing the Disparity Shift value reduces the minimum distance (allowing the camera to get closer to a surface).  It also reduces the maximum observable depth distance, though this is not such a problem when performing close-range sensing.  A good Disparity Shift value to use is '50' for a balance between close range and maintaining some of the background detail.

Reducing the **Depth Scale** from its default of '0.001' to '0.0001' can also enable a better depth image at close range.

If your project was able to change to the **D415 Depth Module** board then you would have a couple more options that would improve the image.  The D415 Depth Module is the only board in the Depth Module range that has an integrated RGB sensor, which would solve the lack of RGB on the currently used D410 board. 

Using the above Depth Scale and Disparity Shift values may enable you to achieve improvement in a depth-only image with D410 though.

[image]

Dalinaffeti, 2021-01-28:
thank you for your reply . you gived of conveinet informations that can help me with my project.
here is my default streaming of depth and infrared a the moment:

[image]
[image]

and here is the custom ones with depth scale of 0.0001 and a Disparity Shift of 50: 

[image]
[image]

MartyG-RealSense, 2021-01-28:
Thanks very much @Dalinaffeti for the update.   Your depth image does look better with a **0.0001** depth scale and an Disparity shift of **50**.  

Do you require advice for further improvement of the image, please?

Dalinaffeti, 2021-01-28:
i guess  i hiave improvements due to my new calibration using the tare calibration tool.
here is my new reults using my app:
DepthFrame:
[image]
ColorizedDepth:
[image]
-i don't know why but as you can see i have a black area at the left top of the infraredFrame which is my main streaming because  i don't have a colorframe
- the drop of frames 
-  and it's looks like i have  a scale in the y-axis
- wenn i calculate the distance between two points i get sometimes values with +- 6cm difference to reality and sometimes huge numbers (maybe random ) like 336 cm when the actuel distance is 64 cm. Mybe due to the fact that i wrote my own deprojection and distance_3d methods because they are not included in .NET wrapper or the weird resolution that my depthFrame have or both.
- static Vector Deproject(DepthFrame depth, int x, int y, Intrinsics intrinsics)

        {

            float z = depth.GetDistance(x, y);

            return new Vector

            {

                x = z * (x - intrinsics.ppx) / intrinsics.fx,

                y = z * (y - intrinsics.ppy) / intrinsics.fy,

                z = z

            };

        }
        float dist_3d(Intrinsics intr, DepthFrame frame, PointF u, PointF v)
        {
            //float upixel[2]; // From pixel
            //float upoint[3]; // From point (in 3D)

            //float vpixel[2]; // To pixel
            //float vpoint[3]; // To point (in 3D)

            // Copy pixels into the arrays (to match rsutil signatures)
            //upixel[0] = u.first;
            //upixel[1] = u.second;
            //vpixel[0] = v.first;
            //vpixel[1] = v.second;

            // Query the frame for distance
            // Note: this can be optimized
            // It is not recommended to issue an API call for each pixel
            // (since the compiler can't inline these)
            // However, in this example it is not one of the bottlenecks
            float udist = frame.GetDistance((int)u.X, (int)u.Y);
            float vdist = frame.GetDistance((int)v.X, (int)v.Y);

            // Deproject from pixel to point in 3D
            Vector upoint = Deproject(frame, (int)u.X, (int)u.Y, intr);
            Vector vpoint = Deproject(frame, (int)v.X, (int)v.Y, intr);

            // Calculate euclidean distance between the two points
            return (float)Math.Sqrt(Math.Pow(upoint.x - vpoint.x, 2) +Math.Pow(upoint.y - vpoint.y, 2) +Math.Pow(upoint.z - vpoint.z, 2));
        }

Dalinaffeti, 2021-01-28:
i am using the 32 Bits librealsense and the following config
cfg.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
cfg.EnableStream(Stream.Infrared, 1280, 720, Format.Rgb8, 30 );
 does that have an impact?

MartyG-RealSense, 2021-01-28:
As far as I know, using 32-bit librealsense should not have an impact on the stream formats.

Regarding the black area in the corner, have you got auto-exposure enabled in your own application?

Dalinaffeti, 2021-01-28:
no  i wanted to write a code one to enable it but it didn't work i forget exactly why but i guess because i don't have IRsensor.

MartyG-RealSense, 2021-01-28:
I wonder whether the empty area could be generated by a lighting condition in that area that the camera is observing that the camera cannot adjust for with auto-exposure disabled.   An example of detail on an infrared image disappearing because of lighting conditions is shown here:

https://github.com/IntelRealSense/librealsense/issues/2875

Dalinaffeti, 2021-01-28:
i took that under consideration and i did a lot of experiences with lighting, i even measured and tracked the temperature of the camera . the lighting does plays a role. i exposed the camera to low and high lighting in my office.
Resultsa:
 when  the camera is exposed to a wall or an area within 2,3 meters the black area transforms into a vertical black stripe at the left side of the frame 
when the camera is exposed to a wide area of my office with a depth of  5 meters as in the following image then the black area takes almost the whole frame
[image]

MartyG-RealSense, 2021-01-28:
I ran extensive tests in the Viewer with infrared with a D415 at 1280x720 and RGB8 mode but was unable to replicate the black areas on the infrared image that you are experiencing.  Does it occur on every IR image or only infrequently?

Dalinaffeti, 2021-01-28:
i managed to use the autoexposure with the following code:
Sensor sensorIR = profile.Device.Sensors[0];
        

        foreach (var opt in sensorIR.Options)
        {
                
                        if (opt.Key == Option.EmitterEnabled)
                        {
                                opt.Value = 0f;
                        }
                        if (opt.Key == Option.EnableAutoExposure)
                        {
                                opt.Value = 1f; 
                        }
                
        }
but stills
[image]

Dalinaffeti, 2021-01-28:
it does at certain areas of my office and while moving the camera (i'm not moving it fast at all).

MartyG-RealSense, 2021-01-28:
Bearing in mind that you are using a D410 module, could you check any enclosure casing that you have around the module to see whether the corners of the enclosure might be in front of the field of view of the infrared lens?  If some part of a custom-made housing intrudes in front of the lens then it can impede the sensor's view and may cause the kind of obstruction seen in these images.

Also, the D415 and depth module numbers D415 and less use **rolling** shutters on the left and right infrared sensors that are slower than the fast **global** shutters on the D435 / D435i and D455 models and so are less suited to capturing motion.

Dalinaffeti, 2021-01-28:
the Panasonic Toughpad  have indeed an enclosure case but  i guess it is perfectly aligned with the camera lens
[image]
[image]

MartyG-RealSense, 2021-01-28:
I have not personally seen RealSense applications such as the Viewer run on one of the Windows tablets that has a D410 built in, such as your Panasonic.  On the depth image though, the black area on the left side is an expected effect.  It is a phenomenon called an **Invalid Depth Band**. 

Page 64 of the current edition of the data sheet document for the 400 Series cameras explains about the Invalid Depth Band.  To quote the section:

"The depth data generated with stereo vision uses the left imager as the reference for stereo matching resulting in a non-overlap region in the field of view of left and right imagers where we will not have depth data at the left edge of the frame.  Closer
scenes result in a wider invalid depth band than scenes at larger distances".

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

[image]

Dalinaffeti, 2021-01-29:
can i ask what is causing me alot of problems with is camera? and can it be solved ?

Dalinaffeti, 2021-01-29:
my Horizontal FOV: 68,45876
my Vertical FOV: 41,88355 

what do you think about those values?

Dalinaffeti, 2021-01-29:
when i use the capture example om the sdk the streaming have no drop of frames

Dalinaffeti, 2021-01-29:
i solved the black area problem by removing Align(Stream.Depth)

MartyG-RealSense, 2021-01-29:
Thanks very much for the update about your successful solution @Dalinaffeti - that is excellent to hear.  Do you require further advice, please?  Thanks!

Dalinaffeti, 2021-01-29:
- of course i do , now i even have another problem my depth frame resolution is smaller than my infrared frame resolution altough i set them to have the same resolution.
- also a distorted depthframe 
- when i calculate the distance between  2 points on it they are always wrong. that's why i opened this issue actually

MartyG-RealSense, 2021-01-29:
As you mention removing **Align(Stream.Depth)**, it sounds as though you are seeing the resolution difference in your own application and not the RealSense Viewer.

If you have still got the **depth unit scale** set to 0.0001 in your application then that is the likely cause of the difference between IR and the depth, whose image is probably zoomed in, restricting the view of the scene compared to the IR image.  That leaves the choice of setting the depth unit scale back to its default of **0.001** to restore the zoomed out view.

Dalinaffeti, 2021-02-02:
i did it on the viewer, i never changed depth unit in my application. if i set a perset in the viewer it will not affect the other applications that uses the camera as far as i know right ?

Dalinaffeti, 2021-02-02:
actually i checked it now and when i change depth unit in the viewer than it changes in my application too

Dalinaffeti, 2021-02-02:
[image]

Dalinaffeti, 2021-02-02:
can u please get me in touch with one of the intel team ?

MartyG-RealSense, 2021-02-02:
Setting a preset in the RealSense Viewer program would not affect the camera configuration used in other programs.

If you are seeing a difference in the Viewer between your depth resolution and infrared resolution, you may have a post-processing filter called **decimation** enabled.  If there is a blue icon beside the decimation filter (showing that it is enabled), please left-click on it to change it to red color (off) and see if that solves the resolution difference.

[image]

Dalinaffeti, 2021-02-02:
still the same i also checked the scale on my app it's 0,001

MartyG-RealSense, 2021-02-02:
As you are running the Viewer and your program on a Panasonic FZ-M1 Toughpad tablet, Intel cannot provide technical support for that tablet as it is not an Intel product.   It is the same for devices by other manufacturers that have a RealSense camera built into them as a component.  Panasonic are responsible for technical support for problems that might be resulting from the D410 camera that is integrated into the tablet.

The next step to take may be to use the **Contact Support** button on the link below to fill in Panasonic's contact form and ask them about the problems that you are experiencing,

https://na.panasonic.com/us/computers-tablets-handhelds/tablets/tablets/toughpad-fz-m1

MartyG-RealSense, 2021-02-04:
Hi @Dalinaffeti   I handled another case where there was a step-shaped gap in the depth image, this time with a D455 model on Linux, so I thought I would draw the case to your attention due to the similarities with your issue.

https://github.com/IntelRealSense/librealsense/issues/8309

MartyG-RealSense, 2021-02-06:
Hi again @Dalinaffeti  I consulted Intel for advice about the 'step shape' problem experienced in this case and in https://github.com/IntelRealSense/librealsense/issues/8309 and they agreed that it was caused by the alignment process.

MartyG-RealSense, 2021-02-12:
Hi @Dalinaffeti  Do you require further assistance with this case, please?  Thanks!

Dalinaffeti, 2021-02-12:
no i will try changing to d415

closed 2021-02-12 by Dalinaffeti

MartyG-RealSense, 2021-02-12:
Okay, thanks very much @Dalinaffeti for the update.