realsenseai/librealsense #8217: Librealsense library not detecting D435i with Ubuntu 18.04 WSL

rk122019, 2021-01-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   WSL (Ubuntu 18.04 LTS) on Windows 10 | 
| Kernel Version (Linux Only)    |  4.4.0-19041-Microsoft | 
| Platform                                 | PC  |
| SDK Version                            |  2.39.0 and 2.40.0 | 
| Language                            |  ROS                          | 
| Segment			|                            | 

### Issue Description
Hello! I'll start by saying I'm pretty new to Intel RealSense cameras / Linux / ROS / etc.  I recently submitted [this issue](https://github.com/IntelRealSense/realsense-ros/issues/1631) to the "realsense-ros" repository regarding an issue with using a D435i on a Windows 10 WSL (Ubuntu 18.04 LTS). Since writing that post and debugging, I have reason to believe the issue may be related to the librealsense library rather than the realsense-ros wrapper itself.

In this issue, I showed an error where the ROS wrapper returns a warning "No RealSense devices were found" when using a roslaunch command, even though the camera is definitely connected. I have tried to completely uninstall and reinstall everything multiple times with different attempts at fixing, with no luck. Those attempts include:

- Building everything from source [using this guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) rather than using the "ros-melodic-realsense2-camera" package
- Building different versions of the librealsense library in combination with the appropriate ROS wrapper (namely v2.40.0 with 2.2.20 wrapper, and v2.39.0 with 2.2.18 wrapper)
- Applying different kernel patches (first using `./scripts/patch-realsense-ubuntu-lts.sh`, then using `./scripts/patch-realsense-ubuntu-xenial-joule.sh`)

I also suspect the issue is with the librealsense library because when I run `realsense-viewer`, the command executes but nothing happens in the Ubuntu WSL. When I open the RealSense Viewer I downloaded on the Windows side of things, the camera works with no problems. 

I checked my settings in Windows 10, and my camera permissions are enabled, so I think Ubuntu should have access to the camera if it wants.

Is there anything special I need to know or do in order to use the librealsense library appropriately with WSL? Any suggestions would be greatly appreciated!

ev-mp, 2021-01-21:
@rk122019 hello,
Linux WSL provides a subset of Linux kernel functionality, and as of summer 20' it didn't have native USB support.
There are community reports about modifying WSL kernel for USB, such as https://github.com/microsoft/WSL/issues/5158, though in order to run Depth camera both USB and UVC/V4l modules must be resident.

rk122019, 2021-01-22:
Well that's a shame. Fortunately, I have the ROS wrapper up and running directly on Windows 10 now. The librealsense library seems to be working.

closed 2021-01-22 by rk122019

jtkhair, 2021-12-31:
I tried the [usbip](https://docs.microsoft.com/en-us/windows/wsl/connect-usb) and it seems that the wsl2 is able to access my device. In this case, the L515. 

However, the realsense viewer is not able to detect the L515/not able to enumerate the usb device.

I ran the `rs-enumerate-devices` and `dmesg -wT`. The outcomes are as follows; 

[image]

[image]

Is there any workaround for this?

gonultasbu, 2022-06-16:
As a follow-up. I followed the necessary steps and installed the native USB support to WSL2. For Ubuntu 18.04 I get the following errors and both D455 cameras are not detected by the launch scripts in realsense-ros.

```
rs-enumerate-devices
 16/06 11:51:21,553 ERROR [140393979213568] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
Could not create device - failed to set power state . Check SDK logs for details
 16/06 11:51:21,554 ERROR [140394059384512] (sensor.cpp:572) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details
 16/06 11:51:21,562 ERROR [140393987606272] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS

realsense-viewer
 16/06 11:50:36,967 ERROR [139922551854848] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 16/06 11:50:36,967 ERROR [139925121763200] (sensor.cpp:572) acquire_power failed: failed to set power state
 16/06 11:50:36,969 WARNING [139925121763200] (rs.cpp:310) null pointer passed for argument "device"
 16/06 11:50:36,969 WARNING [139925121763200] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
 16/06 11:50:40,990 ERROR [139922669291264] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 16/06 11:50:40,990 ERROR [139925121763200] (sensor.cpp:572) acquire_power failed: failed to set power state
 16/06 11:50:40,992 WARNING [139925121763200] (rs.cpp:310) null pointer passed for argument "device"
 16/06 11:50:40,992 WARNING [139925121763200] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
```