realsenseai/librealsense #8218: Realsense crashing and not detected on 5.4.5-rt3 kernel

ifollow-sofiane, 2021-01-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |  Ubuntu 18.04.5 LTS  | 
| Kernel Version (Linux Only)    | 5.4.5-rt3  | 
| Platform                                 | PC |
| SDK Version                            | V2.41.0 and 2.40.0      | 
| Language                            |ROS | 
| Segment			| Robot | 

### Issue Description

Hello all,
I am currently trying to run realsense camera on a rt patched kernel. I built ubuntu following the instructions given by : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md.

I applied patched kernel modules using the script: 
./scripts/patch-realsense-ubuntu-lts.sh 
of the realsense sdk and built the sdk from source installation. 

After installing ros packages (realsense2_camera and building the ros_crapper from https://github.com/IntelRealSense/realsense-ros.git). 
I try to launch rs_camera node (roslaunch realsense2_camera rs_camera.launch).
The node is working well for a while and after approximately 1 hour it crashes. And Realsense camera is no longer detected unless unplug/replug of the camera which is not acceptable in my case.
You will find attached the logs coming from the node first and then the logs from the kernel.
I think the problem is linked to the kernel version. Restarting the driver of the camera doesn’t make any change.
My questions are the following: 
Is there a way to make realsense work on rt kernels ? 
Is there a firmware patch to handle this kernel or is there one programmed ?

Thank you for any help, the logs are following.

### Ros logs:
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,897 ERROR [139692888499968] (types.h:313) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,897 ERROR [139692888499968] (sensor.cpp:547) release_power failed: xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,897 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,898 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,898 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,898 WARNING [139692709525248] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 21/01 15:00:45,898 ERROR [139692888499968] (types.h:313) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 21/01 15:00:45,899 ERROR [139692888499968] (types.h:313) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 21/01 15:00:45,899 ERROR [139692888499968] (sensor.cpp:547) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
process has finished cleanly
log file: /home/ifollow/.ros/log/fe062726-5be7-11eb-af5e-000732704958/camera-realsense2_camera-3*.log
[camera/realsense2_camera_manager-2] process has died [pid 13973, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ifollow/.ros/log/fe062726-5be7-11eb-af5e-000732704958/camera-realsense2_camera_manager-2.log].
log file: /home/ifollow/.ros/log/fe062726-5be7-11eb-af5e-000732704958/camera-realsense2_camera_manager-2*.log

### Kernel logs:

vcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[604582.532776] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
[604582.532779] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
[604582.532782] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
[604582.532969] usbcore: registered new interface driver uvcvideo
[604582.532972] USB Video Class driver (1.1.1)
[604583.928742] hid-sensor-hub 0003:8086:0B3A.0004: No report with id 0xffffffff found
[604587.679649] hid-sensor-hub 0003:8086:0B3A.0004: No report with id 0xffffffff found
[604643.530528] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604643.688390] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.249962] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.300422] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.452033] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.566630] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.617013] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.667437] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[604644.718381] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608410.414668] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608419.472722] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608419.523227] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608420.574603] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608420.625074] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608426.680678] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608426.731182] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608430.784815] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608430.835263] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608431.886576] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608431.936960] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608434.989862] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608435.040349] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608526.161261] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608527.212433] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608527.262830] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608528.314022] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608528.364423] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608529.415543] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608529.465873] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[608531.517348] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -19 (exp. 1024).
[608532.138066] uvcvideo: Failed to query (GET_CUR) UVC control 9 on unit 3: -19 (exp. 4).
[608532.138405] uvcvideo: Failed to query (GET_CUR) UVC control 9 on unit 3: -19 (exp. 4).
[608532.221668] usb 2-1: USB disconnect, device number 2
[608532.313543] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[608532.320492] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[608532.323214] hid-sensor-hub 0003:8086:0B3A.0004: No report with id 0xffffffff found
[608536.398647] nodelet[13995]: segfault at 7f0cc416bf35 ip 00007f0d0400ffd7 sp 00007f0cc8ff1020 error 4 in ld-2.27.so[7f0d04000000+29000]
[608536.398659] Code: 49 8b 18 48 8b 40 08 48 89 ca 48 c1 ea 20 48 8d 34 52 48 8d 34 f0 49 8b 02 48 89 74 24 08 48 01 c3 83 f9 07 0f 85 1d 01 00 00 <f6> 46 05 03 0f 85 97 00 00 00 49 8b 82 c8 01 00 00 45 31 c0 48 85

ev-mp, 2021-01-21:
@ifollow-sofiane hello, according to the log the device is forcibly disconnected while streaming
>[608532.221668] usb 2-1: USB disconnect, device number 2

In many cases this is caused by loose cables/connectors, jitter or insufficient power provided by host.
Apart of verifying the physical connection/switching USb ports, one of the robustness improvements (especially with small form-factor boards) is to use USB3 HUB with external power supply.
Another indication for it being not a physical but rather a power related issue is that it didn't happen momentarily, but the situation evolved for about 6 seconds with non-critical error messages being fired from [608526.161261]  to [608532.221668] till the disconnect event.

ifollow-sofiane, 2021-01-25:
@ev-mp hello and thank you for your comment.
I will try that today and let you know if this solved the issue.

ifollow-sofiane, 2021-01-26:
@ev-mp after testing it several times, the problem remains. The first time after 7 hours running the node died and I was able to relaunch the camera normally. The second time, it was like the realsense nodelet wasn't killed properly while the node died and the device was not found anymore. So using external powered USB looks like not solving the issue.  I should add that when we had the bug, we plugged a Lidar in the same USB hub during the test. Can it be the source of the pb, because it disturbed the voltage somehow ? The Lidar was not powered by the hub (external source)
logs are following 

ROS logs while diying: 

Done Setting Dynamic reconfig parameters.
[ INFO] [1611655763.967263075]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1611655763.967923412]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1611655763.967956987]: setupPublishers...
[ INFO] [1611655763.970685874]: Expected frequency for depth = 30.00000
[ INFO] [1611655763.984792672]: Expected frequency for color = 30.00000
[ INFO] [1611655763.993373165]: setupStreams...
[ INFO] [1611655763.993824622]: insert Depth to Stereo Module
[ INFO] [1611655763.993857862]: insert Color to RGB Camera
 26/01 11:09:24,496 ERROR [140077507610368] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[ERROR] [1611655764.497157607]: An exception has been thrown: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[camera/realsense2_camera_manager-2] process has died [pid 32119, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ifollow/.ros/log/8fd04ff4-5fbe-11eb-af5e-000732704958/camera-realsense2_camera_manager-2.log].
log file: /home/ifollow/.ros/log/8fd04ff4-5fbe-11eb-af5e-000732704958/camera-realsense2_camera_manager-2*.log
^C[camera/realsense2_camera-3] killing on exit
[ WARN] [1611655772.279801515]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down

ROS logs while trying to relaunch:

[ WARN] [1611656081.381060412]: ***************************************************
 26/01 11:14:41,387 ERROR [140165698729728] (types.h:313) get_dev_capabilities: Cannot open '/dev/video4 Last Error: Device or resource busy
 26/01 11:14:41,387 ERROR [140165698729728] (types.h:313) get_dev_capabilities: Cannot open '/dev/video2 Last Error: Device or resource busy
 26/01 11:14:41,387 ERROR [140165698729728] (types.h:313) get_dev_capabilities: Cannot open '/dev/video0 Last Error: Device or resource busy
 26/01 11:14:41,387 ERROR [140165698729728] (backend-v4l2.cpp:633) uvc meta-node with no video streaming node encountered: id- /dev/video1
vid- 8086
pid- b3a
mi- 0
unique_id- 2-1.3-14
path- /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video1
susb specification- 320
 26/01 11:14:41,387 ERROR [140165698729728] (backend-v4l2.cpp:633) uvc meta-node with no video streaming node encountered: id- /dev/video3
vid- 8086
pid- b3a
mi- 0
unique_id- 2-1.3-14
path- /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video3
susb specification- 320
 26/01 11:14:41,387 ERROR [140165698729728] (backend-v4l2.cpp:633) uvc meta-node with no video streaming node encountered: id- /dev/video5
vid- 8086
pid- b3a
mi- 3
unique_id- 2-1.3-14
path- /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.3/video4linux/video5
susb specification- 320
[ WARN] [1611656081.387648196]: No RealSense devices were found!

Kernel logs: 

uvcvideo: Failed to set UVC probe control : -32 (exp. 48).
[1025228.855120] uvcvideo 2-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[1025228.855125] uvcvideo 2-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[1025228.855129] uvcvideo 2-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[1025228.855291] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/input/input23
[1025228.855988] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[1025228.858249] uvcvideo 2-1.3:1.3: Entity type for entity Processing 7 was not initialized!
[1025228.858254] uvcvideo 2-1.3:1.3: Entity type for entity Extension 8 was not initialized!
[1025228.858257] uvcvideo 2-1.3:1.3: Entity type for entity Camera 6 was not initialized!
[1025229.076741] hid-sensor-hub 0003:8086:0B3A.0007: No report with id 0xffffffff found
[1025230.846948] nodelet[31460]: segfault at 7f2dc41ea318 ip 00007f2dd0c804a7 sp 00007f2d95938e80 error 4 in ld-2.27.so[7f2dd0c76000+29000]
[1025230.846955] Code: 08 48 89 44 24 18 48 8b 83 f8 02 00 00 48 85 c0 0f 84 15 01 00 00 8b 54 24 20 23 93 f0 02 00 00 48 8b 3c 24 8b 8b f4 02 00 00 <48> 8b 04 d0 48 89 fa 48 d3 ea 48 89 d1 48 89 c2 48 d3 ea 89 f9 48
[1025232.116940] usb 2-1: USB disconnect, device number 9
[1025232.116949] usb 2-1.1: USB disconnect, device number 10
[1025232.117942] usb 2-1.3: USB disconnect, device number 11
[1025233.605577] usb 2-1: new SuperSpeed Gen 1 USB device number 12 using xhci_hcd
[... 112 lines cut ...]
[1026341.146400] usb 1-1.1.3: SerialNumber: 110033420290738
[... cut ...]
[1026696.266955] usb 2-1.3: Set SEL for device-initiated U1 failed.
[1026696.273964] usb 2-1.3: Set SEL for device-initiated U2 failed.
[1026696.273970] usb 2-1.3: usb_reset_and_verify_device Failed to disable LPM
[1026696.277960] usb 2-1-port3: cannot disable (err = -71)
[1026697.308742] usb 2-1-port3: cannot disable (err = -110)
[1026697.316823] hub 2-1:1.0: hub_ext_port_status failed (err = -71)
[1026697.323819] hub 2-1:1.0: hub_ext_port_status failed (err = -71)
[1026697.323824] usb 2-1-port3: device 2-1.3 not suspended yet

RealSenseSupport, 2021-01-28:
@ifollow-sofiane  do you have any new updates? Can you run a test with the cameras connected directly to the root USB port without using a hub? As you suggested have you / can you perform a test without the Lidar connected?

ifollow-sofiane, 2021-01-29:
Hello @RealSenseSupport, we are still testing but updating the Firmwareto 5_12_10.0 and not plugging anything while the camera is plugged looks to be working. We are still performing other tests in this direction to see if the Firmware is resolving those usb issue even if there is another device plugged and we will let you know if the issue is completely solved.

RealSenseSupport, 2021-02-12:
Hello @ifollow-sofiane , have you been able to determine if the latest firmware resolves your USB issues with additional devices connected? Also, in the past have you experienced the connection problem when stopping and restarting a node?

ifollow-sofiane, 2021-02-16:
Hello, it looks that the latest firmware is solving the connexion issue. 
In the past just the node running alone was crashing and even after restarting the node it was not working until unplug then replug the camera.

closed 2021-02-16 by ifollow-sofiane

ifollow-sofiane, 2021-02-18:
Hello @RealSenseSupport, no I didn't test again this configuration.