realsenseai/librealsense #8219: Glad and KHR

arothenberg, 2021-01-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I'm trying to integrate some code that relies on glad. It throws all sorts of errors right now  using the sdk infrastructure. 
I noticed that in thirdparty/glfw/dep you have glad files. I tried to add includes and I tired other solutions I found to solve the subsequent errors and I  just caused more errors.

This may be a basic programing question, in which case I'll be happy to take it to a Glad message board(if there is one)
but is there  a simple best practices way for me to incorporate code with alternate glad dependencies? 

Your product has been good to me the last few months and I'm almost finished with the project I'm on - but my final task is  creating some graphical indicators (in my case  spinning needles - like a compass) and it's been a struggle figuring out how to use opengl properly because all example code fails in your environment.  I've worked a few tutorials in opengl at this point and they don't port for me and I cant seem to search a  solution. 
If I can get this  https://learnopengl.com/Getting-started/Transformations to work in your environment I could complete this project. 

Thanks.

MartyG-RealSense, 2021-01-21:
Hi @arothenberg  The link below has a discussion about a C++ user who was attempting to link Glad to librealsense in their project.  Whilst the case does not have a confirmation of whether the developer was successful, it may provide you with useful insights.

https://stackoverflow.com/questions/52477124/why-cant-the-c-compiler-find-librealsense

arothenberg, 2021-01-21:
Thank you Marty. Appreciate it. I'll have a look.

arothenberg, 2021-01-21:
I think that one is more about Cmake.  My libreasense sdk cmakes fine. 
And there is  a glad directory in the sdk that looks right and looks like its meant to be used. 
I think I'm making progress though so I'll close this out. It's hard though. 
I'll open something back up if I can't move forward.
Thanks.

arothenberg, 2021-01-21:
Hey, could you give me one link to someone doing a transformation of an image within the Realsense environment. 
That's all I need at this point. The amount of time I'm spending trying to do this is getting prohibitive. All the tutorials I use work fine. If I could get just any example I'd be good to go. It's been a week and I cant solve this. 

Thanks.

closed 2021-01-21 by arothenberg