realsenseai/librealsense #8221: [question] using rs.rs2_deproject_pixel_to_point() to convert 2D points to 3D points

monacv, 2021-01-22:
Hello, I want to learn how to use rs.rs2_deproject_pixel_to_point() in order to convert 25 keypoints that I have in 2D to 3D keypoints.

My keypoints are in this format:

[x0, y0, ....,x24,y24]

Thanks a lot for any guidance.

MartyG-RealSense, 2021-01-22:
Hi @monacv   A good place to start in learning **rs2_deproject_pixel_to_point** may be the tutorial in the link below, which uses Python code and utilizes the 2D coordinate system **left up is (0,0), right is x , down is y**.

https://medium.com/@yasuhirachiba/converting-2d-image-coordinates-to-3d-coordinates-using-ros-intel-realsense-d435-kinect-88621e8e733a

After that tutorial, the discussion in the link below should be a useful reference for further reading.

https://github.com/IntelRealSense/librealsense/issues/2488

MartyG-RealSense, 2021-02-05:
Case closed due to no further comments received.

closed 2021-02-05 by MartyG-RealSense

jcgarciaca, 2021-05-24:
Hi @MartyG-RealSense, 
I need to convert from 2D pixel (RGB) to 3D point, but with some considerations to keep in mind:
1. I have to convert 2D to 3D with previous data. It is, I am saving RGB image, IR image, Depth data and PLY file from SR300 camera (3 or 4 times a day) in a database, and later I need to analyze those data.
2. The RGB image is captured with 1920x1080 resolution

Then I have a couple of questions about `rs2_deproject_pixel_to_point`:
1. `depth` parameter is a float value, but I am not sure how to get that value. Is it got from `get_distance` function in `depth_frame` class? If it is, the `get_distance` function requires a pixel coordinates as input. Does it correspond to 2D pixel in depth frame? How to relate it to RGB image (specially if they have different resolution)?
2. I am just saving the depth frame data with `depth_frame = frames.get_depth_frame()` and `depth_image = np.asanyarray(depth_frame.get_data())` How can I calculate the `depth` value from this data?

Thank you

MartyG-RealSense, 2021-05-24:
1.  I think in order to answer this particular question I would need to know how the depth and RGB images are being stored (e.g as PNG images or in a bag file).

Saving the depth as a PNG image would be problematic, as most of the depth information is lost when saving depth to PNG.  Saving to **.raw** image format preserves the depth data.

Another approach is to save color as an image and depth as an numpy **.npy** file, like in the script in the link below.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

Once an npy file has been exported then you can perfom further work with it, such as loading it into **pytorch**.

https://towardsdatascience.com/beginners-guide-to-loading-image-data-with-pytorch-289c60b7afec

2.  In the link below, a RealSense team member provides a Python script that converts the depth image into numpy array data using ** np.asanyarray(depth_frame.get_data())**.  
 
It then multiplies that numpy depth value by the **depth unit scale** value of the camera (which the script retrieves) to convert the raw depth value to a real-world distance in meters and prints the value of that distance in meters.

https://github.com/IntelRealSense/librealsense/issues/3473#issuecomment-474637827

Given that it sounds as though you will not be able to access the SR300 camera directly at the time that you are performing your depth calculations, you could instead use a fixed hardcoded value as the depth scale to multiply the raw depth value by, as the depth scale does not change unless deliberately altered.   On the SR300 cames, the default depth unit scale value is **0.000125**

jcgarciaca, 2021-05-25:
Hi @MartyG-RealSense,
Thank you for your response. It was really useful.
I save the RGB image as PNG and depth as **npy** file. With your response I was able to run `rs2_deproject_pixel_to_point` and got the 3D point. I haven't tested the accuracy of received data, but shape of object seems well. Just a question to clarify, `intrinsics` and `pixel` for `rs2_deproject_pixel_to_point` are from depth sensor and depth image respectively, right?

If pixel is in depth image, then I need to convert from RGB to Depth pixel. I think I could use `rs2_project_color_pixel_to_depth_pixel` function. I got the extrinsics parameters for my cameras, but -as you mentioned- I don't have access to camera directly when performing depth calculations, do you know how to get `pyrealsense2.BufData` required for `rs2_project_color_pixel_to_depth_pixel` from numpy array?

Thank you

MartyG-RealSense, 2021-05-25:
Hi @jcgarciaca  Intrinsics are accessed from the hardware, whilst pixel relates to the image, yes.  The official **Projection** documentation of the SDK provides the following description. 

***************

Deprojection takes a 2D pixel location on a stream's images, as well as a depth, specified in meters, and maps it to a 3D point location within the stream's associated 3D coordinate space. It is provided by the header-only function rs2_deproject_pixel_to_point().  Intrinsic parameters can be retrieved from any rs2::video_stream_profile object via a call to get_intrinsics()

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

*************

In regard to your other questions: 

1.  In that example, it looks as though a **depth clamp** is being defined to specify the minimum and maximum depth values that can be rendered on th eimage, with depth values outside of that defined min-max range being excluded from the image.  It is similar to doing so using the SDK's **Threshold Filter** post-processing filter like the one implemented in Python in the link below.

https://www.programmersought.com/article/11995485679/

Each camera model has a default minimum depth sensing value in meters.  On D435 / D435i it is 0.1, on D415 is it 0.3 and on D455 it is 0.4.   So it makes sense for the example that you quoted to use a minimum depth value of 0.11 (meters) if it was designed for D435 / D435i.

2.  My understanding is that using Bufdata in the SDK is problematic.  In the link below, RealSense team member **ev-mp** has a detailed discussion with a RealSense user about it. Please read downwards through the comments from the point in the discussion that I have linked to below.

https://github.com/IntelRealSense/librealsense/issues/8394#issuecomment-782811459

jcgarciaca, 2021-05-26:
Hi @MartyG-RealSense,
I tried to create my own `rs2_project_color_pixel_to_depth_pixel` function but it didn't work well. Then, just for testing purpose, I tried to use the original function with instant data from camera but it didn't work either.

First, I get intrinsic and extrinsic parameters as follow:
```
profile = cfg.get_stream(rs.stream.color)
color_intr = profile.as_video_stream_profile().get_intrinsics()
...
profile = cfg.get_stream(rs.stream.depth)
depth_intr = profile.as_video_stream_profile().get_intrinsics()
...
depth_to_color_extrin = cfg.get_stream(rs.stream.depth).as_video_stream_profile().get_extrinsics_to(cfg.get_stream(rs.stream.color))
color_to_depth_extrin = cfg.get_stream(rs.stream.color).as_video_stream_profile().get_extrinsics_to(cfg.get_stream(rs.stream.depth))
```
Then I get depth data with `depth_frame = frames.get_depth_frame()` and color image with `color_frame = frames.get_color_frame()`.

However, the `rs2_project_color_pixel_to_depth_pixel` return value does not match in images. I tested with my original resolution (`1920x1080`) and with `640x480`, but both are wrong.

I do:
```
depth_point_ = rs.rs2_project_color_pixel_to_depth_pixel(
        depth_frame.get_data(), depth_scale,
        depth_min, depth_max, depth_intr, 
        color_intr, depth_to_color_extrin, color_to_depth_extrin, color_point
)
```

Such that red circle shows input pixel in color image and calculated pixel in depth image for:

1. `1920x1080` color image
[image]

2. `640x480` color
[image]

How could I fix it?

Thank you

MartyG-RealSense, 2021-05-26:
Thanks very much for the images.  I researched further and found a case with a similar misalignment between depth and color when using **rs.rs2_project_color_pixel_to_depth_pixel** 

https://github.com/IntelRealSense/librealsense/issues/7581

My understanding from the description on that case is that with their particular mis-alignment, the depth image is vertically higher than the correct color image.

https://github.com/IntelRealSense/librealsense/issues/7581

It was thought that there may be a problem in the Inverse Brown Conrady distortion model.  As a result, changes were introduced to Brown-Conrady in librealsense SDK **2.45.0**.

https://github.com/IntelRealSense/librealsense/pull/8759

pfcouto, 2023-02-26:
> Hi @jcgarciaca
> 
> 1. I think in order to answer this particular question I would need to know how the depth and RGB images are being stored (e.g as PNG images or in a bag file).
> 
> Saving the depth as a PNG image would be problematic, as most of the depth information is lost when saving depth to PNG. Saving to **.raw** image format preserves the depth data.
> 
> Another approach is to save color as an image and depth as an numpy **.npy** file, like in the script in the link below.
> 
> [#4934 (comment)](https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)
> 
> Once an npy file has been exported then you can perfom further work with it, such as loading it into **pytorch**.
> 
> https://towardsdatascience.com/beginners-guide-to-loading-image-data-with-pytorch-289c60b7afec
> 
> 2. In the link below, a RealSense team member provides a Python script that converts the depth image into numpy array data using **np.asanyarray(depth_frame.get_data())**.
> 
> It then multiplies that numpy depth value by the **depth unit scale** value of the camera (which the script retrieves) to convert the raw depth value to a real-world distance in meters and prints the value of that distance in meters.
> 
> [#3473 (comment)](https://github.com/IntelRealSense/librealsense/issues/3473#issuecomment-474637827)
> 
> Given that it sounds as though you will not be able to access the SR300 camera directly at the time that you are performing your depth calculations, you could instead use a fixed hardcoded value as the depth scale to multiply the raw depth value by, as the depth scale does not change unless deliberately altered. On the SR300 camera, the default depth unit scale value is **0.000125**

Hello @MartyG-RealSense how can I save the depth to .raw image format preserving the depth data, as you mentioned in that comment? And how can I read it later on? Thanks!

pfcouto, 2023-02-26:
Hello @MartyG-RealSense ! I am looking for a solution to send the RGB and Depth stream from one device to another. DO you know of any implementation that allows me to do that without losing precision in the depth stream?

However, initially I only need to send a frame (like a picture) and not a stream. Maybe using sockets, or just saving a frame of the RGB stream into a PNG file, and then send the file. However, what about the depth? Should I create a .npy file and send it? On the other end, I have to use the RGB and Depth just as I was using it directly from the camera. 
Thanks!

MartyG-RealSense, 2023-02-27:
Hi @pfcouto  A .raw file can be exported from the depth stream in the RealSense Viewer tool by clicking on the **Snapshot** camera-shaped option at the top of the depth stream's panel whilst the depth stream is enabled.

A RealSense team member shares C++ code for exporting a .raw file with scripting at https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-379703868

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412810987 and https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331 discuss reading a .raw file into the RealSense SDK.

***********

The 'network device' system is being removed from the RealSense SDK in an upcoming version, so it would not be a suitable option for your project anyway if you need to update to a newer SDK version in future.  Its Python-based networking predecessor **EtherSense** should still function though.  It is based around the TCP protocol.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

I do not know of a solution for sending IMU data from one computer to another unless you use ROS and create a networking connection between two machines, like in Intel's guide for RealSense at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

Would it be possible to you to use a computer 'remote access' tool such as  Google's**Chrome Remote Desktop** to control a computer that a camera is attached to from another computer so that you are viewing the camera data on the second machine?

https://remotedesktop.google.com/

pfcouto, 2023-02-27:
Hello @MartyG-RealSense I am tryinng the solution that you suggested `https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server` however, I am getting an error when running `setup.py`, here is the output:

`WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Requirement already satisfied: numpy in /usr/lib64/python3.11/site-packages (1.22.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Requirement already satisfied: python-crontab in /usr/local/lib/python3.11/site-packages (2.7.1)
Requirement already satisfied: python-dateutil in /usr/lib/python3.11/site-packages (from python-crontab) (2.8.2)
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil->python-crontab) (1.16.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Requirement already satisfied: opencv-python in /usr/local/lib64/python3.11/site-packages (4.7.0.72)
Requirement already satisfied: numpy>=1.21.2 in /usr/lib64/python3.11/site-packages (from opencv-python) (1.22.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Requirement already satisfied: cron_descriptor in /usr/local/lib/python3.11/site-packages (1.2.35)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Traceback (most recent call last):
  File "/home/pedro/Documents/Escola/Investigacao/librealsense/wrappers/python/examples/ethernet_client_server/setup.py", line 16, in <module>
    system_cron = CronTab(tabfile='/etc/crontab', user=False)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/crontab.py", line 213, in __init__
    self.read(tabfile)
  File "/usr/local/lib/python3.11/site-packages/crontab.py", line 272, in read
    with codecs.open(filename, 'r', encoding='utf-8') as fhl:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen codecs>", line 906, in open
FileNotFoundError: [Errno 2] No such file or directory: '/etc/crontab'`

I do have pyrealsense2 installed, but it outputs an error. Also, how can I create /etc/crontab ?

pfcouto, 2023-02-27:
Hi @MartyG-RealSense I manged to solve the error. However, when I try to run the server and client, I get these errors. I only changed the IPs highlighted in the picture. Can you help? Thanks!

[image]

MartyG-RealSense, 2023-02-28:
It looks as though the problem is centered around the IP address.

There is a more detailed guide about EtherSense in a PDF document at the link below.

https://www.intelrealsense.com/wp-content/uploads/2019/03/Depth_Camera_powered_by_Ethernet_WhitePaper.pdf

Page 3 of the guide about IP addresses states that EtherSense should be running on the 'server-host', which will be the computer / computing board that the camera is attached to.  Can you confirm please that you have installed EtherSense on the computer that has the camera attached to it, please?