realsenseai/librealsense #8222: Camera tracking / trajectory code samples for Intel D455

zc08, 2021-01-22:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.36.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I'm trying to track the 6DoF camera pose and obtain the trajectory from an Intel D455 camera. 
However I found that the several camera tracking and trajectory code examples in the repository only applies to T265 camera, and does not run with D455.
Are there code samples for camera tracking / trajectory using D455 ? Thanks in advance !

MartyG-RealSense, 2021-01-22:
Hi @zc08   The **rs-motion** SDK example program illustrates how the orientation of the D455 (and also the D435i) can be obtained from the IMU.

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

A RealSense user wrote their own Python solution and shared the script:

https://github.com/IntelRealSense/librealsense/issues/4391#issuecomment-510701377

The angle of the device can also be obtained without using the IMU by using a **plane-fit**, a method used by the SDK's **Depth Quality Tool** program.  Resources about this can be found in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894154/comments/360013322694

zc08, 2021-01-22:
Thank you for the answer ! 
I'm aware of the rs-motion example. But it seems to be only tracking the 3DoF orientation, while I need to track both the orientation and position (6DoF)...

MartyG-RealSense, 2021-01-22:
The **AUBOi5-D435-ROS-DOPE** project uses RealSense D435, Pytorch and ROS to get the 6 degrees of freedom (6-DOF) pose of recognized objects.  It also uses other software components such as **Unreal Engine 4** and **NVidia Deep Learning Data Synthesizer(NDDS)*.  

So whilst it may be a 6DOF solution for the 400 Series cameras, it is a complex project because of the number of software parts involved in it.

https://github.com/avinashsen707/AUBOi5-D435-ROS-DOPE

closed 2021-02-05 by MartyG-RealSense