realsenseai/librealsense #8223: Android 29 running error

whosmyqueen, 2021-01-22:
`2021-01-22 14:43:22.956 20547-20736/com.xiangchuang.myapplication E/UsbManager: exception in UsbManager.openDevice
    java.lang.SecurityException: User has not given 10227/com.xiangchuang.myapplication permission to access device /dev/bus/usb/002/002
        at android.os.Parcel.createException(Parcel.java:2071)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:739)
        at android.hardware.usb.UsbManager.openDevice(UsbManager.java:559)
        at com.intel.realsense.librealsense.DeviceWatcher.addDevice(DeviceWatcher.java:115)
        at com.intel.realsense.librealsense.DeviceWatcher.invalidateDevices(DeviceWatcher.java:78)
        at com.intel.realsense.librealsense.DeviceWatcher.access$000(DeviceWatcher.java:16)
        at com.intel.realsense.librealsense.DeviceWatcher$1.onDeviceAttach(DeviceWatcher.java:44)
        at com.intel.realsense.librealsense.Enumerator.notifyOnAttach(Enumerator.java:102)
        at com.intel.realsense.librealsense.Enumerator.access$200(Enumerator.java:18)
        at com.intel.realsense.librealsense.Enumerator$MessagesHandler.handleMessage(Enumerator.java:144)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:213)
        at android.os.HandlerThread.run(HandlerThread.java:67)`

MartyG-RealSense, 2021-01-22:
Hi @whosmyqueen   There was a case involving the error **exception in UsbManager.openDevice java.lang.SecurityException: User has not given ... permission to access device** in December 2019.  

https://github.com/IntelRealSense/librealsense/issues/5388

The RealSense user who created the case reported that updating their camera's firmware driver resolved the problem.

https://github.com/IntelRealSense/librealsense/issues/5388#issuecomment-623078025

whosmyqueen, 2021-01-23:
@MartyG-RealSense It doesn't solve my problem. My project is Android 29. Your SDK can only run on Android 27.

MartyG-RealSense, 2021-01-23:
Setting **TargetSDKVersion** to '27' is a method used to fix problems with using Android 10 and 11 devices with the RealSense Android wrapper.  

There was a RealSense user in November 2020 who was successfully using the wrapper with Android 29 on some Android devices but not others.  This fits past patterns where problems may occur with using the wrapper on Android devices of a certain model but it works on other devices.

https://github.com/IntelRealSense/librealsense/issues/7691#issuecomment-720876395

The link below has an example of problems occurring with particular models:

https://github.com/IntelRealSense/librealsense/issues/6546#issuecomment-643180328

closed 2021-02-07 by MartyG-RealSense