realsenseai/librealsense #8224: Preset loading seems broken

doisyg, 2021-01-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | lastest| 
| Operating System & Version |   Ubuntu  20.04.01 | 
| Kernel Version (Linux Only)    |  5.8        | 
| Platform                                 | x86 |
| SDK Version                            |  v2.41                         | 
| Language                            |                        | 
| Segment			|  Robot                       | 

### Issue Description

Loading of json preset files seems broken for HighAccuracy Preset.
Steps to reproduce:
- Open realsense viewer
- Activate Stereo module and see that the output is behaving like it is on default
- Switch preset in the dropdown menu to High Accuracy and see that the outputs switches indeed to High Accuracy (much less points)
- Use `Save current stereo modules settings to file` to save the configuration file to HighAccuracySaved.json
- Close and restart realsense viewer
- Use `Load pre-configured stereo modules settings` button to load the HighAccuracySaved.json configuration file that you saved before
- Start the Stereo module and see on the output that the **settings are not applied**, confirm by selecting High Accuracy in the dropdown (the output will then properly switch)

This seems to afect json loading from the SDK and ros wrapper too

doisyg, 2021-01-22:
Continuing, High Accuracy settings (without save/load json) seems to not be applied with the following scenario:

- Open realsense viewer
- Select High accuray with Stereo Module to off
- Start stereo module and see **High Accuracy** settings not applied in the output

doisyg, 2021-01-22:
To add to the confusion, I believe the difference in outputs are due to the laser emitter behind switched off when selecting High Accuracy in some conditions

MartyG-RealSense, 2021-01-22:
Hi @doisyg   I was able to replicate your findings by following your procedure using a D435i.  I found that a workaround was to:

1.  Load in the **HighAccuracySaved.json** preset whilst the camera is inactive, so that it is added to the drop-down lit of presets.
2.   Activate the stream with one of the existing builtin SDK presets applied, such as **Default**.
3.   Select the **HighAccuracySaved** custom preset from the list whilst the stream is active.

The stream then correctly changed to the typical data conditions present when **High Accuracy** is used.

Whilst this workaround is okay for the Viewer, it would be problematic to implement when loading the preset in ROS if you had to set Default as the json first and then the **HighAccuracySaved..json** preset file.  I wonder if you could set the Default preset initially in the roslaunch instruction, and then put an arg in the launch file to overwrite that setting with the path to HighAccuracySaved..json.

https://github.com/IntelRealSense/realsense-ros/issues/753
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L5

doisyg, 2021-01-22:
> The stream then correctly changed to the typical data conditions present when **High Accuracy** is used.

I am not sure anymore if the difference in the output in these conditions is due to improper application  of Advanced Settings from or just a bug turning off the laser emitter (try to toggle the emitter off/on to see if it impacts the output)

MartyG-RealSense, 2021-01-22:
I re-ran the testing and did not experience any change in the displayed **Laser** status of the emitter in the options side-panel.

It is normal though for the depth image to potentially degrade when the emitter staus is changed from **Laser** (On) to **Off**, or if the value of the **Laser Power** setting is reduced.  So if the depth quality degraded if the emitter was manually toggled off, that would be an expected behavior.

If you are referring to how the depth image loses some depth detail when **High Accuracy** settings are used, this is also normal because pixels that have a high confidence of measurement accuracy are prioritized.  If you wanted a detailed image that also had accuracy, the **Medium Density** preset provides a balance between these two priorities.

MartyG-RealSense, 2021-01-29:
Hi @doisyg   Do you require further assistance with this case, please?  Thanks!

doisyg, 2021-02-01:
I believe the bug which is switching off the laser emitter without updating the status when changing the preset in realsense-viewer should be fixed to avoid confusion

MartyG-RealSense, 2021-02-02:
At this point, it may be worth waiting for the next SDK version and coming back to the issue if it is still present.

I note that custom presets are addressed in a PR commited in the past 24 hours at the time of writing this.

https://github.com/IntelRealSense/librealsense/commit/7e72d615bf4c02156c182ad1f44665453830d83d

MartyG-RealSense, 2021-02-16:
Hi @doisyg   The new librealsene version **2.42.0** is now available.  Would it be possible please to check whether your problem is corrected in this version, please?   Thanks!

doisyg, 2021-02-16:
Issue indeed solved with the last lib version. Thx, closing.

closed 2021-02-16 by doisyg

MartyG-RealSense, 2021-02-16:
Excellent news - thanks very much for the update!