realsenseai/librealsense #8226: C# DNN OpenCV Example

bmils, 2021-01-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435/D455 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  v2.41.0  | 
| Language                            |  C# | 

### Issue Description
In April of 2020 I decided to convert the DNN OpenCV C++ example to C#. I was able to do that successfully and it was working perfectly fine. I just recently tested the code again and I keep getting an error when testing on the D435 and D455. The example I based it on is found here:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn

It crashes on the following line:
```C#
 colorMat = colorMat[crop];
```
With the following error:
"OpenCvSharp.OpenCVException: 0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols"

I ran the DNN C++ example and it worked perfectly fine. No code has been changed since I last tested it in April of 2020.
To convert it, I copied line by line over to C# (see below):

```C#
using System;
using System.Linq;
using System.Threading;
using System.Windows;
using System.Windows.Controls;
using System.Windows.Media;
using System.Windows.Media.Imaging;
using Intel.RealSense;
using OpenCvSharp;
using OpenCvSharp.Dnn;

namespace DNNTest
{
    /// <summary>
    /// Interaction logic for MainWindow.xaml
    /// </summary>
    public partial class MainWindow : System.Windows.Window 
    {
        //all dnn setups
        const int inWidth = 300;
        const int inHeight = 300;
        const float WHRatio = inWidth / (float)inHeight;
        const float inScaleFactor = 0.007843f;
        const double meanVal = 127.5;
        //all objects that the DNN is looking for
        String[] classNames = {"background",
                             "aeroplane", "bicycle", "bird", "boat",
                             "bottle", "bus", "car", "cat", "chair",
                             "cow", "diningtable", "dog", "horse",
                             "motorbike", "person", "pottedplant",
                             "sheep", "sofa", "train", "tvmonitor"};
        //pipeline receives all video data
        private Pipeline pipeline;
        //used to help color the depth stream
        private Colorizer colorizer;
        //thread streaming
        private CancellationTokenSource tokenSource = new CancellationTokenSource();
        //intel realsense processing block
        private IProcessingBlock block;

        //Function to write the video frame to the screen
        static Action<VideoFrame> UpdateImage(Image img)
        {
            var wbmp = img.Source as WriteableBitmap;
[... 171 lines cut ...]
            dm = dm * depth_scale;
            return dm;
        }

        //close the threading source
        private void control_Closing(object sender, System.ComponentModel.CancelEventArgs e)
        {
            tokenSource.Cancel();
        }

        //function that sets up the video windows
        private void SetupWindow(PipelineProfile pipelineProfile, out Action<VideoFrame> depth, out Action<VideoFrame> color)
        {
            using (var p = pipelineProfile.GetStream(Stream.Depth).As<VideoStreamProfile>())
                imgDepth.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            depth = UpdateImage(imgDepth);

            using (var p = pipelineProfile.GetStream(Stream.Color).As<VideoStreamProfile>())
                imgColor.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            color = UpdateImage(imgColor);
        }
    }
}
```
I apologize for a sort of source code dump, but I have been debugging for a good while and I don't understand what could have changed to cause this error to keep occurring. Am I missing something?

MartyG-RealSense, 2021-01-23:
Hi @bmils   What was the most recent librealsense SDK version that your code worked on, please?

bmils, 2021-01-25:
@MartyG-RealSense I can't recall the exact version, but whatever the most recent SDK version was in April 2020.

I installed the entire v2.41.0 SDK from [https://github.com/IntelRealSense/librealsense/releases](https://github.com/IntelRealSense/librealsense/releases) (I am assuming that also updates the C# wrapper or does it not?). Then I replaced the DLLs that were in my project with the newly installed DLLs from the 2.41.0 SDK. I then rebuilt the solution and ran it with these results.

I do not have the UWP driver installed, so that shouldn't be an issue.

MartyG-RealSense, 2021-01-25:
If your most recent SDK version that the project worked with was installed in April 2020 then it was likely SDK version **2.34.0**, as the next version 2.35.0 was not released until May 31 2020.  

The transition between 2.34.0 and 2.35.0 was a notable one for the reason that 2.34.0 produced continuously generating timing errors for some RealSense users.  

Because the project apparently jumped from 2.34.0 to 2.41.0, it is difficult to isolate the point at which the project broke.  So I would recommend testing different SDK versions in jumps of two (2.36, 2.38.1, 2.40.0).  That should help to identify the point at which compatibility with your C# project ended.

I am not able to compile and test code, only read scripts posted on the forum and advise on them.

bmils, 2021-01-25:
@MartyG-RealSense Thanks for looking into it, I appreciate the help. I'll do what you suggested and try to test different SDK versions and comment on this if I get it working.

MartyG-RealSense, 2021-02-07:
Hi @bmils   Do you require further assistance with this case, please?   Thanks!

bmils, 2021-02-08:
Hi @MartyG-RealSense I do not require any further assistance with this. Closing the issue now, thanks.