realsenseai/librealsense #8227: Getting Couldn't Resolve Request exception in C# SDK.

rajrajeswaran, 2021-01-22:
Camera Model                              D415
Firmware Version                          05.12.07.100
Operating System & Version        Window 10
SDK Version                            C# SDK version 2.19 
Language                            C#         | 

### Issue Description
I am getting "Couldn't resolve requests" exception in my C# Forms application using Intel C# SDK version 2.19 (Firmware: 05.12.07.100).  The Intel RealSense viewer is working just fine.  The application used to work just fine and not sure what is causing this error.  Logger message below:

22/01 16:34:39,077 DEBUG [14228] (context.cpp:100) Librealsense VERSION: 2.19.0
 22/01 16:34:39,089 INFO [14228] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/01 16:34:39,114 INFO [14228] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/01 16:34:39,135 INFO [14228] (tm-context.cpp:39) LibTm version 0x130003000006af
 22/01 16:34:39,135 DEBUG [14228] (ds5-factory.cpp:568) try_fetch_usb_device(...) failed.
 22/01 16:34:39,172 DEBUG [14228] (context.cpp:100) Librealsense VERSION: 2.19.0
 22/01 16:34:39,173 INFO [14228] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/01 16:34:39,187 WARNING [8848] (win-backend.cpp:194) RegisterClass failed.
 22/01 16:34:39,192 INFO [14228] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/01 16:34:39,205 DEBUG [14228] (tm-context.cpp:34) Failed to create TrackingManager
 22/01 16:34:39,205 DEBUG [14228] (ds5-factory.cpp:568) try_fetch_usb_device(...) failed.
 22/01 16:34:39,488 INFO [9452] (win-helpers.cpp:85) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element not found."
 22/01 16:34:39,501 DEBUG [9452] (ds5-factory.cpp:568) try_fetch_usb_device(...) failed.
 22/01 16:34:39,704 INFO [9452] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 22/01 16:34:39,704 INFO [9452] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 22/01 16:34:39,705 INFO [9452] (win-helpers.cpp:85) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 22/01 16:34:39,705 WARNING [9452] (sensor.cpp:366) Unregistered Media formats : [ W10  ]; Supported: [ ]

MartyG-RealSense, 2021-01-23:
Hi @rajrajeswaran   As a starting point in investigating your case, may I ask whether you have installed version **6.11.160.121** of the UWP Windows driver on your computer?  This driver version has been reported to cause errors with the RealSense C# wrapper when installed.  This driver is not required to use RealSense with Windows unless you need RealSense compatibility with Microsoft Universal Windows Platform (UWP) and so can be removed, or replaced with the driver's earlier **6.1.160.14** version if you do need UWP compatibility.

[image]

There was also a past C# wrapper case where the **Failed to resolve the request** error was found by the RealSense user in that case to be caused by another process trying to access the camera.

https://github.com/IntelRealSense/librealsense/issues/7989#issuecomment-747954425

rajrajeswaran, 2021-01-25:
@MartyG-RealSense - Thank you for your quick response.  Here is what i have for the drivers for both RGB and Depth sensors.  I have 6.1.160.21 version running.
[image]
[image]

and Thank you for referencing my old ticket.  The errors i received earlier and now are different so i was not sure.  I will revert back to another earlier version.  It is possible my organization pushed updates that caused the driver to update.  I will report back.

rajrajeswaran, 2021-01-25:
@MartyG-RealSense - tried to roll back to earlier version but i was unable to.  I then completely uninstall the device and installed really old version of the driver.  The C# SDK now works fine.
[image]

MartyG-RealSense, 2021-01-25:
The driver date of a driver sometimes has little relation to the age of the device that it is associated with, hence why you can get a 2006-dated driver with a device released years later.  I have the same driver details on my recently released D455 camera.  It usually does not have any effect on the successful performance of the device.   

It's great to hear that the C# wrapper is working for you now!

rajrajeswaran, 2021-01-25:
@MartyG-RealSense - Thanks again for your help.  You may close this at your convenience.

MartyG-RealSense, 2021-01-25:
Thanks very much @rajrajeswaran

closed 2021-01-25 by MartyG-RealSense