realsenseai/librealsense #8228: best Realsense model for toothpick

vipelz, 2021-01-23:
Hi everyone,
I would like to purchase an Intel realsense for a project.
Before buying it, I would like to know which model of Realsense is the best to sees a toothpick with stereo/depth module and with what quality? 
Because a toothpick is pretty thin.
Could someone post some pictures of how the different realsense sense the toothpick?
Thanks

This is a picture I've found online:
https://m.imgur.com/a/04pmrWp

This could be an example scenario of my project
[image]

MartyG-RealSense, 2021-01-23:
Hi @vipelz   You are correct that the small thickness of a toothpick makes it a challenging object to take a depth image of.  I set up a scene with a plain backdrop an a toothpick mounted vertically standing like in your image, and ran tests with all of the RealSense depth camera models.  I was able to produce a clear and close-up image of the toothpick only with the **RealSense D435i** camera, and with a specific configuration that mapped RGB color to depth in a 3D **point cloud** image.

[image]

The configuration that this image was achieved with in the **RealSense Viewer** program is:

-  RealSense D435i (it has a minimum depth sensing distance of 0.1 meters, enabling the camera to be positioned close to objects).
- 3D point cloud mode, with RGB color mapped to depth points
- A **depth unit scale** set to **0.0001** instead of the default 0.001 scale.

vipelz, 2021-01-23:
Wow you are so kind.
I just ordered a D435i. Could you please upload here the current stereo module setting to file for the realsense viewer?

Thank you

I just ordered a D435i

MartyG-RealSense, 2021-01-23:
You are very welcome.  :)

The **Custom** json preset that I used to create these images is built into the Viewer's list of pre-made configurations and is its default preset when the Viewer is launched, so there is no need to upload a json.

The **Depth Unit Scale** option is under the **Controls** sub-section of the Stereo Module options.  You can input 0.0001 manually by left-clicking on the pencil icon beside the option.

You can enter 3D point cloud mode by left-clicking on the **3D** option in the top corner of the Viewer.  You can map RGB to the depth points by enabling Depth and RGB, and then left-clicking on the **Texture** option and selecting **Color** from its menu if it does not automatically switch to the RGB-aligned image.

MartyG-RealSense, 2021-01-30:
Hi @vipelz  Do you require further assistance with this case, please?   Thanks!

GabrielMLB, 2021-02-03:
Perfect thank you very much for sharing, I will make the suggested changes and evaluate the RMS error.

I am using the D435i model.

I shared in the link below the example file that I am testing.

[https://1drv.ms/u/s!At7Yi4YwjQcBnxmZ78dZ5mNcAkdX?e=fW06gD](url)

closed 2021-02-10 by MartyG-RealSense