realsenseai/librealsense #8230: Depth changes over a period of time - Intel D415

vishnukumarts, 2021-01-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.35                          | 
| Language                            |  c++                       | 

### Issue Description
I mounted the intel D415 camera at a distance of 2.1m viewing a ground plane.  I ran a test camera sees a ground for 24 hours.  What I am expecting ground depth should not change over a period of time.  I am doing a plane fit algoritm(same like rs-depth-quality tool).   My depth data starting with 2105mm and it increasing upto 2122mm over a period of time.  Data taken for every three minutes, captured in the excel sheet.

Adding attachment file of depth data got over a period of time.  How to avoid [this?]([url](url
[consolidation-depth-data.xlsx](https://github.com/IntelRealSense/librealsense/files/5862098/consolidation-depth-data.xlsx)
))

MartyG-RealSense, 2021-01-24:
Hi @vishnukumarts   May I first ask whether the lighting in the location that the camera is in is the same across the 24 hour period or whether the lighting level can change during that period.

vishnukumarts, 2021-01-24:
Lighting is mostly typical office lighting(100-200 lux).  Lighting won't change much during test.

MartyG-RealSense, 2021-01-24:
Thank you very much.  And is it fluorescent lighting such as ceiling strip-lights instead of lights with bulbs?  Fluorescent lights contain heated gas that can flicker at frequencies that are hard to see with the human eye and introduce noise into the image.  

Using an FPS that is close to the operating frequency of the lights may help to reduce the effect.  For some lights it may be 30 FPS and for others it may be 60 FPS.

vishnukumarts, 2021-01-24:
My powerline frequency is 50Hz.  If i set the powerline frequency to 50Hz,  whether it will be solved?

MartyG-RealSense, 2021-01-24:
The subject of 50hz line frequency in regard to flicker is discussed in the link below.

https://community.intel.com/t5/Items-with-no-label/How-or-when-to-get-d435-IR-25fps-support/td-p/687187

There was also a case with a medical monitoring project that ran for multiple hours including during the night, and the RealSense user in that case addressed flickering under fluorescent lights by doing camera calibration and performing post-processing filtering. 

https://support.intelrealsense.com/hc/en-us/community/posts/360047112993/comments/360011849194

closed 2021-02-07 by MartyG-RealSense