realsenseai/librealsense #8232: T265 is detected in query devices but it erros out with 'not connected' with D435i

yosoufe, 2021-01-25:
This does not work, What do I do wrong? I was trying to follow [this tutorial](https://dev.intelrealsense.com/docs/rs-multicam) but in python.
```.py
import pyrealsense2 as rs
import numpy as np 
import time 
print(rs.__file__)

import atexit

pipelines = []

@atexit.register
def cleanup():
    print('clean up')
    for p in pipelines:
        try:
            p.stop()
        except:
            pass

ctx = rs.context()

for dev in ctx.query_devices():
    print(f'device: {dev.get_info(rs.camera_info.name)}')
    p = rs.pipeline(ctx) # ctx
    cfg = rs.config()
    dev_serial = dev.get_info(rs.camera_info.serial_number)
    cfg.enable_device(dev_serial)
    p.start(cfg) ## <<<<------------------------------------ it errors out here for the T265
    pipelines.append(p)

print('pipelines started')
s = time.time()
time.sleep(1)
```

This generates the following output
```
$ python bug_example.py 
/home/yousof/Software/programming/pyenv/versions/3.7.8/lib/python3.7/site-packages/pyrealsense2/__init__.py
device: Intel RealSense D435I
device: Intel RealSense T265
Traceback (most recent call last):
  File "bug_example.py", line 27, in <module>
    p.start(cfg)
RuntimeError: No device connected
clean up
```

I read somewhere that T265 should get loaded to context first but I don't see any option in the python library to do that.

Sensors do work with the realsense-viewer. I am running on Ubutnu 18.04 and desktop. I tried both the latest release in PyPI server and also from TOT of git repo.

Thanks

yosoufe, 2021-01-27:
What I understood so far, when I call `ctx.query_devices()` and save the result in a python variable, It cannot connect to the device at the `pipeline.start`. 

`ctx.query_devices()` runs the `usb_messenger = usb_device->open(0); // T265 only has one interface, 0` successfully in `tm-device.cpp`. Then when I call the `pipeline.start()` after `ctx.query_devices()`, the `usb_device->open(0);` is being called again and it is not successful and it throws exception. Do you know why?

This can give me hint for a workaround in python script. But I would call it a hack. I think the library should work fine without such a hack. So the following works for me now. But I have to be careful never hold a device object to the T265 device before running `pipeline.start()` in python. So the following works.

```.py
import sys
sys.path.insert(0, '/home/yousof/robotics/libs/librealsense/build_host/wrappers/python')

import pyrealsense2 as rs
import numpy as np 
import time 
print(rs.__file__)

import atexit

pipelines = []

@atexit.register
def cleanup():
    print('clean up')
    for p in pipelines:
        try:
            p.stop()
        except:
            pass

ctx = rs.context()

for dev in ctx.query_devices():
    print(f'device: {dev.get_info(rs.camera_info.name)}')
    p = rs.pipeline(ctx) 
    cfg = rs.config()
    dev_serial = dev.get_info(rs.camera_info.serial_number)
    del dev  ## <<<<------------------------------------ This line is my workaround
    cfg.enable_device(dev_serial)
    p.start(cfg)
    pipelines.append(p)

print('pipelines started')
s = time.time()
time.sleep(1)
```

So the key was to call `del dev` to delete the pointer to the T265 device before calling the `p.start(cfg)`

This should give you a hint on how to fix the bug. To me is like the device pointer somehow is keeping the USB device busy so the next operation cannot be done. Deleting the device object releases the USB device. Interestingly this is only happening for T265, Not the D435i.

egeozguroglu, 2024-08-01:
Did you get this working without the del dev?

closed 2026-10-23 by MartyG-RealSense