realsenseai/librealsense #8233: L515 not detected with C#

AgrinPiaz, 2021-01-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC |
| SDK Version                            |  2.41.0.0                          | 
| Language                            |  C#                          | 
| Segment			|  other                         | 

### Issue Description

In my application, I would use an L515. Actually I work with an SR305. The camera L515 is not detected by my program in C#. I have test with this short code, the SR305 is well detected but the L515 not.

                var ctx = new Context();
                var devices = ctx.QueryDevices();

                Console.WriteLine("There are {0} connected RealSense devices.", devices.Count);
                if (devices.Count == 0) return;
                var dev = devices[0];

                Console.WriteLine("\nUsing device 0, an {0}", dev.Info[CameraInfo.Name]);
                Console.WriteLine("    Serial number: {0}", dev.Info[CameraInfo.SerialNumber]);
                Console.WriteLine("    Firmware version: {0}", dev.Info[CameraInfo.FirmwareVersion]);

The device work well with realsense viewer and with c++ examples from the sdk.

closed 2021-01-25 by AgrinPiaz

manson, 2021-01-25:
Checked this with sdk 2.41 and L515 camera - everything works as should eather with USB2 or USB3

AgrinPiaz, 2021-01-26:
Thank you! The"realsense2.dll" file was an old sdk version.
Now it works well.