realsenseai/librealsense #8235: Cannot build Android SDK librealsense project

hunterman1, 2021-01-25:
While trying to run my librealsense project I'm receiving following error:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not find any matches for com.intel.realsense:librealsense:2.+ as no versions of com.intel.realsense:librealsense are available.
     Required by:
         project :app

Possible solution:
 - Declare repository providing the artifact, see the documentation at https://docs.gradle.org/current/userguide/declaring_repositories.html

Building the same project on saturday worked.
Are you doing some maintenance or any other activities? Or this is a problem on my site?

Regards,
Robert

MartyG-RealSense, 2021-01-26:
Hi @hunterman1  Does the advice in the link below solve your error, please?

https://programingerrors.wordpress.com/2020/07/07/could-not-determine-the-dependencies-of-task-appcompiledebugjavawithjavac-java-and-android-studio-error/

luckycreationsindia, 2021-02-08:
Hi,

Gradle files are missing from bintray repository. Cannot download jar manually too.

This path is also unavailable:
https://dl.bintray.com/intel-realsense/librealsense

`Repository path missing or not specified.`

MartyG-RealSense, 2021-02-08:
Hi @luckycreationsindia  The package location has been updated in the past week.

https://github.com/IntelRealSense/librealsense/issues/8287#issuecomment-770854785

https://github.com/IntelRealSense/librealsense/issues/8287#issuecomment-771179990

luckycreationsindia, 2021-02-08:
> Hi @luckycreationsindia  The packages that were at bintray have been updated in the past week.  Please read the comments at the link below for information about new package locations.
> 
> https://github.com/IntelRealSense/librealsense/issues/8287#issuecomment-770854785

Oh thanks... Didn't noticed that changes... It's actually breaking change and must be moved to master/main branch...

MartyG-RealSense, 2021-02-15:
Hi @hunterman1  Do you require further assistance with this case, please? Thanks!

@luckycreationsindia   The new version **2.42.0** of librealsense is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

luckycreationsindia, 2021-02-15:
> Hi @hunterman1 Do you require further assistance with this case, please? Thanks!
> 
> @luckycreationsindia The new version **2.42.0** of librealsense is now available.
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

This version is not working with Android 10. Just tested.

MartyG-RealSense, 2021-02-15:
@luckycreationsindia   Could you post a comment about your problem on https://github.com/IntelRealSense/librealsense/issues/8287 please?   Thanks!

luckycreationsindia, 2021-02-15:
Version 2.41 ad 2.42 both are ot working on Android (Java Wrapper). And thr maven doesn't include older versions. Please help me on this.

luckycreationsindia, 2021-02-15:
> @luckycreationsindia Could you post a comment about your problem on #8287 please? Thanks!

Sure

ev-mp, 2021-02-16:
@luckycreationsindia , with regards to the new Maven location please follow https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#android-releases---user-notification

MartyG-RealSense, 2021-02-22:
Hi @luckycreationsindia  Do you require further assistance with this case, please?   Thanks!

luckycreationsindia, 2021-02-22:
> Hi @luckycreationsindia  Do you require further assistance with this case, please?   Thanks!

Hi,

No thanks. Actually it might be firmware issue with Intel Camera as it's not working with Android 10. Now problem is I lost my camera and won't be able to test it.

ev-mp, 2021-02-22:
@luckycreationsindia hello,
Android10 has a known bug with permissions settings, so the SDK is supported only with vendors who has picked the proper security patches from the mainline and added it into their respective ROMs. #8287

The other alternatives are to either downgrade to Android 9 and lower, or to upgrade to Android11 where the issue has been apparently rectified.

luckycreationsindia, 2021-02-22:
> @luckycreationsindia hello,
> Android10 has a known bug with permissions settings, so the SDK is supported only with vendors who has picked the proper security patches from the mainline and added it [...]

Well that's not the issue. It was working perfectly fine with SDK version 2.39.x but in latest version it's not working.

ev-mp, 2021-02-23:
@hunterman1 hello, did you manage to apply the recommendations and resolve the build issue?

@luckycreationsindia ,  
>Well that's not the issue. It was working perfectly fine with SDK version 2.39.x but in latest version it's not working.

Please open a dedicated ticket for a follow up.

MartyG-RealSense, 2021-03-07:
Hi @hunterman1  Can you let me know whether you require further assistance with this case, please? Thanks!

closed 2021-03-14 by MartyG-RealSense