realsenseai/librealsense #8239: PyRealSense2, The dependence of the coordinates

RarDay, 2021-01-26:
Hi everyone and @MartyG-RealSense ))

Required Info |  
-- | --
Camera Model | D435i
Intel Model | Neural Compute Stick 2
Operating System & Version | Windows 10
Language | Python 3.7

I have many 3d point after `rs.save_to_ply("1.ply")` for example [0, 1.5, -1] and 2d coord camera [0, 0], [640, 480]. What is the dependence of these coordinates? How do I find all the points on the 3D plane that are in a square, for example from [150, 100] to [250, 250] px. The beginning of 3D [0, 0] coincides with the center of 2D [640 / 2, 480 / 2].
Maybe i must first find maximum and minimum values of the 3D plane. I took measurements and these numbers are always different, usually it is:
X | Y | Z
-- | -- | --
[-2.15, 1.3] | [-1.34, 0.5] | [0, **]

** - here is the depth, so it is clear why the parameter jumps

I drew a square on the 2D plane and adjusted the same on the 3D I got a rough understanding:
Point | 2D  | 3D 
-- | -- | --
1 | [290, 215] | 3D - [-0.093, -0.104]
2 | [345, 270] | 3D - [0.078, 0.125]

Z-not important

Ideally, the task is simple, take 68 points from the 2D plane and draw only them on 3D

[image]

P.S. Does Intel have ready-made solutions for getting 3D face points? Or eyes, mouth, nose
Or maybe even a ready-made solution for remembering a face and finding it on the frame

RarDay, 2021-01-26:
Looks like I found a solution

```
import pyrealsense2 as rs
import numpy as np
import cv2 as cv

pipeline = rs.pipeline()
pipe_profile = pipeline.start()
frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()

# Intrinsics & Extrinsics
depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)

# Depth scale - units of the values inside a depth frame, i.e how to convert the value to units of 1 meter
depth_sensor = pipe_profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()

# Map depth to color
depth_pixel = [240, 320]   # Random pixel
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_scale)
print(depth_point)
color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_point)
color_pixel = rs.rs2_project_point_to_pixel(color_intrin, color_point)

pipeline.stop()
```

MartyG-RealSense, 2021-01-26:
Hi @RarDay   I'm pleased that you were able to find a script solution.

In regard to recommendations for face landmark solutions, the discussion in the link below may be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/8095

Intel also have a ready-made facial authentication technology called **RealSense ID Solution F455** (which will also be available as a module called **F450**) that is shipping in the week of March 1 2021.  I am not able to answer questions about this product at this current time, though the official RealSense store link below provides links to official information resources such as its data sheet document.

https://store.intelrealsense.com/buy-intel-realsense-id-solution-f455-for-facial-authentication.html

closed 2021-01-29 by RarDay