realsenseai/librealsense #8241: Passing through depth data to matlab generated function

bladecell, 2021-01-26:
Hey, I am trying to work with realsense depth data to process them in matlab function. I want to have executable running on Jetson Nano that captures depth data and matlab generated C function will process the data. I want to use gpu coder for that. 

Since I cant do the whole executable in matlab and generate it, I thought I would use the capturing of the depth data in C, as shown in many exmaples, then generate function in matlab that would process generated data and put it in the project and generate the executable with CMake. 
My problem is that I dont konw how to pass the depth data to the matlab generated function. I was thinking, that I would get pointcloud from the camera, but I the GPU coder doesnt accept pointcloud as input argument, so maybe I would get a simple matix of depth data instead of pointcloud and use that as input argument, but I am not sure how to do that. 
Maybe I could use `auto depth = frames.get_depth_frame();` as an input argument, but I dont know, what to tell gpu coder the `depth` variable is. Thank for any suggestions.

MartyG-RealSense, 2021-01-26:
Hi @bladecell   I see that you have asked before In November 2020 about accessing RealSense depth data with Jetson and MATLAB using a GPU.

https://github.com/IntelRealSense/librealsense/issues/7762

The RealSense MATLAB wrapper for Windows 10 has a pointcloud example program called **pointcloud_example.m**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/pointcloud_example.m

Another discussion provided scripting for generating an RGB point cloud from a live D435 with the MATLAB wrapper.

https://community.intel.com/t5/Items-with-no-label/MATLAB-RGB-point-cloud/m-p/584717

*******************

If you would prefer to perform the point cloud calculation directly in MATLAB, on the link below there is a MATLAB function that can be downloaded that converts a depth image to a point cloud in MATLAB.  

https://rgbd-dataset.cs.washington.edu/software.html

Scroll down the page to the heading titled 'Depth Image to Point Cloud (MATLAB)' and click on the **depthToCloud.m** link to launch its download in your browser.  I tested that the download still works.

bladecell, 2021-01-26:
Hey, thank for your response.
I am not really sure, if I need exactly poincloud. I need the matlab script to create countour lines from the depth data. The pointcloud_example.m uses librealsense, which the GPU conder doesnt work with. I dont even think I can use the MATLAB weapper for the same reason. I guess I need the realsense library only "run" in the c code, which captures depth, then the matlab genereted function do it thing and then show the result. I was trying to build on [this](https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp) and the use `depth` variable to serve as an input for the matlab function. I just dont whether the `depth` variable is a simple matrix or librealsense specific variable, in that case I would need to change to to some standard c variable that matlab coder can work with.

MartyG-RealSense, 2021-01-27:
The link below suggests resources for using the in-built features of MATLAB instead of the RealSense MATLAB wrapper to generate 3D world points for a 2D image.

https://github.com/IntelRealSense/librealsense/issues/6665#issuecomment-647406108

There was also a past case where a RealSense user used OpenCV code to write pixel data to an xml file and then used the xml to plot a contour graph in MATLAB.  The OpenCV code is here:

```
cv::FileStorage fs("data_capture.xml", cv::FileStorage::WRITE);
fs << "depth" << ir ;
```

bladecell, 2021-01-27:
I am sorry, I think I may have misled you. I just want to use matlab to generate c code with GPU coder. I think, I need 3d array from rs2::frame depth and use the array as an argument for the c generated function. Thanks again for your help.

MartyG-RealSense, 2021-01-27:
No problem at all.  Does the RealSense MATLAB case in the link below resemble what you are aiming to achieve please in regard to saving RealSense stream data into a 3D XYZ axis?

https://github.com/IntelRealSense/librealsense/issues/2576#issuecomment-432271553

bladecell, 2021-01-27:
Yeah exactly that, get the XYZ data in array, but in C++.

MartyG-RealSense, 2021-01-28:
The C++ script in the link below takes the approach of writng depth and pointcloud into a text file that can be read by MATLAB.

https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380487211

bladecell, 2021-01-28:
I came up with this:

```
    double points[230400]; 

    for (int y = 0; y < 359; y++) {
        for (int x = 0; x < 640; x++) {
            points[y*640 + x] = depth.get_distance(x,y);     
    }
    }
```
but I heard, that `depth.get_distance()` shouldnt be used in loop, so I will have a look at that.

bladecell, 2021-01-29:
Thank you for your help, I think I have what I need. From [this](https://gist.github.com/MRo47/aecd9be1b58455f29b5c1544703589ae) script I found i the thread you linked, I got this:

```
    double pixel_distance[230400]; 

    const uint16_t* p_depth_frame = reinterpret_cast<const uint16_t*>(depth.get_data());

     for (int y = 0; y < height; y++)
    {
        auto depth_pixel_index = y * width;
        for (int x = 0; x < width; x++, ++depth_pixel_index)
        {
            pixel_distance[depth_pixel_index] = depth_scale * p_depth_frame[depth_pixel_index];
        }
    }
```

MartyG-RealSense, 2021-01-29:
You are very welcome @bladecell - I am pleased that you found a solution!

closed 2021-02-05 by MartyG-RealSense