realsenseai/librealsense #8242: Couldn't resolve requests

oscarthorn, 2021-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.10.00) | 
| Operating System & Version |   ARM64 Linux  | 
| Platform                                 | Jetson  |
| SDK Version                            |  Latest  | 
| Language                            |  {c++}                          | 

### Issue Description
Whenever I try to use the camera I get the following error:
`terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
`

This is a minimal version of my code:

```
    rs2::align align_to(RS2_STREAM_COLOR);

    rs2::pipeline pipe;

    rs2::config cfg;

    cfg.enable_stream(RS2_STREAM_COLOR, 1270, 720, RS2_FORMAT_RGB8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 1270, 720, RS2_FORMAT_Z16, 30);

    pipe.start(cfg);

    while (true) {
        // CTRL-C check
        if (!nh.ok()) {
            break;
        }

        rs2::frameset data = pipe.wait_for_frames();

        // Make sure the frameset is spatialy aligned
        // (each pixel in depth image corresponds to the same pixel in the color image)
        rs2::frameset aligned_set = align_to.process(data);

        // Try to get a frame of a depth image
        rs2::frame depth = aligned_set.get_depth_frame();
        rs2::frame rgb = aligned_set.get_color_frame();

       // Do stuff

    }

    return 0;
```

It works perfectly fine in the realsense-viewer using these settings.

MartyG-RealSense, 2021-01-26:
Hi @oscarthorn  The warning **Couldn't resolve requests** refers to the SDK not being able to carry out an instruction, either because it does not understand it because of a typing error or because the requested mode is not currently supported by the camera (e.g because it is operating in the more limited USB 2 mode instead of the full range of capabilities available on a USB 3 connection).

There is a typing error in your cfg stream definition instructions.  You have defined a resolution of 1270x720.  The supported resolution is **1280x720**

oscarthorn, 2021-01-27:
Thanks for spotting that! Silly mistake!

closed 2021-01-27 by oscarthorn