realsenseai/librealsense #8243: Exposure time inconsistencies

sam598, 2021-01-26:
I have a question about the inconsistencies in the way that the RGB and Depth/IR cameras of the D400 cameras handle exposure time.

The exposure time of the depth/IR cameras of the D435 and D455 can be very precisely set in microseconds.

The exposure time of the RGB camera appears to be set in "hundreds of microseconds" which is a strange format. Additionally instead of using the requested exposure time, it will be automatically clamped to a nearest value in an unknown list of exposure values. This can be easily tested in the RealSense Viewer.

This is understandable on a camera like the D435 where the RGB and IR cameras are different hardware. I assume that the RGB camera has some limitation where it needs to be set at a specific value, and maybe that is where the "hundreds of microseconds" comes from.

However on the D455 the RGB camera is apparently the exact same camera as the IR cameras. If the IR cameras can be precisely set in microseconds, why can't the RGB camera? Is this an unnecessary holdover in the SDK from the previous cameras?

Is this something that can be fixed in software/firmware? Or is it a hardware limitation?

ev-mp, 2021-01-27:
@sam598 hello,
The control's granularity in FW is 100usec with effective range of [1-10000], so the bottleneck is definitely is in the host.
You haven't mentioned the OS used, but in Win10 the RGB sensor's exposure control is implemented using logarithmic scale logarithmic scale. If that's the case then unfortunately there is no way to override it.
Linux OS doesn't have this limitation and the user can select an arbitrary exposure value within the full range.

Possible duplicate of #5493

sam598, 2021-01-27:
Fascinating. So on Linux if you set the exposure to 4980usec the firmware will round up to 5000usec?

ev-mp, 2021-01-27:
Almost, it is just you can't set 4980 on Linux as the units are 100usec. 
You can specify integer value of 49 or 50 (4900 or 5000 usec). So there is no need for round up/down by design.

sam598, 2021-01-27:
Are the infrared/depth cameras limited to 100usec as well, or will the sensor have an exposure with 1usec precision?

ev-mp, 2021-01-27:
@sam598 , to emphasize the info provided in the above discussion applies to the **RGB sensor** only. 
IR/Depth stream exposure control can be set with 1usec granularity, uses a different range of [1-165000]usec and has no limitations on neither OS.

sam598, 2021-01-27:
Thank you for the clarifications. Is this information available somewhere in the documentation or data sheets?

closed 2021-01-27 by sam598