realsenseai/librealsense #8244: change minimun distance settings from the python script

agenis, 2021-01-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0.2482                           | 
| Language                            |  python                           | 
| Segment			|  others                          | 

### Issue Description
Hello, i'm working with L515 and python pyrealsense2. My question is related to tuning the camera to see the closest object possible. In the specs it's said around 25cm, but when i change some tunings in the viewer (_short range_ mode, _min distance_ slider, etc.), we get to see objects as close as ~5cm on the 3d viewer. It seems to me that I have to change the setup manually in the viewer, and then the camera saves it when run from python...  But I coulnd't find any example in python on how to do this directly from the code. Are there any examples of such code?
Thanks

gusrbgks, 2021-01-29:
I'm also looking for this issue.
In Viewer, you can see the depth image close by adjusting the Min Distance.
However, it cannot be checked when executed in Python code. 
I would like to know the code to change this.

agenis, 2021-02-15:
I used that code that seems to work:
```
        import pyrealsense2 as rs
        pipeline = rs.pipeline()
        profile = pipeline.get_active_profile()
        profile.get_device().query_sensors()[0]  # 0 for depth sensor, 1 for camera sensor
        sensor.set_option(rs.option.min_distance, 0)
        sensor.set_option(rs.option.enable_max_usable_range, 0)
```
This, and maybe for a real close measurement, one might want as well to reduce the laser power below 50% (`sensor.set_option(rs.option.laser_power, 80)`) and keep receiver_gain rather high value (18 for me).

This reproduces some of the differences between the default mode and the close range mode.

closed 2021-02-15 by agenis

DesmondZuo, 2021-06-27:
Thanks agenis! That actually worked for me!

just one typo, instead of:

`profile.get_device().query_sensors()[0]  # 0 for depth sensor, 1 for camera sensor`

should be 

`sensor = pipeline_profile.get_device().query_sensors()[0]`

again, thanks a lot!