realsenseai/librealsense #8246: adjust odometry position / implement JSON file (T265)

Dronikos, 2021-01-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265  | 
| Operating System & Version |  Linux (Ubuntu 18)  |                                       | 
| Platform                                 |  NVIDIA Jetson  |
| Segment			|  Robot                          | 
|additional info | I am using ROS for that project |

Hello everyone,

I am new to all of this so please excuse me if I am posting this in the wrong space or ask a trivial question.
I am currently working on a robot which uses the T265 for odometry purposes. However, the T265 is in front of base_link so I want to transform the odometry so that it represents the odometry in base_link.
During my research i found this [https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md](url) in which the author describes how to do exactly that using a JSON-File to configure the T265. The Problem is that I have no idea how to feed T265 this file or what to do with it. If it has something to do with the API please explain what to do as well.

Every help is appreciated

seanxu112, 2021-03-07:
Haven't done this myself, but there is an arg in the rs_t265.launch called "json_file_path" which I am assuming to be where you should feed this file.

RealSenseSupport, 2021-03-08:
There is a python example located here:  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/t265_wheel_odometry

closed 2026-10-23 by MartyG-RealSense