realsenseai/librealsense #8248: Callbacks can happen after rs2::sensor::stop()

thomas-slamcore, 2021-01-27:
| Field | |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.8.200 | 
| Operating System | Linux | 
| Kernel Version |          5.10                      | 
| Platform                                 | PC |
| SDK Version                            |  2.39.0                    | 
| Language                            |  C++                     | 

### Issue Description

I recently ran into a use-after-free that was caused by a sensor callback happening after `rs2::sensor::stop()` was called. Having looked at the code (this is the Linux V4L2 backend) it doesn't look like there's anything there to prevent a callback from happening after `uvc_sensor::stop()` since it ends up just setting an atomic flag in `v4l2_uvc_device::stop_callbacks()` which doesn't wait for the `v4l2_uvc_device::poll()` function to return. I feel like this behaviour is unexpected and unsafe.

RealSenseCustomerSupport, 2021-03-17:
----------------------------------------------

Hi thomas-slamcore
Thanks for sharing your finding
When the use after free occurs, will it cause your application (use Librelasense API) crash?
or has issue such as image freeze or stuck for a while?

thomas-slamcore, 2021-03-17:
Hi there,

Honestly I'm not 100% sure anymore whether it crashed that one time, obviously this is rather hard to reproduce. Since then we've implemented our own workaround (we basically ignore callbacks based on an atomic flag), but this is something that should be fixed within librealsense. Your documentation for `rs2::sensor::stop()` is a bit sparse, but it doesn't make any sense that callbacks should be triggered after `stop()` has returned.

Here's a screenshot from ASAN which helped me diagnose this issue.

[image]