realsenseai/librealsense #8249: L515 extrinsic parameters IMU - depth

Amber8606, 2021-01-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |                          | 
| Language                            |  C++                          | 
| Segment			|  Robot        | 

### Issue Description
Hi, 

I am using a L515 but I am not sure the IMU data I gather are already multiplied by the extrinsic matrix that allows for the alignment between depth sensor reference system and IMU reference system. 

 Does anyone know it?

I have found some info about that but for other devices (D435i, T265) here: 

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

"The IMU sensor location and orientation relative to the depth sensors are conveniently embedded into the sensor’s extrinsic data. In order to use the depth sensor as CS origin, each IMU sample is multiplied internally by the extrinsic matrix. The resulted orientation angles and acceleration vectors share the coordinate system with the depth sensor."

Thanks a lot

ev-mp, 2021-01-27:
@Amber8606 hello, the coordinate system of IMU and Depth sensors are __aligned__ in terms of Coordinate System axes orientation (i.e. X,Y and Z point in the same directions) but they don't share the same origin.
When extracted, the IMU data is not multiplied by the extrinsic matrix but provided as-is.

alvinsunyixiao, 2021-02-05:
What coordinate system is used by the Depth camera? It seems like the Y axis of ACCEL reading is pointing upwards instead of downwards. (I get negative Y axis reading when standing in an upright position)

ev-mp, 2021-02-05:
The SDK uses RH Coordinate System - [Documentation](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates).
The additional documentation taken from D435i applies to L515 as well - https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

RealSenseSupport, 2021-03-03:
Is there anything else needed on this thread?  Does the information provided help.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-03-23 by RealSenseSupport