realsenseai/librealsense #8250: Installing from source MacOS on M1 chip

HansBambel, 2021-01-27:
I am trying to use the D415 stereocamera via pyrealsense2, but I am struggling with installing the necessary libraries.
Since I am using not using an intel chip I found out that I need to install from source. I am following this [tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md#building-from-source), but I also need to follow this one for the [Python-wrapper](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation), right?
In point 3 in the latter link it says

> Run the top level CMake command with the following additional flag -DBUILD_PYTHON_BINDINGS:bool=true:
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
Note: To force compilation with a specific version on a system with both Python 2 and Python 3 installed, add the following flag to CMake command: -DPYTHON_EXECUTABLE=[full path to the exact python executable]

What is meant by the top level cmake command? Does this mean that I need to add this to the command that is given in the first link:
```
mkdir build && cd build
sudo xcode-select --reset
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
```
so that the whole command will look like this:
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/Users/kevin/opt/anaconda3/envs/p37/bin/python`

If so, do I execute this command from the `librealsense/build`-folder or the `librealsense/build/wrappers/python`-folder?

I tried almost all combinations, but I receive the following error in the `librealsense/build/CMakeFiles/CmakeError.log`:
[CMakeError.log](https://github.com/IntelRealSense/librealsense/files/5880183/CMakeError.log)

The first error tells me that he can not find -lc++ and I don't know how do install that. Is it from a very old xcode version?

Any help would be greatly appreciated!

MartyG-RealSense, 2021-01-27:
Hi @HansBambel   We are aware of a couple of cases where difficulty has been encountered with installing librealsense on the new **Apple Silicon M1** Mac models with Arm-based M1 processors instead of Intel processors that were launched in late 2020.  Intel are looking into it.  I have no update that I can provide at the time of writing this though.

https://github.com/IntelRealSense/librealsense/issues/8090 

The instructions regarding what constitutes the **top level** when installing the Python wrapper can be a little unclear.  In general though, I see the CMake command initiated from the **build** folder, like you have done.  This is also the method used in a CMake installation guide for the Python wrapper on MacOS that has been successful for some Mac users (though the guide dates to before the introduction of M1 processors).

https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-555830996

Another RealSense user's version of a guide for a successful MacOS Python wrapper installation is here:

https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-643304216

If you encounter an error referencing **target.h** when installing the Python wrapper on MacOS, the advice in the link below may also be helpful.

https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-690595112

Ultimately though, there are currently no guarantees that I can provide about installing on M1.

HansBambel, 2021-01-27:
Thank you for your answer. Sadly, it did not help me directly, but I managed to make work around the `Could NOT find Threads (missing: Threads_FOUND)` error by adding
```
set(CMAKE_THREAD_LIBS_INIT "-lpthread")
set(CMAKE_HAVE_THREADS_LIBRARY 1)
set(CMAKE_USE_WIN32_THREADS_INIT 0)
set(CMAKE_USE_PTHREADS_INIT 1)
set(THREADS_PREFER_PTHREAD_FLAG ON)
```
which I found [here](https://github.com/alicevision/geogram/issues/2#issuecomment-536835320).
Then CMake was able to run through correctly.
Sadly, then I run into the same issue as #8090 :(

MartyG-RealSense, 2021-01-27:
I am sorry that you were not able to make progress with librealsense installation due to the x86 error aparently produced by the Arm-based M1 processor.  I do not currently have further information to offer about using RealSense with M1, unfortunately.

HansBambel, 2021-01-27:
Not your fault! I subscribed to the other issue and wait for a reply from intel then. In the meantime I have the luxury of using an intel-based MacBook as well.

MartyG-RealSense, 2021-01-27:
Thanks very much for your understanding!   I'm pleased to hear that you have an alternative Intel-based Mac available.

jackjansen, 2021-01-29:
@MartyG-RealSense I fully share the understanding by others in this thread that supporting the new M1 architecture is not to be taken for granted:-) but still: it would be nice if we could get some indication of whether Intel has any plans for the foreseeable future, and/or whether the community can help.

Maybe it would be good to create a specific issue "Port librealsense2 to Apple M1", where we could follow easily?

MartyG-RealSense, 2021-01-29:
As https://github.com/IntelRealSense/librealsense/issues/8090 is where I direct RealSense users who are having this issue with M1 Macs, I have placed an **Enhancement** tag on that case as a reminder to keep it open.  It can then be used as a focal point for posting comments about this issue from all affected users, fulfilling the same function as your own idea to create a specific issue.

At the moment, the situation status remains that Intel are looking into it.

MartyG-RealSense, 2021-02-11:
Case closed due to https://github.com/IntelRealSense/librealsense/issues/8090 being kept open as the central location to post Mac M1 comments.

closed 2021-02-11 by MartyG-RealSense

MartyG-RealSense, 2021-04-25:
Hi @HansBambel and @jackjansen   A RealSense user shared the method that worked for them for building librealsense on their M1 Mac.

https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-826164584

HansBambel, 2021-04-25:
Thanks for the info! I might be able to try it out again on the M1 as well

jalapatip, 2021-06-29:
Hello @adamnovak and @HansBambel, I see that there are multiple CMakeLists.txt files in the package hierarchy. Which file do I make the changes to?

HansBambel, 2021-06-29:
That is a good question. I did not try this solution yet. Dynamized from the comment https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-826164584 should know.

jalapatip, 2021-06-29:
> That is a good question. I did not try this solution yet. Dynamized from the comment [#8090 (comment)](https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-826164584) should know.

Thanks for the update! I will check with @Dynamized.

adamnovak, 2021-07-12:
@jalapatip You mean to work around the missing `Threads` CMake package? I think you need to make the changes to set the variables in all `CMakeLists.txt` files that say `find_package(Threads REQUIRED)` in them. [This search](https://github.com/IntelRealSense/librealsense/search?l=CMake&q=Threads+Required) ought to find all of them in this project.

zarathustr, 2021-08-10:
cmake .. -DBUILD_WITH_OPENMP=OFF -DENABLE_CCACHE=OFF -DCMAKE_HAVE_THREADS_LIBRARY=1 -DCMAKE_THREAD_LIBS_INIT="-lpthread" -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl

MartyG-RealSense, 2021-08-10:
Thanks very much @zarathustr for sharing your knowledge . :)

gschian0, 2021-09-02:
I'm stuck at this same spot. after running both cmake comands.. When I run those commands  and try to run either `make -j2` or `make install` get this `make: *** No rule to make target `install'.  Stop.`

gschian0, 2021-09-02:
on cmake command I get this response ... `CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   realsense2-gl

This warning is for project developers.  Use -Wno-dev to suppress it.`

gschian0, 2021-09-06:
ok... I'm making progress... i'm finding out now there is the cpuid.h file that is preventing the Xcode build.  I'm not sure how to change it.
`#if !(__x86_64__ || __i386__)
#error this header is for x86 only
#endif`. and an error that says "This header is for x86 only"

gschian0, 2021-09-06:
**in color-formats-converter.cpp this is causing the error** 
`#include <cpuid.h>
void cpuid(int info[4], int info_type) {
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
}
#endif`

gschian0, 2021-09-06:
**this looks like the culprit!  I'm not sure how to fix it though...** 

`//#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__))
//
//bool has_avx() { return false; }
//
//#else
//
//#ifdef _WIN32
//#include <intrin.h>
//#define cpuid(info, x)    __cpuidex(info, x, 0)
//#else
//#include <cpuid.h>
//void cpuid(int info[4], int info_type) {
//    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
//}
////#endif
//
//bool has_avx()
//{
//    int info[4];
//    cpuid(info, 0);
//    cpuid(info, 0x80000000);
//    return (info[2] & ((int)1 << 28)) != 0;
//} 
//
//#endif
`
**I'm  not 100% sure on how to edit the cpuid.h file but in a search I did find a file that seems to grab an arm64 cpu id.**

`/*
 * Copyright (c) 2017 Apple Inc. All rights reserved.
 *
 * @APPLE_OSREFERENCE_LICENSE_HEADER_START@
 *
 * This file contains Original Code and/or Modifications of Original Code
 * as defined in and that are subject to the Apple Public Source License
 * Version 2.0 (the 'License'). You may not use this file except in
 * compliance with the License. The rights granted to you under the License
 * may not be used to create, or enable the creation or redistribution of,
 * unlawful or unlicensed copies of an Apple operating system, or to
 * circumvent, violate, or enable the circumvention or violation of, any
 * terms of an Apple operating system software license agreement.
 *
 * Please obtain a copy of the License at
 * http://www.opensource.apple.com/apsl/ and read it before using this file.
 *
 * The Original Code and all software distributed under the License are
 * distributed on an 'AS IS' basis, WITHOUT WARRANTY OF ANY KIND, EITHER
 * EXPRESS OR IMPLIED, AND APPLE HEREBY DISCLAIMS ALL SUCH WARRANTIES,
 * INCLUDING WITHOUT LIMITATION, ANY WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE, QUIET ENJOYMENT OR NON-INFRINGEMENT.
 * Please see the License for the specific language governing rights and
 * limitations under the License.
 *
 * @APPLE_OSREFERENCE_LICENSE_HEADER_END@
 */
#ifndef _ARM64_MACHINE_CPUID_H_
#define _ARM64_MACHINE_CPUID_H_

typedef struct {
	uint64_t        el0_not_implemented             : 1,
	    el0_aarch64_only                : 1,
	    el0_aarch32_and_64              : 1,
	    el1_not_implemented             : 1,
	    el1_aarch64_only                : 1,
	    el1_aarch32_and_64              : 1,
	    el2_not_implemented             : 1,
	    el2_aarch64_only                : 1,
	    el2_aarch32_and_64              : 1,
	    el3_not_implemented             : 1,
	    el3_aarch64_only                : 1,
	    el3_aarch32_and_64              : 1,
	    reserved                                : 52;
} arm_feature_bits_t;

/* Debug identification */

/* ID_AA64DFR0_EL1 */
typedef union {
	struct {
		uint64_t debug_arch_version             : 4,
		    trace_extn_version             : 4,
		    perf_extn_version              : 4,
		    brps                                   : 4,
		    reserved0                              : 4,
		    wrps                                   : 4,
		    reserved1                              : 4,
		    ctx_cmps                               : 4,
		    reserved32                             : 32;
	} debug_feature;
	uint64_t value;
} arm_cpuid_id_aa64dfr0_el1;

typedef struct {
	uint32_t        num_watchpoint_pairs;
	uint32_t        num_breakpoint_pairs;
} arm_debug_info_t;

#endif /* _MACHINE_CPUID_H_ */
`

gschian0, 2021-09-06:
these are the changes I had to make to **my global_config.cmake** file to fix the linking errors

`# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)
set(CMAKE_CXX_STANDARD 11)
# View the makefile commands during build
#set(CMAKE_VERBOSE_MAKEFILE on)
set(CMAKE_THREAD_LIBS_INIT "-lpthread")
set(CMAKE_HAVE_THREADS_LIBRARY 1)
set(CMAKE_USE_WIN32_THREADS_INIT 0)
set(CMAKE_USE_PTHREADS_INIT 1)
set(THREADS_PREFER_PTHREAD_FLAG ON)
include(GNUInstallDirs)
# include librealsense helper macros
include(CMake/lrs_macros.cmake)
include(CMake/version_config.cmake)

if(ENABLE_CCACHE)
  find_program(CCACHE_FOUND ccache)
  if(CCACHE_FOUND)
      set_property(GLOBAL PROPERTY RULE_LAUNCH_COMPILE ccache)
      set_property(GLOBAL PROPERTY RULE_LAUNCH_LINK ccache)
  endif(CCACHE_FOUND)
endif()

macro(global_set_flags)
    set(LRS_TARGET realsense2)
    set(LRS_LIB_NAME ${LRS_TARGET})

    add_definitions(-DELPP_THREAD_SAFE)
    add_definitions(-DELPP_NO_DEFAULT_LOG_FILE)

    if (BUILD_GLSL_EXTENSIONS)
        set(LRS_GL_TARGET realsense2-gl)
        set(LRS_GL_LIB_NAME ${LRS_GL_TARGET})
    endif()

    if (ENABLE_ZERO_COPY)
        add_definitions(-DZERO_COPY)
    endif()

    if (BUILD_EASYLOGGINGPP)
        add_definitions(-DBUILD_EASYLOGGINGPP)
    endif()

    if (ENABLE_EASYLOGGINGPP_ASYNC)
        add_definitions(-DEASYLOGGINGPP_ASYNC)
    endif()

    if(TRACE_API)
        add_definitions(-DTRACE_API)
    endif()

    if(HWM_OVER_XU)
        add_definitions(-DHWM_OVER_XU)
    endif()

    if(COM_MULTITHREADED)
        add_definitions(-DCOM_MULTITHREADED)
    endif()

    if (ENFORCE_METADATA)
      add_definitions(-DENFORCE_METADATA)
    endif()

    if (BUILD_WITH_CUDA)
        add_definitions(-DRS2_USE_CUDA)
    endif()

    if (BUILD_SHARED_LIBS)
        add_definitions(-DBUILD_SHARED_LIBS)
    endif()

    if (BUILD_INTERNAL_UNIT_TESTS)
        add_definitions(-DBUILD_INTERNAL_UNIT_TESTS)
    endif()

    if (BUILD_WITH_CUDA)
        include(CMake/cuda_config.cmake)
    endif()

    if(BUILD_PYTHON_BINDINGS)
        include(libusb_config)
    endif()

    if(BUILD_NETWORK_DEVICE)
        add_definitions(-DNET_DEVICE)
        set(LRS_NET_TARGET realsense2-net)
    endif()
    
    if(CHECK_FOR_UPDATES)
        if (ANDROID_NDK_TOOLCHAIN_INCLUDED)
            message(STATUS "Android build do not support CHECK_FOR_UPDATES flag, turning it off..")
            set(CHECK_FOR_UPDATES false)
        elseif (NOT BUILD_GRAPHICAL_EXAMPLES)
            message(STATUS "CHECK_FOR_UPDATES depends on BUILD_GRAPHICAL_EXAMPLES flag, turning it off..")
            set(CHECK_FOR_UPDATES false)
        else()
            include(CMake/external_libcurl.cmake)
            add_definitions(-DCHECK_FOR_UPDATES)
        endif()
    endif()
    
    add_definitions(-D${BACKEND} -DUNICODE)
endmacro()

macro(global_target_config)
    target_link_libraries(${LRS_TARGET} PRIVATE realsense-file ${CMAKE_THREAD_LIBS_INIT})

    set_target_properties (${LRS_TARGET} PROPERTIES FOLDER Library)

    target_include_directories(${LRS_TARGET}
        PRIVATE
            src
            ${ROSBAG_HEADER_DIRS}
            ${BOOST_INCLUDE_PATH}
            ${LZ4_INCLUDE_PATH}
            ${LIBUSB_LOCAL_INCLUDE_PATH}
        PUBLIC
            $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
            $<INSTALL_INTERFACE:include>
            PRIVATE ${USB_INCLUDE_DIRS}
    )

endmacro()

macro(add_tm2)
    message(STATUS "Building with TM2")
    include(libusb_config)
    target_link_libraries(${LRS_TARGET} PRIVATE usb)
    if(USE_EXTERNAL_USB)
        add_dependencies(${LRS_TARGET} libusb)
    endif()
    target_compile_definitions(${LRS_TARGET} PRIVATE WITH_TRACKING=1)
endmacro()`

MartyG-RealSense, 2022-04-24:
Hi everyone, I just wanted to highlight that the **brew install librealsense** method of installing librealsense on Mac now lists support for MacOS Monterey and for Apple Silicon M1.

https://formulae.brew.sh/formula/librealsense