realsenseai/librealsense #8255: need help with pointcloud creation and visualization given only the depth data

monacv, 2021-01-28:
I saw https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb however it also makes use of RGB frames.

If I only have the depth data and color_intrinsics, how can I create a pointcloud using RealSense API and how can I visualize it?

I also saw this method but I am unsure how I could visualize it: 
`pointcloud = rs.convert_depth_frame_to_pointcloud(original_depth_frame, color_intrinsics)`
where:
```
def set_intrinsics(intrinsics_dict):
    intrinsics = rs.intrinsics()
    intrinsics.width = intrinsics_dict['width']
    intrinsics.height = intrinsics_dict['height']
    intrinsics.ppx = intrinsics_dict['ppx']
    intrinsics.ppy = intrinsics_dict['ppy']
    intrinsics.fx = intrinsics_dict['fx']
    intrinsics.fy = intrinsics_dict['fy']
    intrinsics.model = intrinsics_dict['model']
    intrinsics.coeffs = intrinsics_dict['coeffs']
    return intrinsics

color_intrinsics = set_intrinsics(camera['color_intrinsics'])
```

MartyG-RealSense, 2021-01-28:
Hi @monacv   The link below contains a script that performs something similar with depth and color_intrinsics to what I believe you are describing.  The code is in C++, though it may give you some insights for structuring a Python equivalent if Python is the language that you are using.

https://github.com/IntelRealSense/librealsense/issues/5482

monacv, 2021-01-28:
Hi @MartyG-RealSense, thanks a lot for your reply.

I'd be interested to learn how you did this? Do you have the code snippet for reproducing it for Intel RealSense camera?

I have created a version using O3D for now but would like to learn how to use this other version too.
https://github.com/intel-isl/Open3D/issues/2959

MartyG-RealSense, 2021-01-29:
The image I provided above was produced in the RealSense Viewer, but you can control depth colorization with code too so that Python point clouds look more like the ones produced by the Viewer.  

A range of resources for configuring depth colorization in Python are provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

closed 2021-02-13 by MartyG-RealSense