realsenseai/librealsense #8259: Access color and depth frame using frame number

akmaldeen, 2021-01-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {c++ }                          | 

### Issue Description
Hi I'm new to realsense. I have D455 camera. I'm using c++ in windows, visual studio 2009.

I have to access a selected frame in .bag file using frame number. Let's say 50th frame is my requirement, then my code has to access 50th frame and convert it to opencv Mat file.  Also I have to access the color frame depth frame separately.

Please help me with c++ example

MartyG-RealSense, 2021-01-28:
Hi @akmaldeen  It may be best to break this programming goal into two sections.   First, get conversion to cv::mat of all frames working.  And then add a frame index seek mechanism for specific frames once conversion for all frames is working.

In regard to the first step, a C++ bag extraction script that uses CV::mat can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/1919#issuecomment-403777588

For the second step (navigating to a specific frame index) with C++, the link below should be helpful:

https://github.com/IntelRealSense/librealsense/issues/6577

MartyG-RealSense, 2021-02-05:
Hi @akmaldeen  Do you require further assistance with this case, please?   Thanks!

akmaldeen, 2021-02-05:
I have used disable playback technique to poll frame one by one from bag file

 rs2::playback play_back = device.as<rs2::playback>();
    play_back.set_real_time(false);

after I can access the particular frame using frame number and basic loop techniques.

Access frame numeber:
rs2:frameset frames = pipe.wait_for_frames();
        auto color_frame = frames.get_color_frame();
        auto fr_no = color_frame.get_frame_number();

Thanks for the links

closed 2021-02-05 by akmaldeen

MartyG-RealSense, 2021-02-05:
Great to hear that you developed a solution @akmaldeen  - thanks very much for the update!