realsenseai/librealsense #8260: Values in CSV file after using rs-convert

bineethk, 2021-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L5I5 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10  |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { 02.0> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am new to Intel RS. I converted my rosbag file into a CSV file using rs-convert. I know the CSV file is depth matrix.
But when I opened the file, I could see various values. Just want to know what actually each of those value represents?. Can you say a scenario where we can use these values from CSV files?

RealSenseSupport, 2021-02-26:
Hi @bineethk 

Thanks and welcome to using RealSense product.  Just to verify, are you using a L515 camera?  And what specifically brought you to this device?  What usage model or use case that you are trying to tackle with the L515 camera?

We have a ton of information on our site regarding using the L515 camera, here is a great place to start:  https://www.intelrealsense.com/category/lidar/

bineethk, 2021-03-02:
Hi. Thank you for the welcome. I am curious to explore more on the product.
Yes. I am using Intel L515 LiDar camera.
I am interested in the LiDar sensor for object detection and distance computation from the depth maps. And also its smallest size is somewhat comfortable for my use case.
My use-case involves a visual object recognition model along with distance computation for autonomous navigation purposes.

RealSenseSupport, 2021-03-23:
Thanks for outlining your usage model.

Have you been able to review the information we have outlined in our documentation site?

Also what you will find in the csv file is depth data for all pixels captured in a frame.  The convert to CSV will provide a single csv file for every frame captured in the bag file.

closed 2021-04-08 by RealSenseSupport