realsenseai/librealsense #8263: ASAN points out some memory leaks

thomas-slamcore, 2021-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  5.12.8.200| 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  5.10                               | 
| Platform                                 | PC  |
| SDK Version                            |  2.41/master                   | 
| Language                            |  C++                   | 

### Issue Description

I've been running address sanitizer (ASAN) on our software which led me to some fairly minor memory leaks in librealsense. Here is the [output from ASAN.](https://github.com/IntelRealSense/librealsense/files/5888268/asan.log) I believe I have fixed these particular issues and will be opening a pull-request against this ticket.

MartyG-RealSense, 2021-01-28:
Thanks very much @thomas-slamcore for the pull request.  I will mark this case with the **Enhancement** tag to remind to keep it open whilst your PR is active.

thomas-slamcore, 2021-02-01:
Fixed by 3715e57

closed 2021-02-01 by thomas-slamcore

MartyG-RealSense, 2021-02-01:
Thanks very much for the update!

AndreV84, 2021-05-13:
@thomas-slamcore 
can we use the slamcore on jetson nx, agx? folks are reporting issues with detecting of a camera with slamcore vizualizer