realsenseai/librealsense #8265: realsense-viewer has a problem with pixel formatting

takijo0116, 2021-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435} | 
| Firmware Version                   | (Open RealSense Viewer --> 05.12.10.00) | 
| Operating System & Version |  Linux (Ubuntu 20.04) /  | 
| Kernel Version (Linux Only)    |  (5.8.0-40-generic) | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.41.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/ }                          | 

### Issue Description
I installed realsense viewer as per the tutorial.
(https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)

But, I command 

    realsense-viewer

I got the warning

    WARNING [140278774392576] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!

How I can solve this warning?

MartyG-RealSense, 2021-01-28:
Hi @takijo0116   I would begin by clarifying that the installation process installs the **librealsense SDK** software for use with RealSense cameras.  The **RealSense Viewer** is a program that is part of the librealsense SDK that enables easy use of the camera without requiring any programming knowledge.

The RealSense SDK currently officially supports kernels up to **5.4**.  Results from using more recent kernels such as the 5.8 version that you are using cannot be guaranteed.

Also , which type of computer are you installing librealsense on, please (PC, Mac, Raspberry Pi, Nvidia Jetson, etc).  If you are installing on a computer / computing device with an Arm processor such as Pi or Jetson then the dkms package installation method in the link above would be incompatible with such machines, as these packages are designed for computers with x86 / x64 processors.

If you are installing for Ubuntu 20.04 on a PC computer then you can use an installation method called **RSUSB backend**.  This method requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

If you are installing on a Pi or a Jetson then I can direct you to specialist installation instructions for those boards.

takijo0116, 2021-01-28:
Thank you for your answer.

My computer is laptop.

I will try RSUSB backend method.

Also, Is it possible to  return to v4l2 backend ?

MartyG-RealSense, 2021-01-28:
You are not committed to a particular installation method permanently.  I would recommend removing the previous installation before trying a different method, but otherwise it is fine.  For example, removing a package-file installation of librealsense before building it from source code to avoid a 'multiple udev files installed' error.

The link below has guidance for installing with RSUSB using the CMake flag **-DFORCE_RSUSB_BACKEND=true**

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

takijo0116, 2021-01-29:
I cannot find CMakeLists.txt in Librealsense root directory.

Is it need to make the file?

MartyG-RealSense, 2021-01-29:
Did you download the source code zip-file from the **Assets** file list of the SDK **Releases** page and extract the zip-file's contents to obtain the librealsense source-code folder, please?

[image]

Typically, the extracted source code folder's structure takes the form of **librealsense > librealsense**, with the files being in the sub-folder (the real 'parent' folder).  CMakeLists.txt is within this sub-folder.

[image]

takijo0116, 2021-01-30:
Thank you for your response.
I will try it/

MartyG-RealSense, 2021-02-07:
Hi @takijo0116  Do you require further assistance with this case, please?   Thanks!

takijo, 2021-02-07:
No.
I solved it.
Thank you for your help.

MartyG-RealSense, 2021-02-07:
That's great news @takijo - thanks very much for the update!

closed 2021-02-07 by MartyG-RealSense