realsenseai/librealsense #8266: Failing to cmake RealSense OpenVINO wrappers : class error_listener : public InferenceEngine::IerrorListener DEPRECATED

MikeHallettUK, 2021-01-28:
Failing to cmake RealSense OpenVINO wrappers : class error_listener : public InferenceEngine::IerrorListener DEPRECATED
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |  Ubuntu 20 | 
| Kernel Version (Linux Only)    |  4.49                                     | 
| Platform                                 | ODROID N2+  |
| SDK Version                            |  2.41                          | 
| Language                            |  na                        | 
| Segment			|  Robot                         | 

### Issue Description
Trying to cmake RealSense OpenVINO wrappers :  e.g.

```
cd ~/librealsense/build
cmake .. -DBUILD_OPENVINO_EXAMPLES=true -DOpenCV_DIR=/usr/lib/aarch64-linux-gnu -DINTEL_OPENVINO_DIR=/home/odroid/openvino
make -j4
```
it fails with error which seems to be caused because InferenceEngine::IerrorListener is DEPRECATED: e.g. in openvino_helpers.h

`Class error_listener : public InferenceEngine::IerrorListener `

These wrappers look quite "old"... is there any chance they can be updated to fit with current versions ?

Or is there a simple fix I can use please ?

Thanks

MartyG-RealSense, 2021-01-28:
Hi @MikeHallettUK   The examples in the OpenVINO wrapper were updated in late 2020 to work with the 2019 and 2020 versions of the OpenVINO Toolkit.

[image]

The **ODROID N2+** that you own has an **Arm** processor though according to its specification details.  The system requirements for the OpenVINO Toolkit specify requirement for an **Intel** processor.

https://docs.openvinotoolkit.org/latest/openvino_docs_install_guides_installing_openvino_linux.html#system_requirements

MikeHallettUK, 2021-01-29:
OK thanks.  Sorry, I should have spotted that: I have used the latest 2021 openvino = built from source, successfully.  I guess I can go back to 2020 version....

closed 2021-01-29 by MikeHallettUK