realsenseai/librealsense #8267: Removing water

kaosmaja, 2021-01-28:
Is there an easy way to remove the water surface from a video taken from a moving boat?

MartyG-RealSense, 2021-01-28:
Hi @kaosmaja   If you are referring to RGB video recorded by a RealSense camera, there may be video editing techniques that you can use to remove such artifacts from the video.  I suggest googling for **remove water splash from video** for more information.  An example reference found from such a search is the one in the link below:

https://www.researchgate.net/publication/285317526_Video-based_Water_Drop_Detection_and_Removal_Method_for_a_Moving_Vehicle

kaosmaja, 2021-01-28:
I actually hadn't even considered water splash yet, thank you for the link. The problem I have is that I had expected that the RealSense camera's depth feed could be used to judge distance to nearby objects in front of the boat, but it turned out that the depth camera sees the water surface just fine. That's why I'm looking for a solution that would help remove this "noise".

MartyG-RealSense, 2021-01-28:
May I confirm whether you are seeking a solution that ignores the water and just sees objects ahead of the boat, please?  And is the depth sensing for the purposes of obstacle avoidance?

kaosmaja, 2021-01-28:
Yes, that is exactly it.

MartyG-RealSense, 2021-01-28:
You could set a **Threshold Filter** to exclude from the depth data measurements outside a minimum and maximum depth sensing range (for example, excluding depth readings less than 1 meter from the camera).  This should also take care of excluding unwanted depth readings from water spray directly in front of the camera as the boat moves.

Another approach would be to use a technique called RANSAC to remove the floor (in this case, the ocean) from the depth image.

https://github.com/IntelRealSense/librealsense/issues/3330

Another navigation technique is to read image tags attached to objects instead of sensing depth.  This approach was used by an MIT project called **roboats** to navigate robotic boats to docks via RealSense 400 Series cameras on the boat's bottom cushion (navigation was by Velodyne lidar camera atop the boat).  This would not be practical for negotiating objects that have not been pre-tagged though.

https://support.intelrealsense.com/hc/en-us/community/posts/360034814013-Update-on-RealSense-equipped-robot-roboat-autonomous-boats

closed 2021-02-11 by MartyG-RealSense