realsenseai/librealsense #8269: `'numpy.ndarray' object has no attribute 'as_depth_frame'` when using convert_depth_pixel_to_metric_coordinate from the helper_functions.py

monacv, 2021-01-29:
I am using the helper_function that you have for the following:

```

                                    depth_val= get_depth_at_pixel(depth_frame, x, y)
                                    x_m, y_m, z_m = convert_depth_pixel_to_metric_coordinate(depth_val, x, y, intrinsics)

                                    print(" metric x {}, y{} , z{} and also depth{}".format(x_m, y_m, z_m, depth_val))
```

And I get this error:

`'numpy.ndarray' object has no attribute 'as_depth_frame'`

Here's the related part from the helper_functions.py:

```
def get_depth_at_pixel(depth_frame, pixel_x, pixel_y):
        """
        Get the depth value at the desired image point

        Parameters:
        -----------
        depth_frame      : rs.frame()
                                                   The depth frame containing the depth information of the image coordinate
        pixel_x                          : double
                                                   The x value of the image coordinate
        pixel_y                          : double
                                                        The y value of the image coordinate

        Return:
        ----------
        depth value at the desired pixel

        """
        return depth_frame.as_depth_frame().get_distance(round(pixel_x), round(pixel_y))

```

I even tried converting the depth_frame to float but didn't work as well.
`depth_float = original_depth_frame.astype(np.float32)`

Could you please suggest how to fix this problem?
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py

yosoufe, 2021-01-29:
Hi, What is the `depth_frame`? How did you create it? Could you share a bit more code above it?

monacv, 2021-01-29:
@yosoufe 
here's my depth_frame : 
`       frame = cv2.imread(os.path.join(args.depth_input_dir, imgs_dir[frame_idx]), cv2.IMREAD_ANYDEPTH)`

yosoufe, 2021-01-29:
`get_depth_at_pixel` function is not for opencv image. You already have the numpy array. If your frame is depth image (np.array), you just need to  use `frame[x,y]` to get the depth at a pixel.
```py
x_m, y_m, z_m = convert_depth_pixel_to_metric_coordinate(depth_frame[x,y], x, y, intrinsics)
print(" metric x {}, y{} , z{} and also depth{}".format(x_m, y_m, z_m, depth_val))
```

monacv, 2021-01-29:
Then the question is:

How can I convert an image read by opencv into rs.frame?

yosoufe, 2021-01-29:
I am not sure now, Could you share what you have that I could run it myself this evening and explore more? I am also learning.

monacv, 2021-01-29:
@MartyG-RealSense  could you help perhaps?

MartyG-RealSense, 2021-01-29:
Hi @monacv   If the question that you require help with is the one from above - "How can I convert an image read by opencv into rs.frame?" - then the only reference that I know of for mat to frame conversion instead of frame to mat is this C++one:

https://github.com/IntelRealSense/librealsense/issues/2634

closed 2021-02-14 by MartyG-RealSense