realsenseai/librealsense #8272: QueryDevices() throws memory errors

LucaFranceschini, 2021-01-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Windows 10 (version 1507) | 
| Platform                                 | PC  |
| SDK Version                            |  2.0.50727                          | 
| Language                            |  C#                          | 
| Segment			|  ?                         | 

### Issue Description
I'm using the following code to check for connected devices:

```c#
new Context().Devices.Count > 0
```

This occasionally throws memory errors like this:

> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: bad_weak_ptr ---> System.Runtime.InteropServices.ExternalException: rs2_query_devices_ex(context:000000248365EE10, product_mask:254)
	   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData)
	   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
	   at Intel.RealSense.NativeMethods.rs2_query_devices_ex(IntPtr context, Int32 mask, Object& error)
	   at Intel.RealSense.Context.QueryDevices(Boolean include_platform_camera)

Or this:

> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
	   at Intel.RealSense.NativeMethods.rs2_query_devices_ex(IntPtr context, Int32 mask, Object& error)
	   at Intel.RealSense.Context.QueryDevices(Boolean include_platform_camera)

This happens at my application's start up, before I try to use the camera. It is not completely reproducible. Also, the camera was not even connected in the second case (don't know about the first one).

Can this be caused by wrong usage on my end? Can it be caused by multiple processes checking for connected devices simultaneously?

MartyG-RealSense, 2021-01-29:
Hi @LucaFranceschini   The link below features a case where a RealSense user was calling C++ code with a D415 from a C# Winforms application and was encountering the ** Attempted to read or write protected memory. This is often an indication that other memory is corrupt** error when starting and stopping a stream.  Is this scenario similar to your project please?

https://github.com/IntelRealSense/librealsense/issues/6459

LucaFranceschini, 2021-01-29:
> Hi @LucaFranceschini The link below features a case where a RealSense user was calling C++ code with a D415 from a C# Winforms application and was encountering the **Attempted to read or write [...]

Our project is a WPF application (though some Windows Forms libraries are occasionally used). In our case though, the error happens before we have any chance of closing and opening a stream.

Other processes may run at the same time and interact with the camera, if that helps, but those are outside our control and our codebase.

MartyG-RealSense, 2021-01-29:
As a next step @LucaFranceschini I have cross-connected your case with another open case involving possible problems with query_devices().

https://github.com/IntelRealSense/librealsense/issues/8091

If you look at the details of that case then you may find some common ground.

LucaFranceschini, 2021-02-03:
> As a next step @LucaFranceschini I have cross-connected your case with another open case involving possible problems with query_devices().
> 
> #8091
> 
> If you look at the details of that case then you may find some common ground.

Thank you. I see in that case the issue is a memory leak, while what we observed is a memory error.

MartyG-RealSense, 2021-02-03:
As your project uses C# and is on Windows, do you know if the **RealSense UWP driver** might have been installed onto your computer?  If version **6.11.160.21** of this driver is installed then it has been reported to cause errors with the C# wrapper and with the RealSense Unity wrapper (Unity is C# based).

[image]

LucaFranceschini, 2021-02-09:
> As your project uses C# and is on Windows, do you know if the **RealSense UWP driver** might have been installed onto your computer? If version **6.11.160.21** of this driver is installed then it [...]

Sorry for late reply, but no, that driver was not installed.

MartyG-RealSense, 2021-02-09:
It would be useful to know please which version of the RealSense SDK you are using.  You list it as **2.0.50727** at the top of this case but that is not a recognisable SDK version number (it would be between 2.8 and 2.41 at the time of writing this).  

Your listed camera firmware version of **05.11.06.250** is also old as it dates back to June 2019.  I would recommend a firmware of up to **5.12.7.100** for SDK versions up until 2.38, and firmware **5.12.8.200** for SDK versions that are 2.39 or newer.

MartyG-RealSense, 2021-02-16:
Hi @LucaFranceschini  Do you have an update about your case that you can provide, please?   Thanks!

LucaFranceschini, 2021-02-16:
Sorry, I used the C# wrapper reference version from Visual Studio rather than the correct SDK version. We are using 2.40, I updated the first post accordingly.

From your comment I understand we should upgrade both the SDK and the camera firmware. I'll post an update as soon as I manage to do that.

MartyG-RealSense, 2021-02-16:
Thanks very much - I look forward to your update.   Good luck!

MartyG-RealSense, 2021-02-28:
Hi @LucaFranceschini   Do you have an update that you can provide about your case, please?  Thanks!

LucaFranceschini, 2021-03-01:
> Hi @LucaFranceschini Do you have an update that you can provide about your case, please? Thanks!

Sorry, looks like updating libraries on our devices is not really an option at the moment. I guess you can close the issue if you want since we're using an old version, and I'll post again if we update to latest SDK and firmware and observe the problem again. Thanks for your support!

MartyG-RealSense, 2021-03-01:
Thanks very much @LucaFranceschini for the update.  I will close the issue as you suggested - feel free to re-open this case in future or begin a new one if the problem is observed again.

closed 2021-03-01 by MartyG-RealSense