realsenseai/librealsense #8273: multicam example won't start L515 ?(L515 + T265)

jack-wlj, 2021-01-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515 + T265 } | 
| Firmware Version                   | (L515 1.5.0.0) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  { legacy / 2.<38>.<1> }                          | 
| Language                            |  {C/C++/opencv/pcl}                          | 
| Segment			|  {Robot/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I  use both T265 and L515. I can get data from L515+T265 in Win10;

But when I ported the code to Jetson AGX Xavier.

The MultiCam program that comes with the SDK can see the information.Pictures as follows:

[image]

 But the source code I compiled didn't get the L515 data.I only got T265 data.Pictures as follows:

[image]

Who encountered similar problems?
Is there anyone who can solve my problem?
Thanks a lot!

RealSenseSupport, 2021-01-29:
Hi @jack-wlj 

Have you gone through the installation process here:  https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

How about trying the RSUSB Backend and compare it to Native Backend build?

Windows is definitely a different build/installation than the Jetson and uses different USB backend.

jack-wlj, 2021-01-30:
> 嗨@ jack-wlj
> 
> 您是否已在此处完成安装过程：[https](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) : [//github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
> 
> 如何尝试RSUSB后端并将其与本机后端版本进行比较？
> 
> Windows绝对是不同于Jetson的构建/安装，并且使用不同的USB后端。

Thank you for your reply. I will try to compile SDK again ,and try to  building from Source using RSUSB Backend .

I will update this issue when the compilation is complete.

jack-wlj, 2021-02-03:
> 嗨@ jack-wlj
>
> 您是否已在此处完成安装过程：[https](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) : [...]

Hi,
I recompiled the code,
The same problem came up.
I  compiled the application with librealsense using   g++ -std=c++11 rs-multicam.cpp -lrealsense2 -lGL -lglut  -o rs-multicam ,
When I run the program ./rs-multicam, I can only get T265 data.I don't have data on the L515.
I tried it on two devices, the Jetson AGX  and Jetson  NX,  It's the same problem.
Why does the same rs-multicam.cpp get different results when compiled in two different ways?
So how do I compile rs-multicam.cpp code separately to get data from both T265 and L515 at the same time?
Looking forward to your reply. Thank you!

jack-wlj, 2021-02-03:
The version of the SDK I'm using is 2.41,The L515 firmware version is 1.5.3.

RealSenseSupport, 2021-03-02:
Hi,

We are able to build the SDK on our JETSON NANO and successfully run multicam sample with both T265 and L515 on latest repo..

If you build the entire SDK with samples enabled are you able to run the binary rs-multicam that is installed and stream? What about realsense-viewer?

Thanks

closed 2021-04-02 by RealSenseSupport