realsenseai/librealsense #8276: On-chip calibration time-out

rlustig-gpfw, 2021-01-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-132-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.41                         | 
| Language                            |  python                           | 
| Segment			|  Robot                          | 

### Issue Description
I am unable to get the on-chip calibration to finish successfully as it always times-out. I have tried using multiple methods:
1. RealSense viewer app: Running the `OCC All` option, I see the following printed in the terminal multiple times a second:
``` 
29/01 15:14:27,808 ERROR [140224994711296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 
29/01 15:14:27,808 ERROR [140224994711296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
```

2. Python using [depth_auto_calibration_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py): Running the default (with no json options) always ends with the following error:
```
Operation timed-out!
Calibration state did not converged in time
```

Any suggestions on how to get on-chip calibration to work for me?
Follow up question, does the camera have to be perfectly still when running OCC or can it be hand-held?

MartyG-RealSense, 2021-01-30:
Hi @rlustig-gpfw  Holding the camera in the hand should not prevent calibration from completing and offering you a health check score.  I always calibrate with the camera held in hand.

I ran extensive tests with the On Chip Calibration tool in the Viewer with a D435i and a D455 on two different laptops.  At different times I encountered both of the errors in points 1 and 2 above but could not identify a consistent, repeatable cause for them.

The factor that most increased the chances of the test completing successfully seemed to be to position the camera far enough away from the observed surface (e.g a flat wall) that there was a solid box represented in the Viewer's center panel instead of a group of broken multi-colored pixels (which can occur if the camera is close to the surface).

[image]

Do you see any improvement if you select the **Fast** mode of calibration from the On-Chip Calibrator's drop-down menu before clicking the **Calibrate** button?

[image]

rlustig-gpfw, 2021-02-01:
Hi @MartyG-RealSense, thank you for the fast response and for performing additional testing. In my tests, I too make sure that the camera is far enough away from the wall (i.e. not within the camera's "deadzone"). For reference, the camera is ~.75 meters away from the wall.

Changing to the **Fast** calibration mode improves things a bit, but does not solve the issue. I ran the calibration with the **Fast** mode option many times and I saw a few different issues. The issues can be filtered into 2 groups:
1. The calibration process begins, but soon after prints these log messages:
```
 01/02 08:14:40,624 ERROR [139754595145472] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/02 08:14:40,624 ERROR [139754595145472] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
 01/02 08:14:40,658 ERROR [139754595145472] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/02 08:14:40,658 ERROR [139754595145472] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
 01/02 08:14:40,819 WARNING [139754580027136] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 771072 out of 814335 bytes (94%)
 01/02 08:14:44,757 WARNING [139754580027136] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 75000 out of 814335 bytes (9%)
 01/02 08:14:45,247 WARNING [139754580027136] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 115712 out of 814335 bytes (14%)
 01/02 08:14:45,626 ERROR [139754554849024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/02 08:14:45,627 ERROR [139754554849024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
 01/02 08:14:45,659 ERROR [139754554849024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 01/02 08:14:45,660 ERROR [139754554849024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
```
It is not clear what the percentages in the log messages are referring to.
2. Sometimes the calibration progress bar completes and the viewer displays an updated colored depth image for a brief moment (~1 second). This image seems distorted as if the calibration procedure went badly. After about 1 second of displaying this image, the calibration process seems to restart and try again with the calibration progress bar restarting and completing. At this point, the application shows a message saying the calibration process failed. Sometimes during this process the following log messages are printed multiple times:
```
 01/02 07:52:59,511 WARNING [139858018297600] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
 01/02 07:52:59,713 WARNING [139858018297600] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
 01/02 07:52:59,914 WARNING [139858018297600] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
```

Lastly, would the calibration process fail if it receives an incomplete frame? From my testing, it seems that way, but it could just be a coincidence.

MartyG-RealSense, 2021-02-01:
The **Incomplete frame received** warning makes me think that there may be a USB related problem occurring.  The link below discusses possible causes of this warning.

https://github.com/IntelRealSense/librealsense/issues/8202

It may also be worth trying the other available RealSense calibration tool, the **Dynamic Calibrator**.  Page 14 onward of the PDF user guide for the tool explains how to install it on Linux.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

rlustig-gpfw, 2021-02-01:
Hi @MartyG-RealSense, I am using a VM (Parallels) for my testing. Should that have any affect? If so, I will try to find a bare metal linux machine for further testing. As for the other suggestions (ASIC too hot, CPU over-utilized), those were not issues in my case. The only other potential issue suggested is that USB controller is somehow failing to handle the requests.

I've read about the Dynamic Calibrator but my use case for the camera is different and it won't work.

MartyG-RealSense, 2021-02-02:
Using a VM may cause problems regarding interfacing with USB.  If you do need to use a VM, Intel suggest in the opening paragraph of the installation instructions link below to use **VMware Workstation Player**.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

MartyG-RealSense, 2021-02-19:
Case closed due to no further comments received.

rlustig-gpfw, 2021-02-19:
Hi @MartyG-RealSense, can we re-open this ticket? I got a separate laptop this week and it runs Ubuntu 16.04 (kernel 4.15.0-132-generic) exclusively (not a VM), hoping this would remove any software OS issues.

However, I still see the same issues. Running on-chip calibration on the `fast` setting, the calibration process starts for about 1 second and then the depth camera images on the viewer stop updating (the last depth image is frozen on the viewer). At the same time, the following log messages below are printed out (this is just a snippet - I get about 4 of these logs per second).

```
 19/02 11:08:15,847 WARNING [139843394860800] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
 19/02 11:08:16,047 WARNING [139843394860800] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
 19/02 11:08:16,248 WARNING [139843394860800] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
 19/02 11:08:16,449 WARNING [139843394860800] (ds5-auto-calibration.cpp:455) hwmon command 0x80( 14 0 0 0 ) failed (response -10= Integrity error)
```

The only other option I can think of is to try a different camera. Do you have any further thoughts?

MartyG-RealSense, 2021-02-19:
Hi @rlustig-gpfw   Thanks very much for the update.  I will re-open the case now, and will look at it again tomorrow.

MartyG-RealSense, 2021-02-20:
I thoroughly re-tested the On-Chip Calibrator and got the same results as my original set of tests.  The errors were generated in the log but the process was able to complete successfully, provide a camera health-check value and offer to write the calibration to the camera or dismiss it.

Given that you have encountered the same problem on more than one computer, that would lend weight to it being a camera-related problem rather than a computer-related one.  The camera has a **calibration table** stored inside it that can prevent calibration it is has become corrupted.

A corrupted calibration table can be corrected by performing a process called a **gold reset** that restores the camera to its factory-new defaults.  This requires the installation of the RealSense **Dynamic Calibration** tool.  Instructions for doing so for Linux (Ubuntu 16.04) can be found on page 14 onwards of the tool's user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Installation of the Dynamic Calibration tool should also install a tool called **CustomRW** that is discussed on page 58 of the guide.

If the CustomRW tool is launched from the Linux terminal with the **-g** condition using the instruction below then a gold reset of the camera should be initiated that should restore the camera's calibration table.

 **Intel.RealSense.CustomRW -g**

rlustig-gpfw, 2021-03-01:
I was able to get a second camera and re-test all the calibration methods. Everything performed normally and the camera calibration for both intrinsics and extrinsics worked correctly. 

For my other camera, I will try a gold reset and see if this fixes the issues I saw earlier.

MartyG-RealSense, 2021-03-01:
Okay, thanks very much for the update!  I will close this ticket as suggested.  Good luck   :)

closed 2021-03-01 by MartyG-RealSense

YuzheHao, 2021-03-15:
I met exactly the same problem. My camera is D435i.
I also tried to use the another D435 camera, indeed, no similar problem happened on that.
Though I could try to get a new camera or do the gold test to reset it to factory-defaults, but I still want to know why this thing happened.
Since some people like me will use the camera to collect a lot of data, we can not always turn to a new camera every time this problem happened, right?

Thank you for the understanding and hope could get your reply soon.

MartyG-RealSense, 2021-03-15:
Hi @YuzheHao   I re-examined this issue but was not able to find any new information to add to the details above, unfortunately.