realsenseai/librealsense #8277: Building librealsense android app with NDK (native c++ code)

zc08, 2021-01-30:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Android 28+ |
| Kernel Version (Linux Only) | - |
| Platform | Android |
| SDK Version | 2.41.0 |
| Language | C/C++/Java |
| Segment | Robot |

### Issue Description
I'm trying to build an android app. 
1. The app utilizes librealsense android wrapper to get frames (which returns `com.intel.realsense.librealsense.FrameSet`).
2. It then passes the obtained frames to some native C++ functions for further processing. The native C++ functions takes `rs2::frameset` as input argument, e.g. `int NativeFrameProcessing(rs2::frameset& frames)`. This is achieved by using NDK.

By looking at the sample android programs, the frame acquisition part in android seems quite clear. And also the librealsense code has shown passing frames to native part by converting native pointers to `jlong`. 

What I'm not clear with, is how to specify the librealsense dependencies in both the android part and the NDK (native code) part.

In android, the dependency is specified as:
```
    repositories {
        maven{
            url "https://dl.bintray.com/intel-realsense/librealsense-dev"
        }
    }

    dependencies {
        implementation 'com.intel.realsense:librealsense:2.41.0+@aar'
    }
```

In NDK (native code part), I use CMake to generate compile configurations. 
```
    find_package(realsense2 REQUIRED)
```

But I guess in this way, CMake would find the librealsense that is installed in the current system (Ubuntu in this case as I'm doing the project in Ubuntu), which is not built against android devices, and the version may not be consistent with the downloaded AAR android library. 

So how can I correctly specify the dependencies to librealsense ? For example, pointing CMake to somehow use the downloaded AAR library so both the android part and NDK part depends on the same librealsense ?  Thanks in advance !

MartyG-RealSense, 2021-01-30:
Hi @zc08  As a starting point in investigating your case, may I ask some questions in order to better understand the configuration of your Android development environment.

*  It looks as though you are using the **unrooted** version of the Android wrapper that uses AAR.  

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

Can you confirm that this is the wrapper that you are using, please?

*  Are you building the Android wrapper using **Android Studio** or building using a different method?  The information that you have kindly provided above suggests that you are building with **CMake**.

If you are using CMake, advice is provided in the link beow about buiding **librealsense.so** with CMake and configuring the USB permissions for working with Native code.

https://github.com/IntelRealSense/librealsense/issues/3385

zc08, 2021-01-31:
@MartyG-RealSense Thank you for the reply ! 

> It looks as though you are using the unrooted version of the Android wrapper that uses AAR.
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

Yes, I'm using this one.

> Are you building the Android wrapper using Android Studio or building using a different method? The information that you have kindly provided above suggests that you are building with CMake.

I'm building an Android app (not the wrapper itself, I have built the wrapper successfully), which has both android part and native code part. The native code part is called from android java code through NDK. Both parts use some functions from librealsense. 

The whole project is built with android studio, but the native code part uses CMake.  CMake is invoked by specifying the following commands in build.gradle script:
```
        externalNativeBuild {
        cmake {
            path "src/main/cpp/CMakeLists.txt"
            version "3.10.2"
        }
```

MartyG-RealSense, 2021-01-31:
Thanks for your patience,  I examined and researched the details of your particular project carefully.  I have not dealt with a project structured like this before though.

You may have already read the wrapper's native application example, but I will link to it in case you have not.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

I will also tag Android wrapper developer @matkatz into this case to seek advice about your question above in https://github.com/IntelRealSense/librealsense/issues/8277#issuecomment-770350447

zc08, 2021-01-31:
Thank you very much for the reply ! 

> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

I have overlooked this example and it does seem to solve my problem. 
I just need the *.so files and header files corresponding to the aar file, so I can refer to them in CMakeLists.txt. 
I'll try out this solution and see if it works, thanks !

MartyG-RealSense, 2021-02-01:
Remembering that you had referenced **bintray** in your installation, I wanted to direct you to a message about the installation process that a RealSense team member has just posted:

https://github.com/IntelRealSense/librealsense/issues/8287#issuecomment-770854785

zc08, 2021-02-02:
I was able to compile the project using this url: "[https://dl.bintray.com/intel-realsense/librealsense-dev](https://dl.bintray.com/intel-realsense/librealsense-dev)", instead of "https://dl.bintray.com/intel-realsense/librealsense" in the examples.

Should I use the new address "https://egiintel.jfrog.io/artifactory/librealsense" instead ?

MartyG-RealSense, 2021-02-02:
I would recommend asking about it in https://github.com/IntelRealSense/librealsense/issues/8287

closed 2021-02-16 by MartyG-RealSense