realsenseai/librealsense #8279: Depth value mismatch between API call and realsense_viewer

tr-gis, 2021-01-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   {Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-41-generic                                        | 
| Platform                                 | Ubuntu PC  |
| SDK Version                            | 2.0                         | 
| Language                            |  C++                     | 
| Segment			|  others                           | 

I am trying to find depth of certain points of interest using Intel RealSense D435i. I modified source code from hello_realsense example [https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense](url) to get depth data.

The code snippet looks like below:
```

int main(int argc, char * argv[]) try
{
    // Create a Pipeline - this serves as a top-level API for streaming and processing frames
    rs2::pipeline p;

    // Configure and start the pipeline
    p.start();

    while (true)
    {
        // Block program until frames arrive
        rs2::frameset frames = p.wait_for_frames();

        // Try to get a frame of a depth image
        rs2::depth_frame depth = frames.get_depth_frame();

        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(300, 50);

        // Print the distance
        printf("Depth = %f\n", dist_to_center);
    }

    return EXIT_SUCCESS;
}

```
I am interested in finding depth of point (300,50). The output I get when running the executable is 

> Depth = 1.406000

However when launching realsense-viewer and looking at the same pixel co-ordinates, I get a different value (0.569 meters). Why is there this discrepancy ?. Am I missing something?

[image]

You can see the depth value displayed on the bottom in the above image.

MartyG-RealSense, 2021-01-30:
Hi @tr-gis  The **width** and **height** values will be determined by the default depth frame dimensions of the particular camera model that is being used with **rs-hello-realsense** and your adaptation of it, as a custom resolution is not defined in the code.

**float width = depth.get_width();
float height = depth.get_height();**

As the default resolution of the D435i is **848x480**, are you using this resolution for the depth stream in the Viewer, please?

It may also be worth toggling to **Off** (red) the parent **Post-Processing** option in the Viewer's side-panel, since the Viewer applies a range of filters to the image by default whilst a script does not apply these filters unless they are deliberately programmed into the application.

[image]

In addition, to ensure consistency between programs, you could attach the camera to the tripod supplied with the camera if you have the tripod available so that the camera is pointing to the same area when running both your own program and the Viewer.

tr-gis, 2021-02-01:
@MartyG-RealSense Toggling off post-processing option has solved this issue. Thank you :)

closed 2021-02-01 by tr-gis

MartyG-RealSense, 2021-02-01:
Great news that toggling post-processing off worked for you @tr-gis - thanks for the update!