realsenseai/librealsense #8283: Problem with installation:  focal

Prakyathkantharaju, 2021-02-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | | 
| Operating System & Version |   Linux (Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.8                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.48                        | 
| Language                            |                       | 
| Segment			|                      | 

### Issue Description
Hello,

I have facing these two problem while installing the librealsense.

#### 1. NO_PUBKEY
key not found for the Ubuntu focal.
```
W: GPG error: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
### 2. Question?
How to launch realsense-viewer after installing librealsense from source ?

Thank you in advance.

ev-mp, 2021-02-01:
@Prakyathkantharaju, the `focal` repo is up and running.
Please check that you have the public key registered according to [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages):
```sudo apt-key adv --keyserver keys.gnupg.net ...```

RealSenseSupport, 2021-02-08:
@Prakyathkantharaju Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

peci1, 2021-03-03:
Isn't the problem the fact that OP runs HWE 20.04.2 with kernel 5.8, which is stated as unsupported?

RealSenseSupport, 2021-03-08:
@peci1  @Prakyathkantharaju  As replied in https://github.com/IntelRealSense/librealsense/issues/8209#issuecomment-764803780, kernel 5.8 support is on our list but no schedule yet. Please try to switch to other supported kernel version. Thanks!

RealSenseSupport, 2021-03-15:
@peci1 @Prakyathkantharaju Did you get chance to test with other supported kernel version? Any other questions? Looking forward to your update. Thanks!

peci1, 2021-03-15:
How could I test on HWE 20.04.2 with a supported kernel version? There's no such thing (as it comes with 5.8). And installing any older kernel on the HWE seems like you're asking for problems... I won't try that.

RealSenseSupport, 2021-03-15:
@peci1 I just checked my setup of 20.04.1. The kernel version is 5.8.0 and there's no problem to install librealsense according to the link https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages? Could you please try again? Thanks!

peci1, 2021-03-15:
The same wiki you link to says:

> The Realsense DKMS kernel drivers package (librealsense2-dkms) supports Ubuntu LTS kernels 4.4, 4.8, 4.10, 4.13, 4.15, 4.18*, 5.0*, 5.3* and 5.4.

Is it outdated?

Last time I tried on 5.8, I didn't succeed installing the DKMS module. Do you actually try to install the dkms module, too? Because I've seen librealsense came with some alternative that uses a less powerful framework to work without patching kernel, but without e.g. multi camera support...

RealSenseSupport, 2021-03-16:
@peci1 Yes, I installed DKMS module too. But as kernel 5.8 is not listed in officially supported kernel list, the stability under kernel 5.8 is not guaranteed at this point.

RealSenseSupport, 2021-03-22:
@pecil Please also make sure you run "sudo apt update" before you install DKMS. Any update from your side? Thanks!

korin-lf, 2021-03-27:
I am having the kernel 5.8 problem too.
I verified the installation but modinfo uvcvideo | grep "version:"    gives
version:        1.1.1
srcversion:     46554C4AFA224F9EBD0866C

So the only solution is to downgrade the kernel?

peci1, 2021-03-28:
@RealSenseSupport I don't think you could manage to install librealsense-dkms on 5.8. This is what I get:

```
Building initial module for 5.8.0-44-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.17/5.8.0-44-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
```

It seems librealsense is actively protecting anyone from building the DKMS module on anything other than 5.4. How did you manage to install it?

peci1, 2021-03-28:
I installed the dkms module using the normal binary installation process on Ubuntu 20.04.

korin-lf, 2021-03-28:
Hi so i downgraded to kernel 5.4 but I am still getting:
pipe.start(cfg)
RuntimeError: No device connected

though realsense-viewer show the camera properly 
and modinfo uvcvideo | grep "version:" includes realsense string

so what can be done to run the examples?
thanks

RealSenseSupport, 2021-03-29:
@korin-lf Did you get such problem when run rs-enumerate https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices ? And please close RealSenseViewer before you launch other examples. Thanks!

RealSenseSupport, 2021-03-29:
> I installed the dkms module using the normal binary installation process on Ubuntu 20.04.

@peci1 Do you mean that you can install DKMS successfully now? Please note that the Debian Distribution PPA is being transferred and redirected to use new URL with Secure Access - HTTPS instead of HTTP
The new addresses shall start with https://librealsense.intel.com
The users who use seamless Ubuntu update shall modify the server's list according to the updated documentation. https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

peci1, 2021-04-05:
@RealSenseSupport I used the https APT repo when doing the last test. Installation of librealsense-dkms was only partially successful. The package itself installed (which is trivial in case of a DKMS package), but the post-install hook (DKMS build step) was unsuccessful.

RealSenseSupport, 2021-04-06:
@peci1 Could you please provide some snapshots about the failure? Or could you pleas try to uninstall previous DKMS first and reinstall the latest one?

RealSenseSupport, 2021-04-12:
@pecil Any update from your side? Looking forward to you update. Thanks!

peci1, 2021-04-12:
Here's the installation log:

```
peci1@cras-17:~$ uname -a
Linux cras-17 5.8.0-48-generic #54~20.04.1-Ubuntu SMP Sat Mar 20 13:40:25 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
peci1@cras-17:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
peci1@cras-17:~$ LANG=C apt-cache policy librealsense2-dkms 
librealsense2-dkms:
  Installed: (none)
  Candidate: 1.3.17-0ubuntu1
  Version table:
     1.3.17-0ubuntu1 500
        500 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 Packages
        100 /var/lib/dpkg/status
     1.3.16-0ubuntu1 500
        500 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 Packages
peci1@cras-17:~$ LANG=C sudo apt install librealsense2-dkms 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm11 linux-headers-5.6.0-1048-oem linux-image-5.6.0-1048-oem linux-modules-5.6.0-1048-oem linux-oem-5.6-headers-5.6.0-1048
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  librealsense2-dkms
0 upgraded, 1 newly installed, 0 to remove and 63 not upgraded.
Need to get 0 B/2150 kB of archives.
After this operation, 11.6 MB of additional disk space will be used.
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 432031 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.17-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.17-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.17-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.17 DKMS files...
Building for 5.8.0-48-generic 5.11.10-051110-generic
Building initial module for 5.8.0-48-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.17/5.8.0-48-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
Building initial module for 5.11.10-051110-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.17/5.11.10-051110-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.17: added
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d 
```

RealSenseSupport, 2021-04-12:
@peci1 It's a little wired that the log you provided shows that it's trying to build for two kernel versions. Could you please remove the unwanted one and make sure kernel 5.8 installed properly?

Building initial module for 5.8.0-48-generic
Building initial module for 5.11.10-051110-generic

peci1, 2021-04-12:
It's completely okay it tries to build for all installed kernel versions. I didn't expect it to succeed on 5.11. But even the 5.8 build fails...

peci1, 2021-04-12:
I was running this test while running kernel 5.8

peci1, 2021-04-12:
which is the stock kernel installed by ubuntu

RealSenseSupport, 2021-04-19:
@pecil Sorry, my mistake. I just noticed the similar error as you got below when install DKMS. In this case, we need apply the kernel patch manually. 

Error!  The /var/lib/dkms/librealsense2-dkms/1.3.17/5.8.0-48-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.

peci1, 2021-04-22:
Okay. So the conclusion is that 20.04.2 is not (yet) supported, right?

RealSenseSupport, 2021-04-23:
@peci1 Currently the supported version is listed as Release Note.

Ubuntu 16.04/18.04/20.04 LTS  (Linux Kernels 4.4, 4.8 ,4.10, 4.13, 4.15, 4.16(4) , 4.18, 5.0, 5.3, 5.4)

peci1, 2021-04-23:
> @peci1 Currently the supported version is listed as Release Note.

I know that. It was actually you (@RealSenseSupport ) who said it works with 5.8.0:

> @peci1 I just checked my setup of 20.04.1. The kernel version is 5.8.0 and there's no problem to install librealsense according to the link https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages? Could you please try again? Thanks!

RealSenseSupport, 2021-04-23:
@pecil Yes, that's right. RealSenseViewer can work without kernel patch for kernel 5.8. But if you want to enable metadata, then you need kernel patch.

RealSenseSupport, 2021-05-06:
@peci1 Any other questions about this ticket? Please note that DKMS package for kernel 5.8 is still under working.

RealSenseSupport, 2021-05-14:
@pecil Please note that the ticket will be closed if we don't hear from you in another 7 days. Thanks!

RealSenseSupport, 2021-05-18:
@pecil Kernel patch for kernel 5.8 is now supported by latest release. Please refer to https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2450 
Please try to apply the kernel patch with the script before DKMS is ready. Thanks!

RealSenseSupport, 2021-05-25:
@pecil Any other questions from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-06-07:
@pecil The latest SDK v2.47 now support DKMS for kernel 5.8 https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2470 Please have a try and feedback if any issue. Thanks!

RealSenseSupport, 2021-07-13:
@Pecil Any other questions about this? Please note that this ticket will be closed if we don't hear from you for another 7 days. Thanks!

peci1, 2021-07-13:
I tested it and now it works. Closing this issue.

RealSenseSupport, 2021-07-13:
@pecil Thanks for your update! Close the issue accordingly.

closed 2021-07-13 by RealSenseSupport