realsenseai/librealsense #8287: Could not find any matches for com.intel.realsense:librealsense:2.41.0+ as no versions of com.intel.realsense:librealsense are available.

kyosho-, 2021-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | Android  |
| SDK Version                            |  2.41.0 | 
| Language                            |  Java | 
| Segment			|  Smartphone | 

### Issue Description

I built it with reference to the following, but an error was output. 

https://github.com/IntelRealSense/librealsense/blob/d8f5d4212df85522a14b4a7b83bf4d54219b06fa/wrappers/android/examples/java_example/readme.md

```
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':app:debugCompileClasspath'.
   > Could not find any matches for com.intel.realsense:librealsense:2.41.0+ as no versions of com.intel.realsense:librealsense are available.
     Required by:
         project :app
```

I set it as follows. Am I doing something wrong?

```
repositories {
  maven{
       url "https://dl.bintray.com/intel-realsense/librealsense"
   }
}

dependencies {
   implementation 'com.intel.realsense:librealsense:2.41.0+@aar'
}
```

ev-mp, 2021-02-01:
@kyosho- hello,
The bintray hosting is unfortunately not available anymore, and we're transferring the Android packages distribution center to jFrog artifactory.
The newly-generated download links are already available on `development` branch.

With the new release we will issue a formal notice for the users.
Sorry for the inconvenience

mrjoelc, 2021-02-01:
I have the same issue and I don't understand what consequences will have this 

> @kyosho- hello,
> The bintray hosting is unfortunately not available anymore, and we're transferring the Android packages distribution center to jFrog artifactory.
> The newly-generated download links are already available on `development` branch. (see #8280 for details)
> 
> With the new release we will issue a formal notice for the users.
> Sorry for the inconvenience

I have the same issue but I don't understand what do you mean. Can you explain the immediate consequences?

ev-mp, 2021-02-01:
@mrjoelc , the URL links for Android precompiled dev. binaries are updated.
So if you're building an application from source then you update the links to the new location (specified in #8280)

kyosho-, 2021-02-02:
I was able to build. Thank you!

> @kyosho- hello,
> The bintray hosting is unfortunately not available anymore, and we're transferring the Android packages distribution center to jFrog artifactory.
> The newly-generated download links [...]

closed 2021-02-02 by kyosho-

MartyG-RealSense, 2021-02-02:
@kyosho-  Great news that you were able to build successfully!

@ev-mp   Thanks so much for your help!

luckycreationsindia, 2021-02-15:
Hi,

I'm trying to build apk using Java-Wrapper. Version 2.41 ad 2.42 both are not working on Android (Java Wrapper). And the maven doesn't include older versions. Please help me on this. The last version worked for me was 2.39.0.

Intel Device Model: D435i

mrjoelc, 2021-02-15:
For me was useful to update the link to the Android precompiled dev. binaries. 
In order to do this in Android you have to change inside the build.graddle(:app) this:

[image]

This is the new link.
(the older one was this: https://dl.bintray.com/intel-realsense/librealsense in my case)

> Hi,
> 
> I'm trying to build apk using Java-Wrapper. Version 2.41 ad 2.42 both are not working on Android (Java Wrapper). And the maven doesn't include older versions. Please help me on this. The last version worked for me was 2.39.0.
> 
> Intel Device Model: D435i

luckycreationsindia, 2021-02-15:
> https://dl.bintray.com/intel-realsense/librealsense

Older link is missing repository. They moved maven from bintray to jfrog.

mrjoelc, 2021-02-15:
That's the point! 
You have to change the url in the maven repo with this: "https://egiintel.jfrog.io/artifactory/librealsense", this is the new one!

> > https://dl.bintray.com/intel-realsense/librealsense
> 
> Older link is missing repository. They moved maven from bintray to jfrog.

luckycreationsindia, 2021-02-15:
> That's the point!
> You have to change the url in the maven repo with this: "https://egiintel.jfrog.io/artifactory/librealsense", this is the new one!
> 
> > > https://dl.bintray.com/intel-realsense/librealsense
> > 
> > 
> > Older link is missing repository. They moved maven from bintray to jfrog.

The new url doesn't have 2.39.0 which was working for me. Problem here is my app is unable to stream data from device as it's showing the following error:

 java.lang.SecurityException: User has not given 10277/com.xxx.xxx permission to access device /dev/bus/usb/001/002

MartyG-RealSense, 2021-02-15:
@luckycreationsindia   The last time that I saw this error with an Android 10 device, performing a firmware update corrected the problem for the RealSense user in that case.   Which firmware version is your camera using please? 

https://github.com/IntelRealSense/librealsense/issues/5388#issuecomment-623078025

ev-mp, 2021-02-16:
@luckycreationsindia , Android 10 has a known issue with regards to the camera permissions handling. 
While Google has released security patches later on, not all OEM picked and integrated it in their respective ROMs,  and therefore you won't be able to use unrooted Android10 with the SDK, at least not with the stock ROMs.