realsenseai/librealsense #8290: L515 Realsense Viewer not present in device panel

RGring, 2021-02-01:
| | |
|---|---|
|**librealsense**|2.41.0 RELEASE|
|**OS**|Linux|

Hi,
I can't get data from my L515 sensor in the Realsense Viewer. It is not showing up in the device panel.
Though in the log, the device is found, when I plug the sensor in (see attached image). Any ideas what could be the problem?

[image]

ev-mp, 2021-02-01:
@RGring hello, there is no indication of the device being attached/enumerated.
Can you share the content of "dmesg -wT" while L515 is being re-plugged?

As for the logs in the application - they apply to another  Web/embedded camera that is also connected to the host and are not indicative of L515 status

RGring, 2021-02-02:
Sorry, I was plugging in different devices to compare the dmesg and must have mixed them up. I updated the log above!

ev-mp, 2021-02-02:
@RGring , the device is recognized as USB2. There are couple of checks to be made:
- If the device was just unboxed then it needs USB3 to operate. Troubleshoot it by re-plugging and/or switching USB port/cable t.  Also note that upgrading the device later to the latest firmware removes that limitation and allows to operate it with both USB2 and USB3 connections (albeit with a subsets of resolution/fps modes).
- Make sure that the device permissions were set. It part of auto-install with Debians, and when [building from sources](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) it is performed with ```./scripts/setup_udev_rules.sh```

RealSenseSupport, 2021-02-26:
@RGring 

Were you able to get your device working or enumerated by the system under USB3?

RGring, 2021-03-02:
Hi @RealSenseSupport 
I got it working, but can't really report how. Plugged it in many different ways. Sometimes it worked, sometimes not. I was able to install the new firmware now, so hopefully that is resolved (haven't used it since). Thanks.

RealSenseSupport, 2021-03-23:
Glad you got it working.

closed 2021-03-23 by RealSenseSupport