realsenseai/librealsense #8291: Realsense viwer not starting

incinemator, 2021-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | - | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 5.4.0-64-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   -                       | 

Hello all!

I just installed realsense viewer following the instructions provided and everythin went smoothly. However when I type the real-sense viewer command I get the command not found error.. Any help?

Thanks in advance!

ev-mp, 2021-02-01:
@incinemator , the name of the executable is `realsense-viewer` and given the [Debian installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) it is part of  `sudo apt install librealsense2-utils` .
Make sure that the package has been properly installed.

In case of building from source make sure to deploy the executables with `sudo make install`.

incinemator, 2021-02-01:
@ev-mp yes that was it, thanks a lot!

closed 2021-02-02 by ev-mp