realsenseai/librealsense #8293: Unable to load Keras Unet model for denoising example

Spritaro, 2021-02-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  5.4.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0                         | 
| Language                            |  python                          | 
| Segment			|  Robot                         | 

### Issue Description
I'm trying to run denoising example described in [TensorFlow with Intel RealSense Cameras](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras#section-part-5-applying-trained-network-to-real-data). 

After downloading the [script](https://github.com/IntelRealSense/librealsense/blob/v2.40.0/wrappers/tensorflow/example5%20-%20denoise.py) and [Keras Unet model](http://realsense-hw-public.s3.eu-west-1.amazonaws.com/rs-tests/ML/Depth_Learning/DEPTH_Keras_Unet.model.zip) , I run the command below:
```sh
python3 'example5 - denoise.py' DEPTH_Keras_Unet.model
```

However, it raised an error:
```sh
Traceback (most recent call last):
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/training/py_checkpoint_reader.py", line 95, in NewCheckpointReader
    return CheckpointReader(compat.as_bytes(filepattern))
RuntimeError: Unsuccessful TensorSliceReader constructor: Failed to find any matching files for DEPTH_Keras_Unet.model/variables/variables

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/saved_model/load.py", line 633, in load_internal
    ckpt_options)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/keras/saving/saved_model/load.py", line 194, in __init__
    super(KerasObjectLoader, self).__init__(*args, **kwargs)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/saved_model/load.py", line 131, in __init__
    self._restore_checkpoint()
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/saved_model/load.py", line 328, in _restore_checkpoint
    self._checkpoint_options).expect_partial()
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/training/tracking/util.py", line 1275, in restore
    reader = py_checkpoint_reader.NewCheckpointReader(save_path)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/training/py_checkpoint_reader.py", line 99, in NewCheckpointReader
    error_translator(e)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/training/py_checkpoint_reader.py", line 35, in error_translator
    raise errors_impl.NotFoundError(None, None, error_message)
tensorflow.python.framework.errors_impl.NotFoundError: Unsuccessful TensorSliceReader constructor: Failed to find any matching files for DEPTH_Keras_Unet.model/variables/variables

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "example5 - denoise.py", line 23, in <module>
    model = keras.models.load_model(test_model_name)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/keras/saving/save.py", line 187, in load_model
    return saved_model_load.load(filepath, compile, options)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/keras/saving/saved_model/load.py", line 121, in load
    path, options=options, loader_cls=KerasObjectLoader)
  File "/home/spritaro/.local/lib/python3.6/site-packages/tensorflow/python/saved_model/load.py", line 636, in load_internal
    str(err) + "\n If trying to load on a different device from the "
FileNotFoundError: Unsuccessful TensorSliceReader constructor: Failed to find any matching files for DEPTH_Keras_Unet.model/variables/variables
 If trying to load on a different device from the computational device, consider using setting the `experimental_io_device` option on tf.saved_model.LoadOptions to the io_device such as '/job:localhost'.
```

```DEPTH_Keras_Unet.model/variables``` directory does contain ```variables.data-00000-of-00002``` and ```variables.data-00001-of-00002``` though... 
My guess is that it requires ```variables.index``` file in ```DEPTH_Keras_Unet.model/variables``` directory, like the one in [Using the SavedModel format](https://www.tensorflow.org/guide/saved_model#the_savedmodel_format_on_disk). Is there any way to confirm this?

MartyG-RealSense, 2021-02-02:
Hi @Spritaro   I must emphasise upfront that I have not used TensorFlow, so the analysis below is based on a careful examination of the documentation and model / example files to see where the problem might be occuring.

I assume that you have met all of the **installation** requirements already.  Your log certainly indicates that you have Python **3.6** installed, as recommended by the install instructions.

https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras#section-installation

You confirm that you have downloaded the Example 5 script file **example5 - denoise.py** and the Keras Unet model file **DEPTH_Keras_Unet.model**.

The Keras Unet model is downloaded as a zip file.  Although the instructions for Example 5 do not mention it, I make the assumption that the contents of the zip file should be extracted so that the folder named **DEPTH_Keras_Unet.model** can be accessed.  As you quote the **variables** sub-folder in your message, it sounds as though you have indeed extracted the zip.

At this point I believe that I have reached the stage where you are now - accessing the model.

I note that the **Example 5** documentation states that the model should be invoked with the instruction below:

`python camera_simulation.py <path to the model>`

Where 'path to the model' is, I assume, the path to the location on your computer of the  **DEPTH_Keras_Unet.model** folder that was extracted.

I see that the launch command that you used was a different one:

**python3 'example5 - denoise.py' DEPTH_Keras_Unet.model**

So that makes me wonder whether the model is invoked first and **then** the  'example5 - denoise.py' script file is launched.  For example:

`python camera_simulation.py DEPTH_Keras_Unet.model`
`python3 'example5 - denoise.py'`

As mentioned earlier, I am not a TensorFlow user and so the above information is written from the basis of an analysis of the available instructions.  So I apologise for any obvious errors.

Spritaro, 2021-02-03:
@MartyG-RealSense Thank you for your reply.

Yes, I extracted the model and specified the path to the extracted folder. Specifying a different path raises 'file does not exist' error, so I'm pretty sure it's the correct path.

About the ```camera_simulation.py``` script, I couldn't find it anywhere in the librealsense repository. 

Instead, I decided to try example 4 [Training on Depth data using TensorFlow](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras#section-part-4-training-on-depth-data-using-tensorflow) and train the model myself. With [Unet library](https://github.com/zhixuhao/unet) and provided [dataset](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras#section-training-dataset) (and some trial and error), I managed to create the model and run the ```'example5 - denoise.py'``` script successfully.

Although my model did work, the predicted depth was not very accurate. Probably because I haven't figured out the right parameters for training. If possible, I would still like to see the original model for comparison.

Spritaro, 2021-02-03:
@MartyG-RealSense Thank you for you reply.

Yes, I extracted the zip file and specified the path to the extracted folder. Specifying different path raised 'SavedModel file does not exist' error, so I'm pretty sure it was the correct path.

About the ```camera_simulation.py``` script, I could not find it anywhere in librealsense directory. Instead, I think I'll try [example4](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/tensorflow/example4%20-%20train.py) and [Training on Depth data using TensorFlow](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras#section-part-4-training-on-depth-data-using-tensorflow) to generate the model, and see if it runs with example5.

MartyG-RealSense, 2021-02-03:
Hi @Spritaro  The **camera_simulation.py** part of the **Example 5** instructions was a challenge for myself too as I could not find such a script.  I did not know if it was a pre-installed part of Python that could be found by the **python camera_simulation.py** command that referenced it, but internet searches about this file only lead to this librealsense example.  I will seek advice from the developer.

Hi @nohayassin  Thanks very much for your great work on the TensorFlow wrapper!  Would it be possible to offer @Spritaro advice on how to access the **camera_simulation.py** file in Example 5 please?  Thanks!

nohayassin, 2021-02-04:
Hi @MartyG-RealSense it is my pleasure and thanks for pointing out this issue. 
@Spritaro the script **camera_simulation.py** is renamed as **example5-denoised.py**, I will fix it.

MartyG-RealSense, 2021-02-04:
Thanks so much @nohayassin

Spritaro, 2021-02-04:
Thank you @nohayassin. I really appreciate your work.
If possible, could you also check whether the model downloaded from [the link in Part5](http://realsense-hw-public.s3.eu-west-1.amazonaws.com/rs-tests/ML/Depth_Learning/DEPTH_Keras_Unet.model.zip) can be loaded using the **example5-denoised.py**?

nohayassin, 2021-02-06:
Hi @Spritaro I checked the model in that link and I see it wasn't uploaded correctly, variables directory doesn't  contain the file variables.index. 
I will fix it but in the meantime you can download the complete model [from here](https://drive.google.com/file/d/1n_39b51KNUeZldxnB--wtP0YptqGEhMD/view)

Spritaro, 2021-02-07:
Hi @nohayassin, I tried your model and it worked really well!
Thank you very much for your help!

MartyG-RealSense, 2021-02-13:
Hi @Spritaro  Do you require further assistance with this case, please?   Thanks!

Spritaro, 2021-02-13:
Thank you for yor assistance. I will close this issue.

closed 2021-02-13 by Spritaro

MartyG-RealSense, 2021-02-14:
Thanks very much @Spritaro for the update

L-xn, 2022-06-11:
> 你好@L-xn 我相信下面链接中的**Keras Unut 模型**文件就是您所要求的。
> 
> https://librealsense.intel.com/rs-tests/ML/Depth_Learning/DEPTH_Keras_Unet.model.zip

Thank you very much！The link to part2 in the picture is invalid, could you share it too?
[image]

MartyG-RealSense, 2022-06-11:
The highlighted link is working for me.  It is a long 4.8 gb download so if the download appears to start but not progress, I recommend leaving it to download and doing something else.  

https://librealsense.intel.com/rs-tests/ML/Depth_Learning/Dataset/part2_4001_8375.zip