realsenseai/librealsense #8296: How to pass FrameSet object from Android Java code to NDK C/C++ code ?

zc08, 2021-02-02:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Android 8.0+ |
| Kernel Version (Linux Only) |  |
| Platform | Android |
| SDK Version | 2.40.0 |
| Language | Java/C/C++ |
| Segment | Robot |

### Issue Description
I'm building an Android App, which 
1. obtains frameset (`com.intel.realsense.librealsense.FrameSet`) in android java code, 
2. then passes the frameset to some existing C++ code for further processing. The C++ processing function takes argument of `rs2::frameset`.

My question is: How to pass the Java `com.intel.realsense.librealsense.FrameSet` object to C++ code, obtaining a C++ `rs2::frameset` object so I can use the existing C++ processing function ? Thanks in advance !

PS: I see that in `FrameSet.java` file of librealsense, several native functions are declared (e.g. `nFrameCount`, `nExtractFrame`, etc.). But in the corresponding CPP file (`android/jni/frameset.cpp`), it seems that the `jlong` "handles" passed from android to native code, is converted to pointers to `rs2_frame` (e.g. `reinterpret_cast<rs2_frame *>(handle)`). This has confused me.

zc08, 2021-02-04:
What didn't work: 
In Java:
`long frame_handle = frames.getHandle();  // FrameSet frames` 
and passing `frame_handle` to the native part as `jlong`.

In C++:
`rs2::frameset * rs2_frameset_ptr = reinterpret_cast<rs2::frameset*>(frame_handle);`
The fields such as `_size` do not make sense.

zc08, 2021-02-04:
The following code seems to work for several frames, but then the program crashes. I guess something has corrupted the memory. (After commenting out the 3 lines, the program runs well)

```
    rs2_frame *rs2_frameset_ptr = reinterpret_cast<rs2_frame *>(frame_handle);
    rs2::frame rs2_frame(rs2_frameset_ptr);
    rs2::frameset rs2_frameset(rs2_frame);
```
Does anyone know something about the conversion ?  The problem is really haunting me. Thanks in advance ! @MartyG-RealSense

MartyG-RealSense, 2021-02-04:
Hi @zc08   The subject of accessing C++ from Java is outside of my knowledge.  I did research your question though, and the most promising looking reference that I could find about learning Java to C++ communcation in an Android application is at the link below:

https://github.com/IntelRealSense/librealsense/issues/4796

zc08, 2021-02-07:
I'll leave my current solution here in case it might help someone coming across similar problems.

The problem seems to be: The C++ interface (`rs2::frame`) will decrease the reference count of frame upon destruction. Passing the internal pointer as `jlong` effectively bypasses the reference counting mechanism and leads you to error. So you have to manually adjust the reference counts a bit. (I guess using pure C interface will not lead to these troubles as there's no destructor that decreases ref count.)

With too many reference counts the resources are not released and you soon get an "unable to allocate new frame" exception, while too few counts get you segfault. 

For me, 
1. reading frames using the C++ interface
2. manually increasing the ref count by 1 before passing the data to java (`rs2_frame_add_ref`)
3. In java, after using the frames, manually call `Frameset.close()` to release the data

solves the problem.

MartyG-RealSense, 2021-02-07:
Thanks a lot @zc08 for sharing your solution!  

Do you require further assostance with this case, please?

MartyG-RealSense, 2021-02-13:
Case closed due to no further comments received.

closed 2021-02-13 by MartyG-RealSense

ChrisSmeets, 2024-02-14:
I'm also trying a same solution, but I have also troubles to get it working properly. It is possible to elaborate it more with source / example?
I'm using the Android wrapper for the realsense SDK. We are showing a preview in Java and send the frameset via JNI to C++ .
But I get errors in the android logcat like "out of resources" and that some SDK methods give errors . 
Like 
`signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2024-02-14 14:08:31.099 10672-10672 DEBUG                   pid-10672                            A  Cause: null pointer dereference`

MartyG-RealSense, 2024-02-14:
Hi @ChrisSmeets  https://github.com/IntelRealSense/librealsense/issues/5424 may be a helpful reference for your issue.

MartyG-RealSense, 2024-02-15:
@ChrisSmeets  My knowledge of Java programming is very limited.  Comparing your script to a script in https://github.com/IntelRealSense/librealsense/issues/11514#issuecomment-1461847011 though, these lines seem to be okay:

```
try(FrameReleaser fr = new FrameReleaser()) {
try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {
```

What happens if you comment out `DepthAnalyzer.setImage(frames.getHandle());`

ChrisSmeets, 2024-02-19:
@MartyG-RealSense It works like expected when I remove the call. When I use the upload, to show the preview, everything is fine. But I want to use the frames for processing, not for viewing only. I integrated the dlib solution in the examples together with the realsense sdk in our app. But I doesn't work because I can't figure out how to implement the java and C++ part without having problems, I use the handles and I tried a lot of things, but somehow something is not right. There is no one at the realsense team who can assist me? If I receive the handle (I think a kind of pointer?) from the java part, how can I create a frameset in C++ from it and retrieve the color and depth frame and use it in a function like  "rs2_project_color_pixel_to_depth_pixel" correctly?

And for how long does a frameset or frame exists? Is it a kind of queue and is the handle / data removed in the meanwhile ?
It this maybe a problem? And can I save the frameset for later usage ?

MartyG-RealSense, 2024-02-22:
That's great to hear that you achieved a solution, @ChrisSmeets - thanks very much for the update!