realsenseai/librealsense #8299: Python set min/max distance for depth visualization

surefyyq, 2021-02-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.41                     | 
| Language                            |  python                          | 

### Issue Description
In Intel RealSense Viewer, there is a configure at Stereo Module→Depth Visualization→Min/Max Distance, which shows differently for the same scenes when it changed like below
[image]
distance 0.3-2.7:
[image]
distance 0.3-0.6:
[image]
According to my understanding, it is not about depth clamp or threshold filter, but just reduces the scope of visualization to show the depth changes of nearby objects in a more fine-grained way, and loss no depth information.
Am I wrong? If not, how can I set min/max distance of depth visualization in Python?

And another question: there are so many parameters in advanced mode of device, where can I find a detailed explanation and corresponding setting method in Python for each item? Do I have to figure them out literally?

Thanks a lot!

MartyG-RealSense, 2021-02-03:
Hi @surefyyq   The guide documentation for the RealSense Viewer states that the Min Distance and Max Distance sliders under the **Depth Visualization** section of the side-panel options is for the minimum and maximum distance for **color vizualization**.   The color shading of the depth image can be altered manually by the sliders when the **Histogram Equalization Enabled** option directly above the sliders is disabled.

In the Viewer, the minimum and maximum distance at which depth detail is rendered can be set by the **Threshold Filter** in the **Post-Processing** options section.  Depth data outside of the defined range is not rendered on the depth image.

[image]

You can define a threshold filter in Python.  

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.threshold_filter.html?highlight=threshold

In regard to the **Advanced Mode** parameters, they are not fully documented because they interact with each others in complex ways and so Intel chose to control them with machine-learning algorithms instead of producing full documentation for them.  You are free to perform trial and error experimentation with the settings though to observe how changes affect the image.

surefyyq, 2021-02-04:
@MartyG-RealSense Thanks for your quick reply!

Like the python Doc says, Threshold Filter in Post-Processing section will filter out depth values that are either too large or too small,  which means some extreme values of depth information will be discarded. 
However, the max/min distance control in Depth Visualization section doesn't seem to be implemented that way: when distance is set to 0.3-0.6, depth value of pixels less than 0.3 meters away from the sensor are retained. This configure only changes colorized result.

What I want to do is visualizing the depth image of objects close to the sensor (0.3-0.6) without lossing any depth information. Like the two images I post above, the white round objects can be observed more clearly when distance range in Depth Visualization is set smaller. How can I set distance of Depth Visualization in Python? Are there any methods in rs.colorizer I can use?

Maybe @dorodnic knows about it?

Thank you!

MartyG-RealSense, 2021-02-04:
Hi @surefyyq   If your goal is to use Python to set the color scheme and depth color-shading functions featured in the depth visualization options of the Viewer, you can do so with the scripting in the link below.

https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

surefyyq, 2021-02-05:
@MartyG-RealSense Thanks for your research, the link solves my problem. Python code to set max/min distance of depth visualization is like below:

```
colorizer = rs.colorizer()
colorizer.set_option(rs.option.visual_preset, 1) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
colorizer.set_option(rs.option.min_distance, value_min)
colorizer.set_option(rs.option.max_distance, value_max)

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
```

You are free to close this issue now, thank again~

MartyG-RealSense, 2021-02-05:
Great to hear @surefyyq - thanks for the update and the sharing of your solution!

closed 2021-02-05 by MartyG-RealSense