realsenseai/librealsense #8300: Viewer crash on Win7 with SEH exception

noirflux, 2021-02-03:
| | |
|---|---|
|**librealsense**|2.41.0 RELEASE|
|**OS**|Windows 7 x64 Enterprise|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```
This happens after installing 2.41.0.2657, and running the viewer, whether the camera is plugged in or not.  Installing the Win7 driver was selected during installation.  This is on a Lenovo X230T laptop.

noirflux, 2021-02-03:
Hmmm, the depth quality tool runs fine, get good data from the camera (D435).  But trying to run the viewer again gives the above crash and error every time.

Hmmm #2, if I click the Enable Metadata button, I get another crash:
[image]

MartyG-RealSense, 2021-02-03:
Hi @noirflux   I hope that the information in the link below will answer your question and enable you to use the RealSense Viewer program.

https://github.com/IntelRealSense/librealsense/issues/8284#issuecomment-770748587

The technical specification for the Lenovo X230T lists it as having a 3rd Generation Intel processor chip.   The version of the Viewer that is installed as part of the SDK installation requires 6th Generation or newer from SDK version **2.36.0** onwards.  Using the standalone version of the Viewer mentioned in the link above wll hopefully correct your problem.

noirflux, 2021-02-03:
Thanks, but no joy.  I downloaded that viewer, and it still crashes on startup, if the camera is plugged in or not.  But this time with a different error:  Unhandled exception at 0x779CEAEB (ntdll.dll) in Intel.RealSense.Viewer.exe: 0xC0000374: A heap has been corrupted (parameters: 0x77A04270).

Going into the debugger shows:
[image]

The SDK Depth Quality Tool still runs fine.

noirflux, 2021-02-03:
Interestingly, the stand-alone version of the Depth Quality Tool from that v2.38.1 page crashes in exactly the same way.

MartyG-RealSense, 2021-02-03:
Let's focus on the **Unhandled exception escaping from a worker thread next**, as that other error is prominent in your kindly supplied images.

It is possible that you are affected by both problems - an incompatible processor on the SDK-bundled Viewer, and the worker thread error on both the bundled and standalone Viewer versions.

You may have already read about the couple of previous cases involving SDK 2.38.1 that were affected by the worker thread error.

https://github.com/IntelRealSense/librealsense/issues/7666

The advice in those cases was to try upgrading the camera firmware.  Which firmware version is your D435 currently using, please?

For SDK **2.38.1**, firmware version **5.12.7.100** would be suitable.

For SDK version **2.39.0** and newer, the most recent firmware version at the time of writing this - **5.12.10.0**  - would be appropriate.  This is because after SDK 2.39.0, the recommended minimum firmware became **5.12.8.200** due to internal firmware changes.

noirflux, 2021-02-03:
Wow, if down-rev firmware on a camera that's not even plugged in is causing this, that's some SERIOUS quantum entanglement!  These cameras are cooler than I thought!  (:P)  I will try it though, as maybe the error is caused by an unrecognized device and once the firmware is correct it may work.  Fingers entangled!
 
I'll check the firmware on another Win10 laptop, as even the firmware .exe crashes at startup too with heap corruption.  I notice in the above-mentioned thread that the problem was fixed there by using another "interface" (I assume USB port) so not sure how similar these problems are.  But I'll try, thanks.

MartyG-RealSense, 2021-02-16:
Case closed due to no further comments received.

closed 2021-02-16 by MartyG-RealSense

noirflux, 2021-04-04:
This still happens with v2.43.0, but if built from source on the above Win7 machine, the viewer works fine.  Soooo, why isn't the version in the Win7 release built to be compatible, as it apparently can be?

MartyG-RealSense, 2021-04-04:
Are you building Windows 7 from source by using CMake to build the source-folder from the **Assets** file list on your Windows 7 computer, please?   Or obtaining the code from the **Code** controls of the librealsense GitHub?

[image]

[image]

As mentioned in the Windows 7 source-code installation instructions in the link below, librealsense can be built from source on Windows 7 (which automatically generates binaries for Windows 7), **or** on Windows 10 using the CMake build flag **DFORCE_WINUSB_UVC=true**.  I do not have knowledge of which method is used to create the Windows 7 binary on the **Releases** page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md#building-from-source

noirflux, 2021-04-05:
I am building the Win7 on Win7, and Win10 on Win10, using cmake.  I have done so both from the assets and from github code -> download zip.  Again seems like the actual release build should be done in the most compatible way and save other poor (but stable!) Win7 users some pain.  Thanks!