realsenseai/librealsense #8302: Unable to show 2D images when I connect L515

AllenZhaoYB, 2021-02-03:
Following is the required info.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (2.41.0) | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0                                         | 
| Platform                                 | PC(Alienware M15R3) |
| SDK Version                            |  { legacy / 2.<41>.<0> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I am now facing a problem. I build the pyrealsense2 with CUDA. When I connect the L515 camera, the RGD image can not be seen while other depth camera and infrared image can be shown on the screen. 
Bellow are some errors.
When I first  connect the L515, and run realsense-viewer, these errors come out.
03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Backlight Compensation control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Brightness control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Contrast control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Exposure control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Gamma control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Hue control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Saturation control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,732 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Sharpness control for RGB Camera
 03/02 18:59:40,732 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,732 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,733 WARNING [140679260975104] (sensor.cpp:1110) Failed to add White Balance control for RGB Camera
 03/02 18:59:40,733 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,733 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,733 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Enable Auto Exposure control for RGB Camera
 03/02 18:59:40,733 ERROR [140679036530688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/02 18:59:40,733 ERROR [140679260975104] (sensor.cpp:523) acquire_power failed: failed to set power state
 03/02 18:59:40,733 WARNING [140679260975104] (sensor.cpp:1110) Failed to add Enable Auto White Balance control for RGB Camera
 03/02 18:59:56,220 WARNING [140678994567168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 18:59:56,330 WARNING [140678994567168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 18:59:56,590 WARNING [140678994567168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 18:59:56,696 WARNING [140678994567168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 18:59:56,860 WARNING [140678994567168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
----------------------------------------------------------------------------------------------------
Then I run the realsense-viewer with sudo, the error differs:
03/02 19:01:12,642 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 19:01:12,747 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 19:01:13,145 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Backlight Compensation control for RGB Camera
 03/02 19:01:13,157 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Brightness control for RGB Camera
 03/02 19:01:13,169 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Contrast control for RGB Camera
 03/02 19:01:13,182 WARNING [140100373549056] (messenger-libusb.cpp:42) control_transfer returned error, index: 256, error: No data available, number: 61
 03/02 19:01:13,192 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Exposure control for RGB Camera
 03/02 19:01:13,203 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Gamma control for RGB Camera
 03/02 19:01:13,214 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Hue control for RGB Camera
 03/02 19:01:13,224 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Saturation control for RGB Camera
 03/02 19:01:13,235 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Sharpness control for RGB Camera
 03/02 19:01:13,246 WARNING [140100876881920] (sensor.cpp:1110) Failed to add White Balance control for RGB Camera
 03/02 19:01:13,257 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Enable Auto Exposure control for RGB Camera
 03/02 19:01:13,268 WARNING [140100876881920] (sensor.cpp:1110) Failed to add Enable Auto White Balance control for RGB Camera
 03/02 19:01:14,367 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 19:01:14,471 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 19:01:14,635 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 19:02:00,618 WARNING [140100851703808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/02 19:02:01,608 ERROR [140100390334464] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 133
[... cut ...]
 03/02 18:50:09,136 ERROR [139920010100736] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: p_counted_deleterIP4int2ZN6rscuda9alloc_devIS0_EESt10shared_ptrIT_EiEUlS1_E_SaIvELN9__gnu_cxx12_Lock_policyE2EE

----------------------------------------------------------------------------------------------------
I can see NVIDIA GPU use 93MB. 
The questions is: When I connect the L515, I can see the depth and infrared image, while the RGB image part shows "warning! no frames received!"
Can Anyone help me solve this problem! Thanks

AllenZhaoYB, 2021-02-04:
[image]

RealSenseSupport, 2021-03-02:
Hi,

Do you see this issue if you build without CUDA? And are you using the native backend via the kernel patch or RSUSB backend?

Thanks

closed 2021-04-02 by RealSenseSupport