realsenseai/librealsense #8303: Low precision

GabrielMLB, 2021-02-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  5.4.0-65-generic                                         | 
| Platform                                 | PC (Notebook)  |
| SDK Version                            |  2.0                           | 
| Language                            |  Python                          | 
| Segment			|  Others                          | 

### Issue Description

Hello guys, I am with a very low precision when I calculate the distance between 2 points, I use pyrealsense2 and python3 (I use the depth frame), below a code example: 

```python
    def calculate_distance_mean(self, p1, p2):

        z1 = list()
        z2 = list()

        # ix, iy = self.ix, self.iy

        p1_x_range = np.array(range(-self.pixel_window, self.pixel_window + 1)) + p1[0]
        p1_y_range = np.array(range(-self.pixel_window, self.pixel_window + 1)) + p1[1]

        p2_x_range = np.array(range(-self.pixel_window, self.pixel_window + 1)) + p2[0]
        p2_y_range = np.array(range(-self.pixel_window, self.pixel_window + 1)) + p2[1]

        for x1, y1 in zip(p1_x_range, p1_y_range):
            z1.append(self.depth_frame.get_distance(x1, y1))

        for x2, y2 in zip(p2_x_range, p2_y_range):
            z2.append(self.depth_frame.get_distance(x2, y2))

        z1 = np.array(z1)
        z2 = np.array(z2)

        z1_non_zero = np.nonzero(z1)
        z2_non_zero = np.nonzero(z2)

        x1_mean, y1_mean = np.mean(p1_x_range[z1_non_zero]), np.mean(p1_y_range[z1_non_zero])
        x2_mean, y2_mean = np.mean(p2_x_range[z2_non_zero]), np.mean(p2_y_range[z2_non_zero])
        z1_mean, z2_mean = np.mean(z1[z1_non_zero]), np.mean(z2[z2_non_zero])

        point1 = rs.rs2_deproject_pixel_to_point(self.color_intrinsics, [x1_mean, y1_mean], z1_mean)
        point2 = rs.rs2_deproject_pixel_to_point(self.color_intrinsics, [x2_mean, y2_mean], z2_mean)

        dist = math.sqrt(
            math.pow(point1[0] - point2[0], 2) + math.pow(point1[1] - point2[1], 2) + math.pow(
                point1[2] - point2[2], 2))

        return dist
```

Here I pass the coordinates of 2 points, create a pixel window to calculate an average of non-zero values, avoiding noise, occlusion ...

I obtain the average value of (x,y,z) and thus calculate the euclidean distance.

An object of 15cm the function estimates between 20cm and 24cm.

I would like an error <5cm

MartyG-RealSense, 2021-02-03:
Hi @GabrielMLB   You may beneft from added accuracy if you perform depth-color alignment.   In librealsense's own **rs-measure** example program, the approach of aligning **color to depth** (instead of the more comon depth to color) is used.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

Performing alignment can make it easier to distinguish background pixels from foreground pixels..

The alignment approach is also used in a Python conversion of rs-measure by RealSense community member **soarwing52**, though the conversion seems to take the **depth to color** alignment approach.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

GabrielMLB, 2021-02-03:
Hello @MartyG-RealSense, thank you very much for the shared links I will access.

I use the following alignment:

```python

self.align = rs.align(rs.stream.color)
...

     frames = self.pipeline.wait_for_frames()
     frames = self.align.process(frames)

      depth_frame = frames.get_depth_frame()
      color_frame = frames.get_color_frame()

      self.depth_frame = depth_frame
      filter_image = self.post_process_depth_frame(depth_frame)

```

MartyG-RealSense, 2021-02-03:
Thanks very much for sharing your align code.   Does the above code improve your measurements, please?

GabrielMLB, 2021-02-03:
The measurements vary between 20cm and 23cm.

As the distance of the object increases, the error is .also increasing.

The camera is 40cm from the object.

MartyG-RealSense, 2021-02-03:
I cannot see a definition for a custom stream resolution in your code, so I assume that the camera is using the default resolution settings for that particular camera model (which are the ones with optimal depth accuracy).

It is normal for error to increase as distance of the observed object or surface from the camera increases. This phenomenon is called **RMS error**.  However, at a 40 cm (0.4 meter) close range, RMS error should be relatively low.

Which camera model are you using please?  If you were using a **D455** then I can see why you may have depth issues, as the minimum depth distance of that model is 0.4 m.  Around or below that distance from the camera the depth image may start to break up.  Setting the **Disparity Shift** option to a value of **50** to reduce the minimum depth sensing distance of the camera may help.

The next test that you could try for enhancing the close-range depth image is setting the **depth unit scale** to 0.0001 instead of the default 0.001.

https://github.com/IntelRealSense/librealsense/issues/1905

I recently tested the above depth unit scale and disparity shift values with an aligned point cloud to improve near-range image quality.

https://github.com/IntelRealSense/librealsense/issues/8228

MartyG-RealSense, 2021-02-03:
Would it be possible to describe the object / surface that the camera is observing or provide an RGB image of it please?  The physical materials may have a bearing on the depth sensing measurements.  For example, the camera may be less able to analyze a surface for depth if it is colored dark grey / black, if it is reflective or it has a strong light source projecting onto its surface.

It may also be worth taking a distance reading from the camera to the object to test whether that distance is measuring accurately.  If it is, then the measurement problem may be with the code that is calculating the measurement between two points.

GabrielMLB, 2021-02-03:
Perfect thank you very much for sharing, I will make the suggested changes and evaluate the RMS error.

I am using the D435i model.
I shared in the link below the example file and the image (I use the real time algorithm).

[https://1drv.ms/u/s!At7Yi4YwjQcBnxg0cTDTZEXb-GNy?e=YxkzOy](https://1drv.ms/u/s!At7Yi4YwjQcBnxg0cTDTZEXb-GNy?e=YxkzOy)

GabrielMLB, 2021-02-03:
File atulized

[https://1drv.ms/u/s!At7Yi4YwjQcBnxtY-S1WngpdLwxv?e=P42XwJ](https://1drv.ms/u/s!At7Yi4YwjQcBnxtY-S1WngpdLwxv?e=P42XwJ)

MartyG-RealSense, 2021-02-03:
Thanks very much.  I will post the image here for easy reference.

[image]

I can see in the image some of the conditions I described above that may negatively affect the depth image.  The top of the cup is not visible except where the natural light falls upon an edge of it because it is black and so cannot be seen by the camera.  The background is also very faint on the depth image because it is a dark grey background.  

The bright window is another area that would be very hard to get depth information from.  This kind of surface would be more readable if a **linear polarization** physical filter was applied over the top of the camera lens on the outside of the camera casing, though the window may not be important enough to your project to do so.

GabrielMLB, 2021-02-03:
Perfect, thanks a lot for all the help, my final project will be in an open environment exposed to the sun (measuring the size of an animal and estimating the weight with these measurements, very challenging!), but I'm testing several scenarios, to analyze what is the best camera configuration for different types of scenarios. I will test the polarization filter as well and see the results.

Thanks for the feedback

GabrielMLB, 2021-02-03:
I will prepare a better scenario so that it doesn't harm the camera's performance too much, thank you very much!

MartyG-RealSense, 2021-02-03:
You are very welcome.  Thanks very much @GabrielMLB for the detailed feedback, and feel free to return to this forum with new questions if you require further advice as your project develops  Good luck!

GabrielMLB, 2021-02-03:
Thank you @MartyG-RealSense!

MartyG-RealSense, 2021-02-10:
Hi @GabrielMLB   Do you require further assistance with this case, please?   Thanks!

GabrielMLB, 2021-02-10:
Hi @MartyG-RealSense, thanks a lot for your help, we can close the topic.

MartyG-RealSense, 2021-02-10:
Thanks very much @GabrielMLB for the update!  As suggested, I will close the case.  Thanks again!

closed 2021-02-10 by MartyG-RealSense